eclipse-embed-cdt/eclipse-plugins #291: Hide the Toolchains tab for non managed projects

ilg-ul, 2018-04-05:
If the build configuration is not managed, hide the Toolchains tab.

ilg-ul, 2018-04-05:
Fixed on 2018-04-05.

ilg-ul, 2018-04-05:
The 4.3.3.201804051758 pre-release is now available for testing from http://gnu-mcu-eclipse.netlify.com/v4-neon-updates-test.

ilg-ul, 2018-04-19:
Fixed since 4.3.3.201804191501.

ilg-ul, 2018-05-29:
Reopened due to #303.

The condition is too restrictive, it must probably be only related to GME, not to unmanaged.

sebastian-new, 2018-05-30:
Hello @ilg-ul . 

I tried this from #5 in 20180419-1818-gnumcueclipse-4.3.3-oxygen-3a-win32.win32.x86_64.zip

and it works. The settings tab allows you to assign device after you click on 'Generate Makefiles Automatically'

And this is for an unmanaged project.

ilg-ul, 2018-05-31:
The requirement for a managed project was relaxed since 4.3.4, 2018-05-31.

ilg-ul, 2018-05-31:
@nolange, please check the version from updates-test and let me know if the Toolchain tab is now visible in your setup.

closed 2018-07-21 by ilg-ul