eclipse-embed-cdt/eclipse-plugins #292: Global build tools path loads legacy path

paunet, 2018-05-03:
### Description

Global build tools loads legacy path, or previous installation path.

### Steps to Reproduce

1. Install a fresh new copy of eclipse, and the GNU MCU plugin from the marketplace.

2. Open Windows/Preferences/MCU/Global Build Tools Path
3. [image]

**Expected behaviour:**
In a fresh installation the path should be empty

**Actual behaviour:** [What actually happened]
Legacy path set,

### Versions
    [plug-in version] 2.6.2
    [Eclipse version] 4.7.3
    [Java version] 8.0.162
    [operating system] Windows 10 64 bit
    [toolchain version] None

TM1234, 2018-05-03:
What exactly do you mean by "previous/legacy" path here?
I suspect that the plugins and/or CDT use any tools that they find on the path and that is what's happening here?

ilg-ul, 2018-05-03:
> In a fresh installation the path should be empty

nope. the paths should be initialised to reasonable defaults.

in general, if you install the binaries in known locations, the most recent one will be selected for the default path.

on Windows, in previous versions, installing via setup left several keys in the Registry, and the plug-ins uses them in the search.

did you install the old Build Tools v2.8 via the setup?

paunet, 2018-05-03:
No, I downloaded the zip file from the relases folder and unzip in an alternative folder, then in the previous eclipse/gnu mcu eclipse I set the path in the Global Build Tools Path.

You commented keys in Windows Registry, Should I try to find thoses keys and erase or delete them?

TM1234, 2018-05-03:
What is your PATH?

paunet, 2018-05-03:
Windows Path doesn't contain any reference to Build Tools, but Global Build Tools Path starts with the previous location of the Build Tools

ilg-ul, 2018-05-03:
> but Global Build Tools Path starts with the previous location of the Build Tools

yes, but this is just a default. if you don't like it, simply define your own path.

paunet, 2018-05-03:
If I try to delete the Global Build Tools Path, after restarting eclipse I end up with this:

[image]

ilg-ul, 2018-05-03:
> If I try to delete the Global Build Tools Path

an empty field will automatically search forthe default.

don't delete, define the actual path.

paunet, 2018-05-03:
It is solved now, I checked and erase all the entries in regedit with "GNU ARM Eclipse"
After that, leaving empty the Global Build Tools Path don't add any extra path in the Environment

closed 2018-05-03 by paunet