eclipse-embed-cdt/eclipse-plugins #293: Disable Toolchain & Build Tools Path Check (project level)

paunet, 2018-05-08:
### Description
Disable Toolchain & Build Tools Path Check (project level)

### Steps to Reproduce
We need to set a Windows System Variable as a Toolchain Path
1) Using Project\Properties\MCU
[image]

2) I f we manually edit (.settings\ilg.gnumcueclipse.managedbuild.cross.arm.prefs):
  eclipse.preferences.version=1
  toolchain.path.962691777=${GCC_ARM_7_Q4}
It works but when we are in Project\Properties\MCU we can't leave the window.

3) The question is if there is a flag to disable project toolchain/build tools path check at the project level like 

//Set to false to disable the folder check.
global.toolchain.path.strict=true
global.buildTools.path.strict=true
workspace.toolchain.path.strict=true
workspace.buildTools.path.strict=true

for global/workspace

### Versions
[plug-in version] 2.6.2
[Eclipse version] 4.7.3
[Java version] 8.0.162
[operating system] Windows 10 64 bit
[toolchain version] None

ilg-ul, 2018-05-08:
Yes. `project.toolchain.path.strict`.

But, if you need it, I would say that you somehow missed the point with the path mechanism.

Did you try the new method to install the toolchain with `xpm`?

ilg-ul, 2018-05-08:
> where should I store

https://gnu-mcu-eclipse.github.io/developer/eclipse/runtime-preferences/#defaults

> The main reason to use System Paths is due the lack of tools in our CI machine.

`xpm` is a command line tool, you can use it on any CI.

The link below shows a sequence of test builds for one of my projects:

https://ci.appveyor.com/project/ilg-ul/sifive-templates-xpack

What happens there is to crete projects by instantiating a template, satisfying dependencies and building the project (both debug & release).

I run similar builds on Travis.

paunet, 2018-05-08:
We are trying:
   project/.settings/<pluginId>.prefs

project\.settings\ilg.gnumcueclipse.managedbuild.cross.arm.prefs (project.toolchain.path.strict=false)
project\.settings\ilg.gnumcueclipse.managedbuild.cross.prefs (project.buildTools.path.strict=false)

but nothing works

Can we have a project (project\.settings\*) file/s containing the toolchain/buildtools (using windows system variables ${GCC_ARM_7_Q4} ${GCC_BUILD_TOOLS_2_11}) and if needed path check disabled?

project.toolchain.path.strict=false
project.buildTools.path.strict=false

We will have a look at xpm, thanks

ilg-ul, 2018-05-08:
> but nothing works

of course they don't, they are defaults:

https://gnu-mcu-eclipse.github.io/developer/eclipse/runtime-preferences/#defaults

paunet, 2018-05-08:
> of course they don't, they are defaults:

So, the only way to disable path check at project level is to disable the defaults.

That means that we can't have a portable way of disabling paths checks (stored in project\.settings)

ilg-ul, 2018-05-08:
> the only way to disable path check at project level is to disable the defaults.

that's correct. 

these are not properties/preferences to be reconfigured dinamically, but defaults to be set once, before starting Eclipse and aplying globally.

paunet, 2018-05-08:
Many Thanks!

closed 2018-05-08 by paunet