eclipse-embed-cdt/eclipse-plugins #294: When starting Eclipse, I get the message "Incompatible JVM...

raysimar, 2018-05-10:
### Description

When starting Eclipse, I get the message "Incompatible JVM: Version 1.6.0_65 of the JVM is not suitable for this product.  Version: 1.8 or greater is required.

Thanks!

### Versions
* 20180419-1818-gnumcueclipse-4.3.3-oxygen-3a-macosx.cocoa.x86_64.tar.gz
* Java 8 Update 171
* macOS High Sierra - Version 10.13.4

TM1234, 2018-05-10:
You must have 1.6 installed and eclipse is picking that up perhaps from the system path.
Put 1.8 in a folder named .../eclipse/jre and eclipse will use that instead.

Hope this helps.

ilg-ul, 2018-05-10:
Tommy is right, you need a more recent Java.

closed 2018-05-10 by ilg-ul

raysimar, 2018-05-10:
Hi Tommy,

Thanks for the tip.  Where should the folder …/eclipse/jre reside?

All the best,
Ray

ilg-ul, 2018-05-10:
please don't use email to post here.

better don't bother to keep multiple Java version; remove your Java 1.6 and install the latest Java from Oracle.

TM1234, 2018-05-10:
The jre folder containing your jre or jdk (I always use the former) should be in your eclipse folder containing your eclipse/cdt (I actually use the eclipse platform runtime binary bundle and manually add just the specific plugins needed for c/c++ and game development but just using eclipse/cdt is easier if a bit more bloated).

Alternatively you can configure eclipse.ini to point to a specific Java installation. There are lots of tips on this if you do a search.

TM1234, 2018-05-10:
In that case have one Java installation and use eclipse.ini to point to it.

https://wiki.eclipse.org/Eclipse.ini

We (Microsemi) include everything in one bundle in SoftConsole so we use the jre folder approach to ensure that the specific Java jre that we test with is used automatically by default.

raysimar, 2018-05-10:
> should be in your eclipse folder
>
> if you have multiple eclipse folders, you install jdk in all of them?

Hi all,

What I did that seemed to fix the problem for me was to download the JDK (10.0.01).  I had been downloading just the JRE.  Then I ran this command and got this result:

Rays-MacBook-Pro:~ Ray$ /usr/libexec/java_home --verbose
Matching Java Virtual Machines (3):
    10.0.1, x86_64:	"Java SE 10.0.1"	/Library/Java/JavaVirtualMachines/jdk-10.0.1.jdk/Contents/Home
    1.6.0_65-b14-468, x86_64:	"Java SE 6"	/Library/Java/JavaVirtualMachines/1.6.0.jdk/Contents/Home
    1.6.0_65-b14-468, i386:	"Java SE 6"	/Library/Java/JavaVirtualMachines/1.6.0.jdk/Contents/Home

/Library/Java/JavaVirtualMachines/jdk-10.0.1.jdk/Contents/Home

So, the 1.6 had been lurking around and was creating the problem, but installing the JDK seemed to get things back on track.  If others see a similar problem, maybe have them use the command above to see what the system is seeing.

Thanks for the help!

All the best,
Ray

TM1234, 2018-05-10:
Or just use the jre folder or eclipse.ini approach that I mentioned above to ensure that eclipse uses a specific Java.

ilg-ul, 2018-05-10:
> the 1.6 had been lurking around

unless you have very old Mac applications that explictly ask for Java 1.6, you better remove 1.6 completely.

if you insist on keeping it, as long as it works you're fine, but after a system update things might break again, so at that time remember Tommy's advice to use `eclipse.ini` to tell Eclipse the exact location of the desired Java.

raysimar, 2018-05-10:
Thank for the advice!

You’ve done a great job on this tool.  I’m planning on using it for RISC-V related programming.

All the best,
Ray