eclipse-embed-cdt/eclipse-plugins #302: Problems with "General Registers" view

pvonmoradi, 2018-05-27:
Thanks for your great work in developing this much appreciated plugin :D
I'm using openOCD + bluepill (STM32F103). I've also downloaded the necessary Pack for the device and chose the device in project settings. 
1- There is no General Registers (R0,R1,...,PC,SP,...) peripheral available. The webpage explanation about Keil/Arm Peripherals mentioned that some vendors *may* not include them but maybe these registers could be accessed from somewhere else?
2- There is a General Register section in Registers view but I guess this comes from query from openOCD not some address in MCU.
3- Anyway, How can I make the Value field fixed width hex? I can select different formats but fixed 32-bit hex is not among them.

Thank you again

TM1234, 2018-05-27:
I suspect that Liviu will close this as it's not really a valid bug reports in my opinion,
But anyway ...

#1 and #2 don't make sense to me. General registers are not peripherals and are not memory mapped so they have to be read (e.g. by OpenOCD) from the CPU and not from the memory map. CDT provides a "general"  CPU registers view which displays the relevant RISC-V integer, floating point and CSR regs so that's where you find them. 

My recollection is that Eclipse/CDT is very quirky (and possibly buggy) when it comes to remembering size/format settings for the likes of registers so if you can't configure this so that it persists then it's more likely to be an Eclipse/CDT rather than a GNU MCU Eclipse plugin bug/issue and should be logged there not here.

Hope this helps.

TM1234, 2018-05-27:
Sorry - I see that you're using Cortex-M3 and not RISC-V but my comments still stand.

pvonmoradi, 2018-05-27:
@TM1234 Thanks for your reply.
You are right. For a moment, I thought this MCU had FPU and I wondered "where are other registers?" but it doesn't have FPU. And the fact that register names were lower-case made me doubt that there might be some other view for General Registers :D
However, about the fixed-width issue, TI's CCS is based on Eclipse. For example here is a debug view [()](https://youtu.be/3ha72Y8pyD4?t=404) of a CCS session. As you can see, Core Register values are fixed width and there are shown alongside other peripherals.

TM1234, 2018-05-27:
In the Register view during a debug session can't you right click in the view and select the format/size of the registers - e.g. so that they are all displayed in 32 bit hex format (0x01234567)?
I can't check this myself at the moment as I don't have access to hardware.

pvonmoradi, 2018-05-28:
@TM1234 It does have hex but it would not be rendered as fixed-width.
[It looks like this](https://imgur.com/a/jBygjkG)

TM1234, 2018-05-28:
OK - I see what you mean.
But I'm pretty sure that that's a CDT and not a GNU MCU Eclipse issue so you should log it against CDT and not here.
I can't see any option to control/enforce this rendering in Eclipse/CDT.
If TI's Code Composer Studio can render fixed width 32 bit hex values then they might be using a modified version of the relevant plugin.
BTW - I think that Liviu may be offline/on time off at the moment so you may not get anything from him on this for the moment.

pvonmoradi, 2018-05-28:
@TM1234 Thank you anyway :D

ilg-ul, 2018-05-29:
The list of general registers comes from the GDB server, and the CDT Eclipse debug plug-ins do nothing more than display them. The GNU MCU Eclipse debug plug-ins do not change this behaviour.

As for the format, it is also under CDT control, the GME debug plug-ins do nothing to change this.

The truth is that the CDT debug plug-ins are a big concern, since any improvement in the way they work or the way they interract with the user, generally requires major rewrites, and most commercial products had to take this path.

closed 2018-05-29 by ilg-ul