eclipse-embed-cdt/eclipse-plugins #303: nRF52840 peripherals view

sebastian-new, 2018-05-28:
**Description**
Peripherals view is empty as described in 
https://github.com/gnu-mcu-eclipse/eclipse-plugins/issues/278

**Steps to Reproduce**
Install packs as explained in normal pack installation. I am not sure where the nRF pack comes from
but the GNU ARM C/C++ Packs version is 2.2.2.20170425
assign nRF52840 as device in the project properties,
enter debug perspective
open Peripherals view
Expected behaviour: [What you expected to happen]

In the peripherals view there should be a list of peripheral registers of nRF52840. There is none and a red warning appears on the bottom of the IDE noting: 'Assign a device to the project'

Actual behaviour: [What actually happened]

No registers visible in the list - list empty

**Versions**
[plug-in version]
4.3.3.20180419

[Eclipse version]
Oxygen.3a Release (4.7.3a)

[Java version]
1.8.0_171-b11

[operating system]
Windows 10

[toolchain version]
gcc-arm-none-eabi-7-2017-q4-major

I also tried PheeL79 workaround from #286  by editing the nRF52840.svd file and removing the BOM character and saving the file with UTF-8 and still did not get it to work.

Please excuse my reporting. I am new to github and eclipse.

ilg-ul, 2018-05-28:
do you see any messages in the packs console? in the workspace .metadata/.logs ?

sebastian-new, 2018-05-29:
Hi, 

Thanks for coming back. **The packs console log is the following**:

2018-05-29 06:55:34
Loading repos summaries...
Parsing cached content file "C:\nRF52840\Packages\.cache\.content_www_keil_com_pack_index_pidx.xml"...
File parsed in 236ms.
Identifying installed packages...
Found 13 installed packages.
Completed in 321ms.

2018-05-29 06:55:34
Collecting devices...
Found 283 device(s), from 29 vendor(s).
Completed in 6ms.

2018-05-29 06:55:34
Collecting packs...
Found 1580 package version(s), from 36 vendor(s).
Completed in 19ms.

I don't know eaxctly where to find the workspace .metadata/.logs but in the workspace console I don't get any errors or messages related to the packs. Everything seems to load normally and perspective opens up with no errors however, no peripherals view is populated and also I cannot add and track variables.

Any help is greatly appreciated.

Thank you.

TM1234, 2018-05-29:
.metadata/.log is in your workspace directory. It may be hidden by default depending on what system you're using.

sebastian-new, 2018-05-29:
Well I am using Windows 10. It must be because in the workspace .metadata I cannot find a /.logs directory.

sebastian-new, 2018-05-29:
Hi,

Ok. Found it. In Windows 10 it is a text file that appears with no name. When you open it it shows as .log.

I am attaching for your reference.
[Log_Copied.txt](https://github.com/gnu-mcu-eclipse/eclipse-plugins/files/2047160/Log_Copied.txt)

ilg-ul, 2018-05-29:
nothing useful there. you did a lot of useless updates, even with wrong urls, but exceptions related to svd parsing.

I'm afraid you are doing something wrong.

are you sure you installed the desired packages and assigned the device to the project? do you see your device name stored in the .cproject file?

sebastian-new, 2018-05-29:
Hello,

I am sure I have probably installed some wrong packages but that shouldn't be a problem. I see the packages and I can surely say that I assigned the correct device to the project.

Please see this abstract from .cproject file:

<storageModule moduleId="org.eclipse.cdt.core.externalSettings"/>
			<storageModule moduleId="ilg.gnuarmeclipse.managedbuild.packs">
				<option id="cmsis.device.name" value="nRF52840_xxAA"/>
				<option id="cmsis.family.name" value="nRF52 Series"/>
				<option id="cmsis.device.vendor.name" value="Nordic Semiconductor"/>
				<option id="cmsis.device.vendor.id" value="54"/>
				<option id="cmsis.device.pack.vendor" value="NordicSemiconductor"/>
				<option id="cmsis.device.pack.name" value="nRF_DeviceFamilyPack"/>
				<option id="cmsis.device.pack.version" value="8.16.0"/>
				<option id="cmsis.core.name" value="Cortex-M4"/>
				<option id="cmsis.compiler.define" value="NRF52840_XXAA"/>
				<memory section="IRAM1" size="0x40000" start="0x20000000" startup="0"/>
				<memory section="IROM1" size="0x100000" start="0x00000000" startup="true"/>

The device is NRF52840_XXAA and it is there.

TM1234, 2018-05-29:
Why not start from scratch with a clean installation of just Eclipse/CDT, Java, GME plugins, tools and try again and see if it still happens?

ilg-ul, 2018-05-29:
> a clean installation of just Eclipse/CDT, Java, GME plugins

you have a distribution with the Eclipse CDT & plug-ins already installed, all you have to do is unpack it to a different folder and use it. be sure you also create a new workspace, in a different folder than before, since this will createa a new Packages folder. 

alternatively you can simply remove the Packages folder, restart Eclipse and install the packs again.

ilg-ul, 2018-05-29:
I did a quick test, I assigned the NRF52840_XXAA device to one of my projects, and the peripherals looked like this:

[image]

sebastian-new, 2018-05-29:
Hello,

I tried doing what you suggested and remove the packages folder and re-install the packs again. 
The problem still persists. 

What I noticed is that although I assign a device to the project, when I open up the debug configuration, the project name and application are filled automatically but the device name is NOT. 

For some reason although the project has an assigned device this is not 'seen' by the debugger (?)

If this means anything I have 64-bit windows but installed the 32-bit Eclipse distribution because the 64-bit didn't work. 

I'll try re-installing the whole distribution from scratch and test both 32 and 64 bit.

Any comments on the above are welcome.

sebastian-new, 2018-05-29:
@ilg-ul the photo you are showing is what it should look like. 

I get an empty peripherals list like this

[image]

TM1234, 2018-05-29:
> If this means anything I have 64-bit windows but installed the 32-bit Eclipse distribution because the 64-bit didn't work.

What do you mean "didn't work"?
While 32 v 64 probably shouldn't matter here it's arguably advisable to use 64 bit stuff wherever possible these days and the 64 bit Eclipse/CDT/GME plugins/Java etc. definitely work in the general case.

sebastian-new, 2018-05-29:
When I try to run the Windows 64 version I get this:

[image]

ilg-ul, 2018-05-29:
Tommy is right, if you install everything as 64-bits it should work. 

In your test it fails because you use the 32-bits Java.

ilg-ul, 2018-05-29:
> Download and extract the Eclipse/CDT ... install the GME plugins

that's the old and hard way.

the new and short way is to directly get and unzip the packed distribution:

https://github.com/gnu-mcu-eclipse/org.eclipse.epp.packages/releases

TM1234, 2018-05-29:
Thanks Liviu. But the reason I do it manually is that I prefer to use the "lean and mean" PRB (Product Runtime Binary) as the basis for my setup containing, add it does, the bare minimum eclipse and then add only whatever else (e.g. selected cdt and other plugins) GME etc absolutely needs or that I want. Rather than taking the (arguably unnecessary) bloat that eclipse/cdt normally provides.

ilg-ul, 2018-05-29:
In your scenario it probably makes sense, but you suggested to download CDT and separatelly install the GME plug-ins; you don't avoid any bloat, you get exactly the same (many) CDT plug-ins, just that you do it in two manual steps instead of one.

TM1234, 2018-05-29:
Fair enough. :-)

sebastian-new, 2018-05-29:
Hi,

I have done as you suggested. I installed jre 8u172 64-bit runtime engine, removed all packages and workspace directories and used 20180419-1818-gnumcueclipse-4.3.3-oxygen-3a-win32.win32.x86_64.zip
to turn to 64-bit version of eclipse CDT.

It runs fine with the new java engine. 

The packages are pre-installed and I also installed the nRF_DeviceFamilyPack as per instructions from Nordic.

**This is the GNU MCU Eclipse packs console:**

2018-05-29 12:42:12
Extracting devices & boards... 
Parsing cached file "C:\nRF52840\Eclipse\Packages\.cache\.installed_devices_boards_books.xml".
Completed in 65ms.

2018-05-29 12:44:37
Loading repos summaries...
Parsing cached content file "C:\nRF52840\Eclipse\Packages\.cache\.content_www_keil_com_pack_index_pidx.xml"...
File parsed in 835ms.
Identifying installed packages...
Found 1 installed packages.
Completed in 931ms.

2018-05-29 12:44:38
Collecting devices...
Found 297 device(s), from 29 vendor(s).
Completed in 10ms.

2018-05-29 12:44:38
Collecting packs...
Found 1784 package version(s), from 37 vendor(s).
Completed in 21ms.

2018-05-29 12:44:49
Parsing "C:\nRF52840\Eclipse\Packages\NordicSemiconductor\nRF_DeviceFamilyPack\8.16.0\NordicSemiconductor.nRF_DeviceFamilyPack.pdsc"...
Parse completed in 151ms.

2018-05-29 12:48:13
Collecting devices...
Found 297 device(s), from 29 vendor(s).
Completed in 5ms.

2018-05-29 12:48:13
Collecting packs...
Found 1784 package version(s), from 37 vendor(s).
Completed in 18ms.

The problem now is that when i get to the project->properties->settings tab I don't get the option to assign a device to the project. It just doesn't come up. I can only see this:

[image]

Any more help appreciated.

ilg-ul, 2018-05-29:
> I don't get the option to assign a device to the project.

the project is not a GME project.

sebastian-new, 2018-05-29:
How do I start a new GME project from existing makefile code? Or generally how do i start a GME project?

sebastian-new, 2018-05-29:
Hello,

OK finally, using the example for Generic GNU MCU project (https://gnu-mcu-eclipse.github.io/templates/cortexm/)

I managed to create, build and debug a project and now the peripherals are indeed visible with no problem. 

The question now is -> Nordic provides all its examples with makefiles. Is it possible to create a GME project from existing makefile code?

TM1234, 2018-05-29:
> Is it possible to create a GME project from existing makefile code?

I don't think so - Makefile projects and (GNU or bog standard CDT) Managed Make projects are completely different types of project.

And isn't your question (ultimately about having the target device associated with a Makefile project) the same as the question asked here - to which the answer seems to be "no"?

https://github.com/gnu-mcu-eclipse/eclipse-plugins/issues/284

Hope this helps.

ilg-ul, 2018-05-29:
That's correct, you cannot create a functionally equivalent GME managed proiect from a makefile code, because you cannot easily extract the list of source folders, include folders and compile options by parsing the make files; you need to redefine them manually.

sebastian-new, 2018-05-29:
Thank you both very much for your help.