eclipse-embed-cdt/eclipse-plugins #308: debugging pb

zawbaz, 2018-07-04:
I have a problem with debugging and read previous documentation but could not find the solution there. I am using DS5 IDE the default Arm compiler 5 (DS-5 Built in ) on windows 10 . DS5 is based on Neon.3 (4.6.3). My CPU is Arm Cortex M4F apollo2 . I installed the GNU MCU Eclipse Plugin from this web http://gnu-mcu-eclipse.netlify.com/v4-neon-updates in DS5 through Eclipse Marketplace and  I am using the Jlink GDBServer.  I am using the gdb client from Arm Gnu Toolchain 7 2018-q2-update. Java 10.0.1 is installed in my pc and it is also added in environment variable path.  The main.c file is the CMSIS template of RTOS2.Keil RTX5. When I hit the run buttton . This is the result I get it and keep executing this forever.

SEGGER J-Link GDB Server V6.32g Command Line Version

JLinkARM.dll V6.32g (DLL compiled Jun 15 2018 17:26:15)

Command line: -if swd -device AMAPH1KK-KBR -endian little -speed 1000 -port 2331 -swoport 2332 -telnetport 2333 -vd -ir -localhostonly 1 -singlerun -strict -timeout 0 -nogui
-----GDB Server start settings-----
GDBInit file:                  none
GDB Server Listening port:     2331
SWO raw output listening port: 2332
Terminal I/O port:             2333
Accept remote connection:      localhost only
Generate logfile:              off
Verify download:               on
Init regs on start:            on
Silent mode:                   off
Single run mode:               on
Target connection timeout:     0 ms
------J-Link related settings------
J-Link Host interface:         USB
J-Link script:                 none
J-Link settings file:          none
------Target related settings------
Target device:                 AMAPH1KK-KBR
Target interface:              SWD
Target interface speed:        1000kHz
Target endian:                 little

Connecting to J-Link...
J-Link is connected.
Firmware: J-Link OB-SAM3U128 V3 compiled Sep 21 2017 14:14:50
Hardware: V3.00
S/N: 483028403
Checking target voltage...
Target voltage: 3.30 V
Listening on TCP/IP port 2331
Connecting to target...Connected to target
Waiting for GDB connection...Connected to 127.0.0.1
Reading all registers
Read 4 bytes @ address 0x00000000 (Data = 0x10001F28)
Read 2 bytes @ address 0x00000000 (Data = 0x1F28)
Received monitor command: speed 1000
Target interface speed set to 1000 kHz
Received monitor command: speed 1000
Target interface speed set to 1000 kHz
Received monitor command: flash breakpoints 0
Flash breakpoints disabled
Downloading 16240 bytes @ address 0x00000000 - Verified OK
Downloading 4720 bytes @ address 0x00003F70 - Verified OK
Downloading 60 bytes @ address 0x10000000Comparing flash   [....................] Done.
Verifying flash   [....................] Done.
 - Verified OK
Writing register (PC = 0x00000100)
Read 4 bytes @ address 0x00000100 (Data = 0xF802F000)
Received monitor command: clrbp
Received monitor command: reset
Resetting target
Received monitor command: halt
Halting target CPU...
...Target halted (PC = 0x000000C0)
Received monitor command: regs
R0 = 00000000, R1 = 00000000, R2 = 00000000, R3 = 00000000
R4 = 00000000, R5 = 00000000, R6 = 00000000, R7 = 00000000
R8 = 00000000, R9 = 00000000, R10= 00000000, R11= 00000000
R12= 00000000, R13= 10001F28, MSP= 10001F28, PSP= 00000000
R14(LR) = FFFFFFFF, R15(PC) = 000000C0
XPSR 01000000, APSR 00000000, EPSR 01000000, IPSR 00000000
CFBP 00000000, CONTROL 00, FAULTMASK 00, BASEPRI 00, PRIMASK 00
Reading all registers
Starting target CPU...
...Breakpoint reached @ address 0x00004E52
Reading all registers
Starting target CPU...
...Breakpoint reached @ address 0x00004E52
Reading all registers
Starting target CPU...
...Breakpoint reached @ address 0x00004E52
Reading all registers
Starting target CPU...
...Breakpoint reached @ address 0x00004E52
Reading all registers
Starting target CPU...
...Breakpoint reached @ address 0x00004E52
Reading all registers
Starting target CPU...
...Breakpoint reached @ address 0x00004E52
Reading all registers
Starting target CPU...
...Breakpoint reached @ address 0x00004E52

This keeps on repeating. In the error window it is always showing this 
[image]

When I click the debug button ,  it always goes in fault_isr before jumping to main. Sometimes I  see this No source available for (gdb[59].proc[42000].threadGroup[i1],gdb[59].proc[42000].OSthread[1]).thread[1].frame[0]"
or No source available for "0x0"
 Any hep would be highly appreciated. Is there any reference manual to set all the configuration correctly  If yes Please share it too . 

Regards, 
Faisal 

 

### Prerequisites

In case you encounter problems while using the plug-ins, please consider the following:

* [ ] read the documentation pages carefully
* [ ] check the [Known issues](http://gnu-mcu-eclipse.github.io/support/known-issues/)
* [ ] check the [FAQ](http://gnu-mcu-eclipse.github.io/support/faq/)
* [ ] check the [GNU MCU Eclipse forum](http://gnu-mcu-eclipse.github.io/support/forum/) for similar problems
* [ ] check the GitHub Issues [tracker](https://github.com/gnu-mcu-eclipse/plug-ins/issues/)

If you still could not find a solution, if you have interesting use cases, if you have custom 
configurations, and generally if you have any experience that you want to share with others, 
use the [GNU MCU Eclipse forum](http://gnu-mcu-eclipse.github.io/support/forum/).

If you are convinced that you identified a bug (if you have doubts, use the forum), or you have a 
pertinent suggestion how to enhance the GNU MCU Eclipse plug-ins, continue and register
an issue.

### Description

[Description of the bug or feature]

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behaviour:** [What you expected to happen]

**Actual behaviour:** [What actually happened]

### Versions

* [plug-in version]
* [Eclipse version]
* [Java version]
* [operating system]
* [toolchain version]

Please understand that without being able to reproduce the bug we cannot identify your problem.

---

Note: Remove unnecessary text after reading, before entering your ticket.

ilg-ul, 2018-07-04:
For support please use the project forum.

TM1234, 2018-07-04:
Why do you think that this is a GNU MCU Eclipse problem?
I can't see anything in what you've posted that suggests this rather than some other problem with your setup or program?

zawbaz, 2018-07-04:
Hi TM1234,
               Because with the same CMSIS main.c template of RTX5 RTOS2, I am having no issue in Keil uVision. So I do not know , may be I am wrong . Here in DS-5 I am using  GNU MCU Eclipse Plugin for j-link debugging and I am using here  Gnu Toolchain 7 2018-q2-update . So may be there is compatibility problem between GDB server and GDB client . My project is running fine in Keil uVision with its default settings. 
Dear ilg-ul , could you provide me the link of the project forum . Many Thanks

closed 2018-07-04 by ilg-ul

TM1234, 2018-07-04:
> Because with the same CMSIS main.c template of RTX5 RTOS2, I am having no issue in Keil uVision

That was not made clear in your original post but is still doesn't necessarily indicate a GME plugin problem. It could be how your are compiling your program, your jlink configuration etc.

Anyway, as Liviu says, the forum is the correct place to discuss this...

> could you provide me the link of the project forum

https://gnu-mcu-eclipse.github.io/support/forum/