eclipse-embed-cdt/eclipse-plugins #312: Disabling "Allocate console for GDB server" for J-Link debugger breaks debugging

jacek-s, 2018-07-25:
As in the title - when de-selecting "Allocate console for GDB server" in the debugger options, it stops debuggning from functioning properly.  When starting the debugger, it stops at "Launching: Configuring GDB" and doesn't continue further.

My intention was only disabling the unnecessary console.

ilg-ul, 2018-07-25:
Oops!

This is tricky, at a certain point the logic to tell if the server is started was changed to hunt for a string in the console output. It seems that disabling the console prevents this logic to detect when the server is ready. :-(

The solution for the plug-in is to continue processing the console as normal, but do not display it.

I don't know now, but the changes might not be trivial.

For now I suggest to avoid deselecting this option. I might even disable it until a fix will be available.

jacek-s, 2018-07-26:
I'll also add to those who use the "Show Console When Standard Out Changes" option and want to see the SVW console output but are annoyed with it changing to JLinkGDBServerCL.exe console every time you step in the program or read/modify variables or memory - the workaround is to use "Pin Console" on the SWV console. It'll stay there until you terminate the session.

This has to be done every time a new debugging session starts, as each session creates a new set of consoles.

ilg-ul, 2019-09-23:
Fixed in 4.6.1.

closed 2019-09-23 by ilg-ul