eclipse-embed-cdt/eclipse-plugins #313: Stuck for short time when try to open project settings

bureaua, 2018-08-04:
### Description

Stuck for short time when try to open project settings

### Steps to Reproduce

1. Open Eclipse with installed gnu-mcu-eclipse plugins
2. Import project
3. Open project setting (C/C++ Build -> Settings)

**Actual behaviour:** [What actually happened]
When I am try to to open  project settings -- Eclipse shuck for some time (~5-30 sec)

### Versions

* [plug-in version]  2.2.2.201807211108 (but same issue with older versions)
* [Eclipse version] Oxygen, Photon (does not matter)
* [Java version]  1.8.0_181
* [operating system] Ubuntu 16.04 x64, Windows 10 x64 (does not matter)
* [toolchain version] gcc-arm-none-eabi-7-2017-q4-major

--------------------
If description does not clear, I can provide more information, just tell me which information I should provide.
May be current issue is connected with other Eclipse modules communication, but with which ?

ilg-ul, 2018-08-05:
Did you try: https://gnu-mcu-eclipse.github.io/plugins/install/#ubuntu-gtk-issue ?

bureaua, 2018-08-05:
With all projects which is use gnu-mcu-eclipse plugin

ilg-ul, 2018-08-05:
I cannot reproduce it. What does 'short time' mean?

bureaua, 2018-08-05:
Eclipse stuck for some time (~5-30 sec)

bureaua, 2018-08-05:
See link
https://fex.net/199516849935?fileId=640704147
At the time when mouse cursor disappeared -- I did press on Settings

ilg-ul, 2018-08-05:
Public access to that link is not possible.

I just tested again, with the latest release from https://github.com/gnu-mcu-eclipse/org.eclipse.epp.packages/releases and with a project created with the STM32F4 template; the time to enter the Settings page is a fraction of a second.

Could you try exactly the same steps?

bureaua, 2018-08-05:
Same issue, like for the new just created project from STM32F4 template and also for old imported project

ilg-ul, 2018-08-05:
If you close the settings windows and open it again, does it stumble again, or is it faster?

ilg-ul, 2018-08-05:
You reported that the problem occur on Windows too. Is this accurate?

bureaua, 2018-08-05:
On windows it takes 2-3 sec. 
The same case with  the latest release from https://github.com/gnu-mcu-eclipse/org.eclipse.epp.packages/releases 

If I will open Properties window -> next, go to the Settings -- in this case first run on Windows takes  2-3 sec. 
If I will go to another subwindow (for example Logging, without closing Properties window) and after go to the Settings again -- second trying to open Settings window is very fast

ilg-ul, 2018-08-05:
> first run on Windows takes 2-3 sec. ... second trying to open Settings window is very fast

then the problem is Ubuntu specific, and most probably is the well known GTK issue.

I'm afraid there is not much the GME plug-ins can do to fix it.

bureaua, 2018-08-05:
OK, understood.
Thanks for your time and help!

For the future newcomers:
Solution for current thread could be found on this link:
[https://gnu-mcu-eclipse.github.io/plugins/install/#ubuntu-gtk-issue](https://gnu-mcu-eclipse.github.io/plugins/install/#ubuntu-gtk-issue)

closed 2018-08-05 by bureaua

bureaua, 2018-08-05:
May be it would be good to place this information on some web page
http://www.eclipse.org/swt/faq.php#gtkstartup

And also information that
`--launcher.GTK_version 
2
`
sould be placed before 
`--launcher.appendVmargs`

TM1234, 2018-08-05:
@bureaua, are you saying that --launcher.GTK_version 2 DID work without the earlier error once you put it in the correct location in the eclipse.ini file?

bureaua, 2018-08-05:
@TM1234 
Yes

ilg-ul, 2018-08-06:
Tommy, if you confirm this, we probably have to update the web page.

TM1234, 2018-08-06:
Confirm what?
The correct location of the --launcher.GTK_version 2 setting in the eclipse.ini file?
I could not find an authoritative source for this but several sites say that it must go before --launcher.appendVmargs setting.

ilg-ul, 2018-08-06:
> Confirm what?

that `GTK_version` must be placed before `appendVmargs`.

TM1234, 2018-08-06:
That requirement seems to be anecdotal but does work... The official eclipse documentation doesn't seem to say anything about the correct placement in the eclipse.ini file of the GTK_version setting.

ilg-ul, 2018-08-06:
ok, I mentioned it in the documentation.

TM1234, 2018-08-06:
Thanks Liviu.
I searched further but still can't find any authoritative info about the correct placement of the --launcher.GTK_version 2 setting (and I think that it also needs to be split over two lines?):

```
--launcher.GTK_version
2
```

Some people say "before -vmargs", some say "before --appendVmargs" some say at the start of the file, some don't say anything about the placement being significant even though, in practice, it obviously is....