eclipse-embed-cdt/eclipse-plugins #315: resolvemacros messages always output to log

JohnKBartholomew, 2018-08-14:
### Description

When I set a value in the my project's ARM Toolchain Paths toolchain folder, like ${my_var}/tools/gcc-7.2.1-1.0/bin, I end up with lots of unneeded messages of the following form output in to my Eclipse log:
 
!ENTRY ilg.gnumcueclipse.managedbuild.cross 4 1 2017-09-30 12:29:31.413
!MESSAGE arm.PathEnvironmentVariable.resolveMacros("${my_var}\tools\gcc-5.4.1-1.0\bin", "Debug") = ""
${my_var} in this case is a build variable my own plugin defines.

### Steps to Reproduce

1. Create an Eclipse CDT project
2. Go the project's properties, under MCU -> ARM Toolchain Paths
3. Enter a value in the Toolchain Folder field that uses a valid ${<whatever>} expression in it.
4. Build the project

**Expected behaviour:** [What you expected to happen]

No messages produced to Eclipse error log.

**Actual behaviour:** [What actually happened]

Lots of messages produced to Eclipse error log.

The code in:
ilg.gnumcueclipse.managedbuild.cross.arm\src\ilg\gnumcueclipse\managedbuild\cross\arm\EnvironmentVariableSupplier.java
in the method resolveMacros() always adds this message to the log file with this line:
	Activator.log("arm.PathEnvironmentVariable.resolveMacros(\"" + str + "\", \"" + configuration.getName() + "\") = \"" + "\"");
It's not checked with Activator.getInstance().isDebugging() first as are most other places in this class. 

### Versions

* [plug-in version] ilg.gnumcueclipse.managedbuild.cross v1.1.2.201707111115
* [Eclipse version] Oxygen 1a
* [Java version] 1,8
* [operating system] Windows 7 x64
* [toolchain version] GCC 7.2.1

ilg-ul, 2018-08-15:
Please install from http://gnu-mcu-eclipse.netlify.com/v4-neon-updates-experimental and let me know if it is ok.

JohnKBartholomew, 2018-08-15:
Thanks Liviu – your update has removed the extra messages appearing in the Eclipse error log.
Can you tell me in which release of GNU ARM Eclipse this fix will be included?

ilg-ul, 2018-08-15:
The next release will be 4.4.2 or 4.5.1, depending on the changes. It is scheduled after the next Eclipse release.

What are your plug-ins doing?

JohnKBartholomew, 2018-08-15:
Our code is programmatically setting the ARM Toolchain Paths toolchain folder value to a location that includes a build variable: e.g., "${our_install_dir}/gcc-7.2.1/bin". In the managedbuild.cross plugin code (see the Description above), any value containing a "${" will cause resolveMacros() to be called, which was always printing the observed message.

If you set the toolchain folder value to an absolute path on your machine (i.e., no ${build_var} values), then resolveMacros() is not called, and no log messages are generated.

closed 2018-08-15 by JohnKBartholomew

ilg-ul, 2018-08-15:
Do you have a custom toolchain? Why do you need a custom path management?

JohnKBartholomew, 2018-08-15:
Yes, it's delivered with our Eclipse RCP application, to which we've added our own plugins and a number of scripts, examples, etc. In this way, when a team of developers is working with our installation, they will all be pointing by default to the same GCC included with our release. (Building firmware for some of our Cypress hardware devices requires a particular GCC release, for example.) It decreases the support burden for us by pre-setting the ARM installation this way.

ilg-ul, 2018-08-15:
Did you check the new method to install toolchains via xpm?

JohnKBartholomew, 2018-08-15:
No - can you add a link to information on this thread?