eclipse-embed-cdt/eclipse-plugins #318: Again, can we use QEMU and OpenOCD from their own official websites?

jiapei100, 2018-08-26:
I mean, can we just use QEMU from [https://www.qemu.org/](https://www.qemu.org/), particularly, 
[https://download.qemu.org/qemu-3.0.0.tar.xz](https://download.qemu.org/qemu-3.0.0.tar.xz) .

And, can we just use OpenOCD from repository of Ubuntu 18.04.1 ? Or OpenOCD 0.10.0 downloaded from [http://openocd.org/](http://openocd.org/) ?

Cheers
Pei

TM1234, 2018-08-26:
No. There's no RISC-V support in the released version of 0.10.0. And only bits of the support have been upstream end to the current OpenOCD github development source base (confusingly still called 0.10.0 rather than 0.11.0-dev or something like that as in the past).

You have to build OpenOCD from the RISC-V github or using Liviu's docker based scripts to get RISC-V support. The former is up to date with the latest RISC-V and most upstream OpenOCD changes. The latter is not and is slightly behind the curve.

ilg-ul, 2018-08-26:
If you prefer to build your own binaries, sure, you can use any version you like.

Please check if the official QEMU supports your device.

closed 2018-08-26 by ilg-ul