eclipse-embed-cdt/eclipse-plugins #320: Is it possible to have multi-line block statements in the launcher?

AntonKrug, 2018-09-18:
### Description

The GME launchers allow adding few commands to the initialization and run/restart. But it seems that they do not allow the multi-line block statements. Single line statements get executed without issue, but a **if** condition or **while** loop which are in multiple lines and need the whole block to function correctly do not work from the launcher.

In the end more complex behaviour can be added to the gdbinit as a defined function and only called from the launcher when needed. And that works well, but I wonder if this is by design or it's a symptom that the commands are sent one by one to the gdb?

### Steps to Reproduce

1. Enable GDB traces: Windows -> Preferences -> C/C++ -> Debug -> GDB -> Show the GDB traces consoles with character limit (default value of characters should be enough)
1. Use regular launcher of a working project (in my case RISCV, but I think this is not target specific).
2. Go to launcher settings -> Startup tab
3. In the "Initilization Commands" type the following:

```
if (1)
  print "if condition true"
end
```

3.  Click debug and open the GDB traces console

**Expected behaviour:** 
Get the condition to be executed correctly:

```
654,867 ~"$1 = \"if condition true\""
```

**Actual behaviour:** 
The launcher gets stuck the moment it reaches the block statement:

```
750,171 &"if (1)\n"
750,171 ~" >"
750,172 ~" >"
750,172 ~" >"
750,177 ~" >"
750,181 ~" >"
750,182 ~" >"
```

### Versions

* GME 4.4.1-201807211108
* Eclipse Photon 4.8
* Java 10

ilg-ul, 2018-09-18:
How do you suggest to implememt such a feature?

AntonKrug, 2018-09-18:
No proper idea to be honest. But is my suspicion right, or did I do something stupid on my part? I was trying to fool it and send multiple statements as one line. Tried to escaped new lines but they got escaped again.

Not sure if this is worth fixing. One idea I had (if it's even possible) is to detect the block statements and send them in one command?

Even if this won't be fixed I wanted to at least report it, so it's known limitation/issue.

ilg-ul, 2018-09-18:
You need to start a shell and pass it the commands.

AntonKrug, 2018-09-18:
So probably the most convenient way of doing it is to add whatever is needed into the gdbinit (or other gdbscripts) as a define-function and then just call the function form the launcher?

ilg-ul, 2018-09-18:
Or start `sh -c 'xxx'` or similar.

AntonKrug, 2018-09-18:
I'm not sure if I follow, shell scripting from within the gdbscript? These loops and conditions need to be part of the of the gdb as they can interact with the target memory/variables etc...

Are we talking about the same Startup tab options?
[image]

Just noticed that my title might sound demanding, it was not a request to add the functionality, more just question if

ilg-ul, 2018-09-18:
> shell scripting from within the gdbscript?

ah, those commands... 

I have no idea how o address this.
 
feel free to experiment and suggest a solution.