eclipse-embed-cdt/eclipse-plugins #324: Print removed sections (-Xlinker --print-gc-sections) reports as build failure

AntonKrug, 2018-11-05:
### Description
This issue is not an issue of GME plugins, but the underlining CDT (even with regular native gcc tools). Regardless this affects the GME as well so for completeness I reported it here as well, but there is no point trying to solve it here. So probably this should be marked as won't fix? If anybody is affected by this and want to contribute to the discussion, the Eclipse/CDT bug the link is here:
https://bugs.eclipse.org/bugs/show_bug.cgi?id=539927

In simple terms CDT 9.5.x will mark build failed on any **stderr** message regardless if the build works or not, while **--print-gc-sections** by design always put the removed sections verbose messages into **stderr***:

```
--print-gc-sections
--no-print-gc-sections
List all sections removed by garbage collection. The listing is printed on stderr. This option is only effective if garbage collection has been enabled via the ‘--gc-sections’) option. The default behaviour (of not listing the sections that are removed) can be restored by specifying ‘--no-print-gc-sections’ on the command line.
```

This issue was not experienced in Oxygen/CDT 9.4.3 while it is experienced in Photon/CDT (tested 9.5.0, 9.5.2 and 9.5.3) regardless of the host OS and target platform.

### Steps to Reproduce

1. Use GME plugins with CDT 9.5.x which comes with Eclipse Photon
2. Use a project which builds correctly under normal conditions.
2. Clean the project (if the project is built already it will not trigger the error message)
3. Enable the "Print removed sections":
3a. For RISC-V projects: **Project Properties -> C/C++ Build -> Settings -> Tool Settings -> GNU RISC-V Cross C Linker -> General -> Print removed sections (-Xlinker  --print-gc-sections)**
3b. For ARM based projects: **Project Properties -> C/C++ Build -> Settings -> Tool Settings -> Cross ARM GNU C Linker -> General -> Print removed sections (-Xlinker  --print-gc-sections)**

**Expected behaviour:** 

```
12:22:50 Build Finished. 0 errors, 0 warnings. (took 6s.935ms)
```

**Actual behaviour:** [What actually happened]

```
riscv64-unknown-elf/bin/ld.exe: Removing unused section '.text._kill' in file './riscv_hal/syscall.o'
riscv64-unknown-elf/bin/ld.exe: Removing unused section '.text._link' in file './riscv_hal/syscall.o'
riscv64-unknown-elf/bin/ld.exe: Removing unused section '.text._lseek' in file './riscv_hal/syscall.o'
riscv64-unknown-elf/bin/ld.exe: Removing unused section '.text._read' in file './riscv_hal/syscall.o'
riscv64-unknown-elf/bin/ld.exe: Removing unused section '.text._stat' in file './riscv_hal/syscall.o'
riscv64-unknown-elf/bin/ld.exe: Removing unused section '.text._times' in file './riscv_hal/syscall.o'
riscv64-unknown-elf/bin/ld.exe: Removing unused section '.text._unlink' in file './riscv_hal/syscall.o'

(I trimmed the log for sake of brevity)

12:23:21 Build Failed. 182 errors, 0 warnings. (took 2s.482ms)
```

The binaries will be present and if they worked before they still work, only Eclipse thinks the build failed.

### Versions

* Not dependant on GME plugin version
* Eclipse Photon and CDT 9.5.x
* Observable on multiple Java versions
* Can be replicated on Linux, Mac and Windows
* Independent on toolchain version or target as the linker was always reporting removed sections to stderr

ilg-ul, 2019-01-01:
Is this problem still present in 2018-12?

I'm not sure what is the best way to approach this.

RufusVS, 2023-02-11:
Problem still exists in 2022.

TommyMurphyTM1234, 2023-02-11:
It's claimed that it was fixed in the upstream CDT:

* https://bugs.eclipse.org/bugs/show_bug.cgi?id=539927

Maybe the fix was inadequate or there has been a regression?
Perhaps @jonahgraham might know more?

jonahgraham, 2023-02-11:
It appears to be fixed, at least the original issue. Perhaps there has been a spelling change or similar in gcc output. 

Please confirm version you are using and include the exact output from gcc that is being misidentified.

TommyMurphyTM1234, 2023-02-11:
I cannot reproduce this using the following:

1. Eclipse IDE for Embedded C/C++ Developers (includes Incubating components) Version: 2022-12 (4.26.0) Build id: 20221201-1913
2. xPack RISC-V GCC toolchain riscv-none-elf-gcc (xPack GNU RISC-V Embedded GCC x86_64) 12.2.0
Results:
```
19:40:22 **** Build of configuration Debug for project riscv-test ****
make all 
Building file: ../xpacks/sifive-hifive1-board/src/board-functions.cpp
Invoking: GNU RISC-V Cross C++ Compiler
riscv-none-elf-g++ -march=rv32imac_zicsr -mabi=ilp32 -mcmodel=medany -msmall-data-limit=8 -mno-save-restore -O0 -fmessage-length=0 -fsigned-char -ffunction-sections -fdata-sections -fno-move-loop-invariants -Wall -Wextra -g3 -DDEBUG -DTRACE -DOS_USE_TRACE_SEMIHOSTING_DEBUG -DSIFIVE_FE310 -DSIFIVE_HIFIVE1_BOARD -I"../include" -I"../xpacks/micro-os-plus-diag-trace/include" -I"../xpacks/micro-os-plus-semihosting/include" -I"../xpacks/sifive-devices/include" -I"../xpacks/sifive-hifive1-board/include" -I"../xpacks/micro-os-plus-c-libs/include" -I"../xpacks/micro-os-plus-riscv-arch/include" -I"../xpacks/micro-os-plus-startup/include" -std=gnu++14 -fabi-version=0 -fno-exceptions -fno-rtti -fno-use-cxa-atexit -fno-threadsafe-statics -MMD -MP -MF"xpacks/sifive-hifive1-board/src/board-functions.d" -MT"xpacks/sifive-hifive1-board/src/board-functions.o" -c -o "xpacks/sifive-hifive1-board/src/board-functions.o" "../xpacks/sifive-hifive1-board/src/board-functions.cpp"
Finished building: ../xpacks/sifive-hifive1-board/src/board-functions.cpp
 
Building file: ../xpacks/sifive-hifive1-board/src/trace-uart.cpp
Invoking: GNU RISC-V Cross C++ Compiler
riscv-none-elf-g++ -march=rv32imac_zicsr -mabi=ilp32 -mcmodel=medany -msmall-data-limit=8 -mno-save-restore -O0 -fmessage-length=0 -fsigned-char -ffunction-sections -fdata-sections -fno-move-loop-invariants -Wall -Wextra -g3 -DDEBUG -DTRACE -DOS_USE_TRACE_SEMIHOSTING_DEBUG -DSIFIVE_FE310 -DSIFIVE_HIFIVE1_BOARD -I"../include" -I"../xpacks/micro-os-plus-diag-trace/include" -I"../xpacks/micro-os-plus-semihosting/include" -I"../xpacks/sifive-devices/include" -I"../xpacks/sifive-hifive1-board/include" -I"../xpacks/micro-os-plus-c-libs/include" -I"../xpacks/micro-os-plus-riscv-arch/include" -I"../xpacks/micro-os-plus-startup/include" -std=gnu++14 -fabi-version=0 -fno-exceptions -fno-rtti -fno-use-cxa-atexit -fno-threadsafe-statics -MMD -MP -MF"xpacks/sifive-hifive1-board/src/trace-uart.d" -MT"xpacks/sifive-hifive1-board/src/trace-uart.o" -c -o "xpacks/sifive-hifive1-board/src/trace-uart.o" "../xpacks/sifive-hifive1-board/src/trace-uart.cpp"
Finished building: ../xpacks/sifive-hifive1-board/src/trace-uart.cpp
 
Building file: ../xpacks/sifive-devices/src/fe310/device-interrupts.cpp
Invoking: GNU RISC-V Cross C++ Compiler
riscv-none-elf-g++ -march=rv32imac_zicsr -mabi=ilp32 -mcmodel=medany -msmall-data-limit=8 -mno-save-restore -O0 -fmessage-length=0 -fsigned-char -ffunction-sections -fdata-sections -fno-move-loop-invariants -Wall -Wextra -g3 -DDEBUG -DTRACE -DOS_USE_TRACE_SEMIHOSTING_DEBUG -DSIFIVE_FE310 -DSIFIVE_HIFIVE1_BOARD -I"../include" -I"../xpacks/micro-os-plus-diag-trace/include" -I"../xpacks/micro-os-plus-semihosting/include" -I"../xpacks/sifive-devices/include" -I"../xpacks/sifive-hifive1-board/include" -I"../xpacks/micro-os-plus-c-libs/include" -I"../xpacks/micro-os-plus-riscv-arch/include" -I"../xpacks/micro-os-plus-startup/include" -std=gnu++14 -fabi-version=0 -fno-exceptions -fno-rtti -fno-use-cxa-atexit -fno-threadsafe-statics -MMD -MP -MF"xpacks/sifive-devices/src/fe310/device-interrupts.d" -MT"xpacks/sifive-devices/src/fe310/device-interrupts.o" -c -o "xpacks/sifive-devices/src/fe310/device-interrupts.o" "../xpacks/sifive-devices/src/fe310/device-interrupts.cpp"
Finished building: ../xpacks/sifive-devices/src/fe310/device-interrupts.cpp
 
Building file: ../xpacks/sifive-devices/src/device-functions.cpp
Invoking: GNU RISC-V Cross C++ Compiler
riscv-none-elf-g++ -march=rv32imac_zicsr -mabi=ilp32 -mcmodel=medany -msmall-data-limit=8 -mno-save-restore -O0 -fmessage-length=0 -fsigned-char -ffunction-sections -fdata-sections -fno-move-loop-invariants -Wall -Wextra -g3 -DDEBUG -DTRACE -DOS_USE_TRACE_SEMIHOSTING_DEBUG -DSIFIVE_FE310 -DSIFIVE_HIFIVE1_BOARD -I"../include" -I"../xpacks/micro-os-plus-diag-trace/include" -I"../xpacks/micro-os-plus-semihosting/include" -I"../xpacks/sifive-devices/include" -I"../xpacks/sifive-hifive1-board/include" -I"../xpacks/micro-os-plus-c-libs/include" -I"../xpacks/micro-os-plus-riscv-arch/include" -I"../xpacks/micro-os-plus-startup/include" -std=gnu++14 -fabi-version=0 -fno-exceptions -fno-rtti -fno-use-cxa-atexit -fno-threadsafe-statics -MMD -MP -MF"xpacks/sifive-devices/src/device-functions.d" -MT"xpacks/sifive-devices/src/device-functions.o" -c -o "xpacks/sifive-devices/src/device-functions.o" "../xpacks/sifive-devices/src/device-functions.cpp"
Finished building: ../xpacks/sifive-devices/src/device-functions.cpp
 
Building file: ../xpacks/sifive-devices/src/plic-functions.cpp
Invoking: GNU RISC-V Cross C++ Compiler
riscv-none-elf-g++ -march=rv32imac_zicsr -mabi=ilp32 -mcmodel=medany -msmall-data-limit=8 -mno-save-restore -O0 -fmessage-length=0 -fsigned-char -ffunction-sections -fdata-sections -fno-move-loop-invariants -Wall -Wextra -g3 -DDEBUG -DTRACE -DOS_USE_TRACE_SEMIHOSTING_DEBUG -DSIFIVE_FE310 -DSIFIVE_HIFIVE1_BOARD -I"../include" -I"../xpacks/micro-os-plus-diag-trace/include" -I"../xpacks/micro-os-plus-semihosting/include" -I"../xpacks/sifive-devices/include" -I" [... cut ...] riscv-none-elf/lib/libg_nano.a(lib_a-reent.o)'
/home/user/.local/xPacks/@xpack-dev-tools/riscv-none-elf-gcc/12.2.0-3.1/.content/bin/../lib/gcc/riscv-none-elf/12.2.0/../../../../riscv-none-elf/bin/ld: removing unused section '.text._reclaim_reent' in file '/home/user/.local/xPacks/@xpack-dev-tools/riscv-none-elf-gcc/12.2.0-3.1/.content/bin/../lib/gcc/riscv-none-elf/12.2.0/../../../../riscv-none-elf/lib/libg_nano.a(lib_a-reent.o)'

19:40:26 Build Finished. 0 errors, 0 warnings. (took 4s.395ms)
```

ilg-ul, 2023-02-11:
Thank you @TommyMurphyTM1234 for taking the time to test.

@RufusVS, could you also check with the latest toolchains?

RufusVS, 2023-02-12:
I just ran with the following configuration for My Renesas e2studio:

Board:   Custom User Board (Any Device) 
Device:  R7FS5D97E2A01CLK 
Toolchain:   GCC ARM Embedded 
Toolchain Version:   10.3.1.20210824 
SSP Version:   2.2.0 

All components seem to have 2021 or 2022 build dates, which should be up-to-date yet exhibited the same problem.

ilg-ul, 2023-02-12:
> Toolchain Version: 10.3.1.20210824

Please try with 12.x, maybe the problem was in the toolchain.

TommyMurphyTM1234, 2023-02-12:
> I just ran with the following configuration for My Renesas e2studio:

What version?
What version of Eclipse, Eclipse CDT, and Eclipse Embedded CDT does is use?

I tried my previous setup with the xPack arm toolchain v10.3.1-2.1.1 instead of the latest and I still did not get the problem.
```
/home/user/.local/xPacks/@xpack-dev-tools/arm-none-eabi-gcc/10.3.1-2.1.1/.content/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: removing unused section '.bss.__lock___atexit_recursive_mutex' in file '/home/user/.local/xPacks/@xpack-dev-tools/arm-none-eabi-gcc/10.3.1-2.1.1/.content/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/lib/thumb/v6-m/nofp/libg_nano.a(lib_a-lock.o)'
/home/user/.local/xPacks/@xpack-dev-tools/arm-none-eabi-gcc/10.3.1-2.1.1/.content/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: removing unused section '.bss.__lock___dd_hash_mutex' in file '/home/user/.local/xPacks/@xpack-dev-tools/arm-none-eabi-gcc/10.3.1-2.1.1/.content/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/lib/thumb/v6-m/nofp/libg_nano.a(lib_a-lock.o)'
/home/user/.local/xPacks/@xpack-dev-tools/arm-none-eabi-gcc/10.3.1-2.1.1/.content/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: removing unused section '.bss.__lock___env_recursive_mutex' in file '/home/user/.local/xPacks/@xpack-dev-tools/arm-none-eabi-gcc/10.3.1-2.1.1/.content/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/lib/thumb/v6-m/nofp/libg_nano.a(lib_a-lock.o)'
/home/user/.local/xPacks/@xpack-dev-tools/arm-none-eabi-gcc/10.3.1-2.1.1/.content/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: removing unused section '.bss.__lock___sfp_recursive_mutex' in file '/home/user/.local/xPacks/@xpack-dev-tools/arm-none-eabi-gcc/10.3.1-2.1.1/.content/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/lib/thumb/v6-m/nofp/libg_nano.a(lib_a-lock.o)'
/home/user/.local/xPacks/@xpack-dev-tools/arm-none-eabi-gcc/10.3.1-2.1.1/.content/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: removing unused section '.bss.__lock___sinit_recursive_mutex' in file '/home/user/.local/xPacks/@xpack-dev-tools/arm-none-eabi-gcc/10.3.1-2.1.1/.content/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/lib/thumb/v6-m/nofp/libg_nano.a(lib_a-lock.o)'
/home/user/.local/xPacks/@xpack-dev-tools/arm-none-eabi-gcc/10.3.1-2.1.1/.content/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: removing unused section '.bss.__lock___tz_mutex' in file '/home/user/.local/xPacks/@xpack-dev-tools/arm-none-eabi-gcc/10.3.1-2.1.1/.content/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/lib/thumb/v6-m/nofp/libg_nano.a(lib_a-lock.o)'

14:26:26 Build Finished. 0 errors, 0 warnings. (took 10s.405ms)
```

Yuo need to test with the latest versions of ALL tools.

RufusVS, 2023-02-12:
e2studio didn't like running with a non-standard toolchain.   I finally got the new toolchain to compile the program by selecting "non-installed" toolchain and giving it the path.  Then I had to fix a syntax error on the code's call to _Static_assert.  But then no errors.  So it *was* the toolchain version.
Thanks everyone helping me with that.  
You don't need to keep this issue open on my account.

ilg-ul, 2023-02-12:
Glad to know that it is no longer a problem.

closed 2023-02-12 by ilg-ul

TommyMurphyTM1234, 2023-02-12:
> So it _was_ the toolchain version. Thanks everyone helping me with that. You don't need to keep this issue open on my account.

That's very odd. I didn't get the problem with the older toolchain that you were using when I used it with the latest Eclipse for Embedded C/C++ Developers. But, either way, with the latest xPacks and Eclipse the issue is not present.

AntonKrug, 2023-02-14:
The best for the GME/CDT issues reported on these forums is to use vanilla Eclipse. If a vendor has some quirk that can't be replicated on the vanilla, then probably it's not an issue of the upstream tools.