eclipse-embed-cdt/eclipse-plugins #325: ARM project settings stall and are very unresponsive

AntonKrug, 2018-11-07:
### Description

In some cases open project settings C/C++ Build -> Settings -> Tool Settings on ARM projects can make the whole UI unresponsive for ~30 seconds. This is due to GTK3 handling in Eclipse (and CDT) was not efficient and structures were increasingly slower the more content they had. Where ARM dropdowns can have ~100 entries.

### Steps to Reproduce

1. Use GTK3 on Linux Eclipse Oxygen or Eclipse Photon.
2. Create example ARM project with GME plugin wizard.
3. Open settings with C/C++ Build -> Settings -> Tool Settings

**Expected behaviour:** 
Open within 1-2 seconds.

**Actual behaviour:** 
Opens in ~30 seconds.

### Versions

* Plugin version doesn't effect the problem
* Eclipse Oxygen or Photon
* Java doesn't affect the problem
* Linux

The issue can be worked around on Eclipse 4.7 Oxygen by setting:
```
export SWT_GTK3=0
```
Which will force GTK2. But for Eclipse 4.8 Photon this will create problem as the GTK2 is in process of removal, so even ARM settings might open in timely manner there still might be other problems/glitches. For ARM projects it's best to use either 4.7 Oxygen with GTK2 forced, or use GTK3 on Eclipse 4.9 where the GTK3 slow downs were sorted out. 

In essence the Eclipse 4.8 Photon users should either downgrade to 4.7 or upgrade to 4.9.

The GME developers do not have to do anything, as this is out of their hands, but probably this could be mentioned in the 
https://gnu-mcu-eclipse.github.io/support/known-issues/
or 
https://gnu-mcu-eclipse.github.io/support/faq/

I think there was somewhere the GTK2 trick, but AFAIK there no mention about this being fixed in 4.9 or that the GTK2 trick on 4.8 might create other undesired effects (especially when libcairo is pretty recent version).

ilg-ul, 2018-11-07:
Thank you, Anton, for investigating this.

I added a section in the Known Issues page:

https://gnu-mcu-eclipse.github.io/support/known-issues/#eclipse-48-photon-on-linux-is-unresponsive

AntonKrug, 2018-11-07:
No problem, and actually it feels a bit better when the TODO list is getting smaller and not bigger :)