eclipse-embed-cdt/eclipse-plugins #326: Unable to get debugger console

pccdac, 2018-11-12:
### Prerequisites

In case you encounter problems while using the plug-ins, please consider the following:

* [ ] read the documentation pages carefully
* [ ] check the [Known issues](http://gnu-mcu-eclipse.github.io/support/known-issues/)
* [ ] check the [FAQ](http://gnu-mcu-eclipse.github.io/support/faq/)
* [ ] check the [GNU MCU Eclipse forum](http://gnu-mcu-eclipse.github.io/support/forum/) for similar problems
* [ ] check the GitHub Issues [tracker](https://github.com/gnu-mcu-eclipse/plug-ins/issues/)

If you still could not find a solution, if you have interesting use cases, if you have custom 
configurations, and generally if you have any experience that you want to share with others, 
use the [GNU MCU Eclipse forum](http://gnu-mcu-eclipse.github.io/support/forum/).

If you are convinced that you identified a bug (if you have doubts, use the forum), or you have a 
pertinent suggestion how to enhance the GNU MCU Eclipse plug-ins, continue and register
an issue.

### Description

[Description of the bug or feature]

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behaviour:** [What you expected to happen]

**Actual behaviour:** [What actually happened]

### Versions

* [plug-in version]
* [Eclipse version]
* [Java version]
* [operating system]
* [toolchain version]

Please understand that without being able to reproduce the bug we cannot identify your problem.

---

Note: Remove unnecessary text after reading, before entering your ticket.

pccdac, 2018-11-12:
Description
Unable to get debugger console using latest version of eclipse  (https://github.com/gnu-mcu-eclipse/org.eclipse.epp.packages/releases/download/v4.4.2-20180930-2018-09/20180930-0922-gnumcueclipse-4.4.2-2018-09-linux.gtk.x86_64.tar.gz).

Steps to Reproduce
-> Installed latest eclipse version
-> configured openocd and gdb in debugger configurations.
-> Debugger console is not working using this latest version.
But 
When i installed eclipse oxygen version https://github.com/gnu-mcu-eclipse/org.eclipse.epp.packages/releases/download/v4.4.1-20180721-o3a/20180721-1240-gnumcueclipse-4.4.1-oxygen-3a-linux.gtk.x86_64.tar.gz it is working.
Is there any problem with latest version of eclipse ?

Versions
Eclipse Oxygen
Linux
riscv64-unknown-elf-gcc version 7.2.0

ilg-ul, 2018-11-12:
> So the `${cross_prefix}` macro is not resolved when it's checking the version. 

aha. so you think the new Eclipse changed something in the way macros are processed, and `${cross_prefix}` is no longer resolved?

> Shouldn't the Configuration.getGdbServerCommand used on the macro executable to get the resolved actually executable before running it?

I don't know, I need to check the code. can you elaborate to help me get up to speed?

AntonKrug, 2018-11-12:
I would say resolving it manually before calling Eclipse part should work on old and new Eclipse.

ilg-ul, 2018-11-13:
right, but where do you suggest to add the resolving code?

ilg-ul, 2018-11-13:
I confirm that by replacing the `${cross_prefix}gdb${cross_suffix}` with `arm-none-eabi-gdb`, the debugger console is visible even with 2018-09.

I'll try to investigate, but it might take some time, so any suggestions are welcome.

ilg-ul, 2018-11-13:
FYI: https://bugs.eclipse.org/bugs/show_bug.cgi?id=541069

ilg-ul, 2018-11-19:
CDT 9.5.5 fixed the problem. update your Eclipse and the Debugger Console should be fine.

AntonKrug, 2018-11-19:
Should this be closed as the CDT update fixes it? Or you want to wait for your release to include it?

ilg-ul, 2018-11-19:
My next release will probably be done after 2018-12 is out, so we'll probably leave the current release as it is, with the mention to perform an upgrade.

closed 2018-11-19 by ilg-ul

AntonKrug, 2018-11-19:
The CDT update is painless now so there shouldn't a problem for users to upgrade if they need to. Or resolve the macro by hand if they do not want upgrade CDT.

AntonKrug, 2018-11-21:
Actually there might be similar/related issues to this. The Debug Console view still is named without resolving `${cross_prefix}gdb${cross_suffix}`. And there is on the linux/windows terminal the message: `Unable to find full path for "${cross_prefix}gdb${cross_suffix}"`

ilg-ul, 2018-11-21:
Yes, I know, we'll solve it at a later date.

ilg-ul, 2018-11-21:
Well, they are not cosmetic, the CDT debug plug-ins somehow adapt to the GDB version; without the macro subtitution, they assume a very old version, so new features are not used. This may or may not affect embedded targets, I did not study the details.

AntonKrug, 2018-11-21:
Oh, I was thinking that was happening before that it went to the worst implementation.

ilg-ul, 2018-11-24:
Bug [541069](
https://bugs.eclipse.org/bugs/show_bug.cgi?id=541069) was fixed in CDT 9.5.5, released a few days ago, so the solution is to update your Eclipse to the latest available version.