eclipse-embed-cdt/eclipse-plugins #327: Peripherals view doesn't actually show register contents, despite showing the correct peripherals from SVD

bigbrett, 2018-11-23:
### Description

Peripherals view shows the proper peripherals from the SVD file, but is unable to actually provide any register values. Also, once a peripheral is "checked" in the view, it cannot be unchecked. All the SVD parsing in the debug console looks to be correct and successful. I am using Segger JLink to debug. 

### Steps to Reproduce

1. Start a debug session
2. Click the Peripherals view and select a peripheral to view
3. Watch as nothing happens...
4. Try to "uncheck" a peripheral
5. Watch as it cannot be unchecked 

**Expected behaviour:** 
One can view the status of the peripheral registers

**Actual behaviour:** 
Nothing shows up for the peripheral registers, and the checkbox for a peripheral cannot be unchecked. 

### Versions

* Plugin version: 4.4.2.201809300659

* [Eclipse version]
```
Eclipse IDE for C/C++ Developers
Version: 2018-09 (4.9.0)
Build id: 20180917-1800
```
* [Java version]: Java version: 10.0.2
* [operating system]: OS: Linux (Kubuntu 18.04) v.4.15.0-39-generic, x86_64 / gtk 3.22.30
* [toolchain version]: GNU Tools for Arm Embedded Processors 7-2017-q4-major

bigbrett, 2018-11-23:
Screencaptured a nice GIF demonstrating the issue. Please lmk what else I can do to help debug. 

[image]

AntonKrug, 2018-11-23:
Could it be related to this issue? https://github.com/gnu-mcu-eclipse/eclipse-plugins/issues/326 could you try to resolve the GDB macro into the actual name?

bigbrett, 2018-11-23:
> try to resolve the GDB macro into the actual name
Could you pls explain what you mean by this? I thought the peripherals were just the macros defined in the CMSIS headers (so for me, LPC54606.h)? 

If it adds any additional info, I'm able to see the ethernet peripheral in the "Expressions" view, if I add a watch expressinon of `(ENET_Type*)<address of enet peripheral>`, however the `ENET` macro defined in the CMSIS header doesn't resolve on the fly (see below image)
[image]

bigbrett, 2018-11-23:
Although I am able to see the output from my JLink Server, I just realized that the Debugger Console view says "No Console to display at this time" while I'm debugging. Is this symptomatic of the issue you referenced above?

bigbrett, 2018-11-23:
Ah okay I now understand what you are saying (sorry for the barrage of messages). 

I just changed the gdb command in the debug configuration from `${cross_prefix}gdb${cross_suffix}` to `arm-none-eabi-gdb` and I got the debug console back, but the peripherals view is still broken :(

ilg-ul, 2018-11-24:
I cannot confirm the issue, on my mac the peripheral view window is functional:

[image]

ilg-ul, 2018-11-24:
as reference, I suggest you try the blinky tutorial with the emulated stm32f4discovery board.

ilg-ul, 2018-11-24:
BTW, the solution to https://github.com/gnu-mcu-eclipse/eclipse-plugins/issues/326 is to update your Eclipse to CDT 9.5.5 or later.

bigbrett, 2018-11-24:
I've used GNU ARM Eclipse many times for various STM32 devices, I fully believe it works. This time it isn't working for me for the LPC546xx series, so wanted to raise the issue. 

**@ilg-ul  is there any debug output I can provide that might assist us in narrowing down what is going on?** I created a new project and the same thing happens in a new workspace, so at least it is consistently occurring on my machine.

ilg-ul, 2018-11-24:
> I've used GNU ARM Eclipse many times for various STM32 devices, I fully believe it works

since I've seen quite a lot of bugs, I'd like you to check the stm32f407 device, to rule out a plug-in general bu and restrict the search to the LPC546xx svd case.

> any debug output

first check the workspace/.metadata/.log file for any exceptions

as for debug options, yes, there are quite a lot, but you need to start Eclipse with a `-debug file`, as explained in https://gnu-mcu-eclipse.github.io/developer/build-procedure/; an example of debug options can be seen at https://github.com/gnu-mcu-eclipse/eclipse-plugins/blob/develop/debug.options

bigbrett, 2018-11-24:
@ilg-ul okay I'll do some digging and get back to you. Thanks

bigbrett, 2018-11-24:
Ah, this potentially looks promising (from the .log file)

```
!ENTRY org.eclipse.ui 4 0 2018-11-24 11:09:01.960
!MESSAGE Unhandled event loop exception
!STACK 0
java.lang.NullPointerException
        at ilg.gnumcueclipse.debug.gdbjtag.services.PeripheralMemoryService.getAddressSize(PeripheralMemoryService.java:372)
        at ilg.gnumcueclipse.debug.gdbjtag.memory.PeripheralMemoryBlockExtension.<init>(PeripheralMemoryBlockExtension.java:198)
        at ilg.gnumcueclipse.debug.gdbjtag.memory.PeripheralMemoryBlockRetrieval.getExtendedMemoryBlock(PeripheralMemoryBlockRetrieval.java:131)
        at ilg.gnumcueclipse.debug.gdbjtag.memory.MemoryBlockMonitor.addMemoryBlock(MemoryBlockMonitor.java:155)
        at ilg.gnumcueclipse.debug.gdbjtag.memory.MemoryBlockMonitor.displayPeripheralMonitor(MemoryBlockMonitor.java:94)
        at ilg.gnumcueclipse.debug.gdbjtag.datamodel.PeripheralDMContext.displayPeripheralMonitor(PeripheralDMContext.java:126)
        at ilg.gnumcueclipse.debug.gdbjtag.viewmodel.peripherals.PeripheralsVMModelProxyStrategy.setChecked(PeripheralsVMModelProxyStrategy.java:60)
        at org.eclipse.debug.internal.ui.viewers.model.TreeModelContentProvider.setChecked(TreeModelContentProvider.java:269)
        at org.eclipse.debug.internal.ui.viewers.model.InternalTreeModelViewer.handleSelect(InternalTreeModelViewer.java:1833)
        at org.eclipse.jface.viewers.StructuredViewer$4.widgetSelected(StructuredViewer.java:1254)
        at org.eclipse.jface.util.OpenStrategy.fireSelectionEvent(OpenStrategy.java:243)
        at org.eclipse.jface.util.OpenStrategy.access$4(OpenStrategy.java:238)
        at org.eclipse.jface.util.OpenStrategy$1.handleEvent(OpenStrategy.java:403)
        at org.eclipse.swt.widgets.EventTable.sendEvent(EventTable.java:89)
        at org.eclipse.swt.widgets.Display.sendEvent(Display.java:5797)
        at org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1374)
        at org.eclipse.swt.widgets.Display.runDeferredEvents(Display.java:5051)
        at org.eclipse.swt.widgets.Display.readAndDispatch(Display.java:4583)
        at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine$5.run(PartRenderingEngine.java:1173)
        at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:339)
        at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.run(PartRenderingEngine.java:1062)
        at org.eclipse.e4.ui.internal.workbench.E4Workbench.createAndRunUI(E4Workbench.java:156)
        at org.eclipse.ui.internal.Workbench.lambda$3(Workbench.java:628)
        at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:339)
        at org.eclipse.ui.internal.Workbench.createAndRunWorkbench(Workbench.java:563)
        at org.eclipse.ui.PlatformUI.createAndRunWorkbench(PlatformUI.java:151)
        at org.eclipse.ui.internal.ide.application.IDEApplication.start(IDEApplication.java:155)
        at org.eclipse.equinox.internal.app.EclipseAppHandle.run(EclipseAppHandle.java:199)
        at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.runApplication(EclipseAppLauncher.java:137)
        at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.start(EclipseAppLauncher.java:107)
        at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:391)
        at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:246)
        at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
        at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
        at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
        at java.base/java.lang.reflect.Method.invoke(Method.java:564)
        at org.eclipse.equinox.launcher.Main.invokeFramework(Main.java:659)
        at org.eclipse.equinox.launcher.Main.basicRun(Main.java:595)
        at org.eclipse.equinox.launcher.Main.run(Main.java:1501)
        at org.eclipse.equinox.launcher.Main.main(Main.java:1474)
```

ilg-ul, 2018-11-24:
it looks like the LPC SVD uses a syntax that is not very well supported by the plug-in. I'll investigate, but it may take a few days.

thank you,

Liviu

ilg-ul, 2018-11-25:
I installed the LPC546xx_DFP package, v2.6.0, and selected the LPC54608J512BD208 device, but still cannot replicate the issue:

[image]

Can you provide more details?

bigbrett, 2018-11-29:
Sorry for the delay, was out of town. 

I ended up switching the device in the settings dialog, and then switching it back, and all seems to be working again. If it happens again I will report back, but I guess we can close this one for now.....

closed 2018-11-29 by bigbrett

ilg-ul, 2018-11-29:
ok