eclipse-embed-cdt/eclipse-plugins #328: atollic build analyezer plugin?

mgiaco, 2018-12-11:
Hello, 

Dose someone know any similar plugin von gnu-mcu-eclipse? It is really bad that atollic do not offer there tools as plugins only.

many thanks
cheers

mgiaco, 2018-12-11:
Yes, look here: [image]

mgiaco, 2018-12-11:
I also tried this IDE the first time today. It is really nice. The bad thing all the eclipse based IDE´s have is that there is no good cmake support. I hope that will change some day.

ilg-ul, 2018-12-11:
Ok, it basically shows how much memory the program uses. 

The standard GNU solution is to use the 'size' command, and the GNU MCU Eclipse plug-ins provide a convenient way to run it after the build.

I know that the graphical solution looks better, but it is way more complicated to implement.

ErichStyger, 2018-12-11:
You get that memory map as well with --print-memory-usage, see https://stackoverflow.com/questions/41389450/can-gnu-ld-print-memory-usage-by-memory-space-rather-then-just-as-a-bulk-percen. What Atollic seems to do is to parse that data and show it in a custom view.

ilg-ul, 2018-12-11:
As for cmake, I'm afraid I can't be of much help, I'm not a fan of it.

Actually right now I'm working on a simpler alternativ, called xmake, which I plan to integrate into my workflow and the plug-ins.

For now xmake builds projects with make/ninja, and builds/runs tests. Right now I'm working on the exporter to Eclipse; next the importer from Eclipse is planned, and later an internal builder, which will run the build directly, without the need of a separate tool (like make/ninja).

ilg-ul, 2018-12-11:
> `--print-memory-usage`

You mean add `-Wl,--print-memory-usage` to the linker, and parse the linker output?

ErichStyger, 2018-12-11:
it works for me like this:
Invoking: MCU Linker
arm-none-eabi-gcc -nostdlib -Xlinker -Map="K22FX512xxx12_FreeRTOS.map" -Xlinker --gc-sections -Xlinker -print-memory-usage -mcpu=cortex-m4 -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mthumb -T K22FX512xxx12_FreeRTOS_Debug.ld -o "K22FX512xxx12_FreeRTOS.axf"  ./startup/startup_mk22f12.o  ./source/Application.o ./source/main.o ./source/semihost_hardfault.o  ./freertos/source/croutine.o ./freertos/source/event_groups.o ./freertos/source/list.o ./freertos/source/queue.o ./freertos/source/tasks.o ./freertos/source/timers.o  ./freertos/portable/fsl_tickless_systick.o ./freertos/portable/heap_4.o ./freertos/portable/port.o  ./board/board.o ./board/clock_config.o ./board/peripherals.o ./board/pin_mux.o  ./CMSIS/system_MK22F12.o   
Memory region         Used Size  Region Size  %age Used
   PROGRAM_FLASH:       17956 B       512 KB      3.42%
      SRAM_UPPER:       19048 B        64 KB     29.06%
      SRAM_LOWER:          0 GB        64 KB      0.00%
        FLEX_RAM:          0 GB         4 KB      0.00%
Finished building target: K22FX512xxx12_FreeRTOS.axf

ilg-ul, 2018-12-11:
Yes, you can manually add `-Wl,--print-memory-usage` and inspect the build console.

Do you suggest to add this as a separate entry in the settings page?

mgiaco, 2018-12-11:
@ilg-ul and @ErichStyger  many thanks.

mgiaco, 2018-12-11:
@ilg-ul  do you mean that https://xmake.io

ilg-ul, 2018-12-11:
I mean: https://www.npmjs.com/package/xmake

The public version 0.x is a bit old, v1.x will bring many changes.