eclipse-embed-cdt/eclipse-plugins #333: Can't see all Debug configuration settings

BrendanSimon, 2018-12-24:
### Description

When I go to view/edit the Debug settings (e.g. for J-Link), I can't see some of the settings towards the bottom as the Window isn't large enough and the panel of widgets does not scroll.  You can test/verify this by making the window smaller and will notice you can't access the hidden widgets.

The panel of widgets should be scrollable.  E.g. with the C++ Build Settings, I can scroll the panel around to access all the settings.

I'm using a MacBook with 15" screen, Eclipse 2018-09 (4.9).

### Steps to Reproduce

1. Create a new J-Link Debugger configuration
2. Click on the `Debugger` or `Startup` tab
3. Some of the settings at the bottom are not visible (e.g. `Force thread list on suspend` or `continue` checkboxes).  This may be dependent on system font size, scaling, etc.
4. The problem can be seen by making the window smaller.
5. Try panning around with two button pan gesture (I could not get the contents to pan up or down).

**Expected behaviour:** [What you expected to happen]

Expected to be able to pan around the contents to gain access to all fields.

**Actual behaviour:** [What actually happened]

All attempts of panning do not work.  No pan occurs and only the top/left fields are visible.

The C++ Build Settings behaves slightly differently and I can pan around using my trackpad gestures (two finger pan) if I apply it the right areas.

With the Debugger Settings the scrollbars don't seem to be present at all.  i.e the don't even show up momentarily and the whole pane of widgets just doesn't seem to be scrollable at all.

If you could get the Debugger settings to behave the same as the C++ Build Settings then that would be the best outcome.  At the moment I am lucky enough to have a larger external monitor available to get to some of the other settings, but that's not always the case.

Other possible workarounds:

- Decrease the large vertical space between the cancel and ok buttons at the bottom of the window, and the bounding box of the tabs.  It's just wasted space.
- Rearrange some of the widgets so as to use more horizontal space (i.e. wider dialog box and less 

height).

### Versions

* GNU MCU Eclipse plug-ins v4.4.2-201809300659
* Eclipse 2018-09 (4.9)
* Mac OS X 10.13.6

ilg-ul, 2019-01-01:
I just checked again and Eclipse displays the scroll bars as long as you resize the window, plus a few seconds after releasing the window margin. If you grab the scroll bar whilevisible, it works as expected.

I agree that this is not optimal, since the user needs to be fast to grab the scroll bar, but this behaviour is part of the Eclipse implementation for macOS, and I don't know how to change it.

BrendanSimon, 2019-01-01:
I checked again too.  These are my observations.

The scrollbars **DO** appear if the debug settings dialog is invoked via the `Debug Configurations...` menu (e.g. via `Run` menu or bug icon drop down menu.

The scrollbars **DO NOT** appear if the debug settings dialog is invoked via the "Gear" icon in the "Debug Launcher" widget (i.e. from left to right we have "Build Hammer Icon", "Debug Bug Icon", "Debug Terminate Icon", "Debug/Run mode selection drop down", then "Debug Laucher".  This is all in Eclipse 2018-09.

lorcap, 2019-11-08:
Same problem and same behavior differences between `Run` and `Debug Configurations...` here with:
* GNU MCU Eclipse Plug-in 4.7.1 201911052135
* Eclipse 2019-09
* OpenJDK Runtime Environment (build 10.0.2+13)
* ArchLinux

ilg-ul, 2019-11-08:
The content of the debugging configuration windows is displayed by the Eclipse CDT, using the classes in the Eclipse platform code. The plug-ins do not have any contribution to this, they define only the specific content in some of the windows.

Apparently the Eclipse platform code for macOS and Linux is not very fortunate when dealing with scrolling.

TM1234, 2019-11-08:
That being the case presumably the same effective issue can be reproduced using the default CTD debug dialog and can then be reported against CDT itself (or Eclipse if appropriate)? Somebody would need to take the time to reproduce the issue and report it though...