eclipse-embed-cdt/eclipse-plugins #334: OpenOCD RTOS-aware debugging fails

Novakov, 2019-01-10:
### Description
Device: Microsemi SmatfFusion2 M2S010 
Program: simple program FreeRTOS 10.0 and two tasks.

I've built simple program using FreeRTOS that starts two threads. After flashing on chip it is working properly. I can debug it without problems when not using RTOS support in OpenOCD. After enabling RTOS support in OpenOCD (I've added --command "$_TARGETNAME configure -rtos FreeRTOS" to OpenOCD command line) I can debug program correctly using command-line arm-none-eabi-gdb. Also debug configuration in Eclipse that only attaches to running target works fine. 

However when running debug configuration that restarts MCU it is not possible to debug target: MCU is halted but Eclipse is not reflecting that (no source line is highlighted/no stack trace/etc). Attempt to pause program execution results in 'Interrupt failed' exception. 

In gdb traces I can see that Eclipse tries to switch to thread and gdb server is responding with "Thread ID 2 not known." 

Logs (https://gist.github.com/Novakov/4772ec4142f067dc3c9ec8d2f77f67e0):
* OpenOCD log output (default verbosity) 
* gdb traces log

### Steps to Reproduce

1. Use SmartFusion2 device
2. Create simple FreeRTOS application (two blink tasks)
3. Create debug configuration for OpenOCD 
4. Use Microsemi fork of OpenOCD (bundled in SoftConsole)
5. Use following command line arguments for OpenOCD: `--command "set DEVICE M2S010"  --file board/microsemi-cortex-m3.cfg --command "`$_TARGETNAME configure -rtos auto"`
6. Debug program

**Expected behaviour:** Program is running on MCU, hitting breakpoint shows list of threads.

**Actual behaviour:** MCU is halted, Eclipse looks like waiting for something. Hitting Pause results in 'Interrupt failed' exception.

### Versions

* GNU MCU Eclipse 3.3.3.201901011632
* Eclipse 2018-12 (4.10.0)
* Java 1.8.0_171
* Windows 10 x64
* Toolchain: GNU Tools for Arm Embedded Processors 8-2018-q4-major
* OpenOCD (Microsemi version): GNU MCU Eclipse (Microsemi SoftConsole build) 64-bits Open On-Chip Debugger 0.10.0+dev-00084-gae68d7da-dirty (2018-06-28-11:40)

ilg-ul, 2019-01-10:
Multi-threaded debug configuration are handled by Eclipse CDT itself, the GNU MCU Eclipse plug-ins do nothing special for them.

If you start OpenOCD by hand, externally to Eclipse, and connect the GDB Hardware debug plug-in (NOT the GME OpenOCD plug-in) to it, does the multi-threaded debug session run as expected?

Novakov, 2019-01-11:
Ok, I've checked:
* OpenOCD with RTOS enabled running in console
* GDB Hardware debugging with Generic TCP/IP configured for OpenOCD

Result: a bit different:
* After starting it's stopping on random line of code and shows correct task list
* Resuming and stopping causes 'Interrupt failed' exception

It's seems to me like GDB is getting tasklist before OpenOCD performs flash operation (which include halt and MCU restart). I have no idea (unfortunately) how the internals works. Unfortunately I don't have any MCU that can work with OpenOCD now, so I'm unable to rule out possible problem with OpenOCD modified by Microsemi.

If someone here is willing to verify the issue with another MCU, I would be greatful.

AntonKrug, 2019-01-11:
Hi @Novakov can you clarify what you mean exactly under "debug configuration that restarts MCU"?

AntonKrug, 2019-01-11:
@Novakov Do I understand it correctly? When the ELF file is already loaded (like from envm) then you can debug it, but when you load the elf while you are attaching to the target it doesn't. I'm just guessing but are the other Initial and Pre-run reset settings correct?

Novakov, 2019-01-11:
@AntonKrug Yes, that's correct. And the problem is only when RTOS support is enabled in OpenOCD. I didn't change the Initial/Pre-run settings as they work fine when RTOS support is disabled.

TM1234, 2019-01-11:
Hi @Novakov  - thanks for the clarifications. When you're debugging "standalone" from SoftConsole/Eclipse/GME are you sure that you're telling GDB to use the gdbinit/softconsole.gdbinit script bundled with SoftConsole as it does some stuff that may make a different when debugging? For example if the debug session is downloading the program to envm and then executing it from there then the gdbinit script will set the $sp and VTOR (Vector Table Offset Register - the address of the vector table) correctly. If you don't use this then these will probably be wrong and this might explain the behaviour that you're seeing?

Novakov, 2019-01-11:
@TM1234 That might be the cause but... I've just updated SoftConsole to 6.0 (from v5.3) and it works correctly (or at least seems to be working). What's funny that the only thing that changed is the OpenOCD version (I'm using arm-none-eabi from ARM and standalone Eclipse).

OpenOCD version in SoftConsole 5.3:
 `GNU MCU Eclipse (Microsemi SoftConsole build) 64-bits Open On-Chip Debugger 0.10.0+dev-00084-gae68d7da-dirty (2018-06-28-11:40)`

OpenOCD version in SoftConsole 6.0:
`GNU MCU Eclipse (Microsemi - a Microchip company - SoftConsole build) 64-bit Open On-Chip Debugger 0.10.0+dev-00352-gaa6c7e9b-dirty (2018-10-22-10:08)`

Unfortunately none of commits mentioned in version string is publicly available so we can only guess what changed.

Anyway, that's for helping with sorting out the issue and I'm sorry for not doing the obvious thing - update everything :)

TM1234, 2019-01-11:
Hi @Novakov  - thanks for the update - that's interesting/strange alright. When targeting SmartFusion2 it is important, in the general case, that gdb uses the gdbinit/softconsole.gdbinit script bundled with SoftConsole. That is the same with both SoftConsole v5.x and v6.0. The SoftConsole Eclipse/CDT/GME based IDE is preconfigured to cause this to happen but when using gdb "standalone" you need to provide the script to gdb yourself. If that's not the problem here then I'm not sure why the problem arose with SoftConsole v5.3 at the moment and would need to replicate your setup (or something similar) to check. In the meantime are you OK with using SoftConsole v6.0 if things are working there? 

BTW what M2S010 board are you using? Is it the Maker board?

https://www.digikey.com/eewiki/display/microcontroller/Getting+Started+with+the+Microsemi+SmartFusion+2+Maker-Board

Novakov, 2019-01-11:
@TM1234 I will keep that in mind :) I see no problem in using SoftConsole 6.0.

The board I'm using is this one: https://shop.trenz-electronic.de/en/TEM0001-01A-010C-SMF2000-FPGA-Module-with-Microsemi-SmartFusion2-8-MByte-SDRAM?c=486

BTW. Do you know if there are any plans of adding the microsemi-flashpro interface to the "official" OpenOCD? In this case it would be easier to track down the issue :)

closed 2019-01-11 by Novakov

ilg-ul, 2019-01-11:
> thanks for facilitating this discussion

you're welcome!

AntonKrug, 2019-01-11:
> is a big fan. :-)

That is an understatement, hope the Trenz guys will do more boards with our FPGAs.

When the $target_arm variable is not set then the gdbinit will go into the auto-detection branch and maybe then together with the rtos could get confused?

Wonder what would happen with the SC 5.3  with $target_arm set.

Yes, the OpenOCD situation is not ideal.

If you will give me STAPL files I will try your design with SC 5.3 to replicate the issue, so we will understand exactly what was the cause.

Novakov, 2019-01-11:
@AntonKrug I've attached .stp file to the gist I've linked before (hopefuly that's the one you were asking for). I think that any FreeRTOS program would result in the same behavior but if you want I can send you the exact one I was working with (just need to reduce it a bit)

AntonKrug, 2019-01-14:
@Novakov Thank you very much. If you could attach zip of a small project which can replicate it directly to rule out any ambiguity, that would help.

Novakov, 2019-01-18:
@AntonKrug Sorry for the delay. In attachment you will find very simple FreeRTOS program with two tasks. It is based on the Reference design for SMF2000 board but I guess it will work with anything (apart from systick frequency)

[freeRTOS.zip](https://github.com/gnu-mcu-eclipse/eclipse-plugins/files/2773795/freeRTOS.zip)

AntonKrug, 2019-01-18:
@Novakov No problem, thank you very much.