eclipse-embed-cdt/eclipse-plugins #335: Certain Silicon Labs CMSIS packs error with a -1 length

jahess, 2019-01-11:
While updating the pack lists, certain packages from Silicon Labs report "illegal PDSC file size -1."

The devices with similar errors are:
 
EFM32GG11B_DFP.pdsc
EFM32LG_DFP.pdsc
EFM32TG11B_DFP.pdsc
EFM32WG_DFP.pdsc
EFM32GG12B_DFP.pdsc

Screen shots at:  [https://www.element14.com/community/message/270275/l/some-silicon-labs-cmsis-packs-report-a-length-of-1#270275](url)

TM1234, 2019-01-11:
Link to screenshots above seems to be incorrect (text is correct but the URL that is linked is wrong):

https://www.element14.com/community/message/270275/l/some-silicon-labs-cmsis-packs-report-a-length-of-1#270275

ilg-ul, 2019-01-11:
Which URL is wrong? 

I manually tried:

```console
$ curl -L https://www.silabs.com/documents/public/cmsis-packs/SiliconLabs.EFM32GG11B_DFP.pdsc
```

and I got the .pdsc file.

TM1234, 2019-01-11:
When I click on the link in the original post it goes to:

https://github.com/gnu-mcu-eclipse/eclipse-plugins/issues/url

instead of this (which is the text of the URL):

https://www.element14.com/community/message/270275/l/some-silicon-labs-cmsis-packs-report-a-length-of-1#270275

jahess, 2019-01-11:
Fixed the link.

ilg-ul, 2019-09-21:
Fixed in 4.6.1, on 2019-09-21.

The URL conn.getContentLength() returns -1 when the file is sent in chunks, so it cannot be used; instead it must be computed.

closed 2019-09-23 by ilg-ul