eclipse-embed-cdt/eclipse-plugins #336: Peripheral view uses incorrect address for registers in SVD cluster array

cypress-neil, 2019-01-14:
### Description

This [SVD file](https://github.com/gnu-mcu-eclipse/eclipse-plugins/files/2757320/subset.svd.txt) (excerpt from a SVD file for an ARM device) contains a peripheral with registers organized in clusters. Each cluster contains the dim and dimIncrement elements to define an array. In the Peripherals view, the first element has the correct register addresses but each subsequent element has the same register addresses as the first element.

[image]

### Steps to Reproduce

1. Create a GNU ARM Eclipse debug configuration. As far as I know it should be possible to view the register addresses in the Peripherals view even if the addresses are not compatible with the actual target. It should not be necessary to build and load an actual program to use the Peripherals view. (I was not able to get the QEMU debug configuration to work with this version of GNU MCU Eclipse; I used an OpenOCD target with a Cypress PSoC 6 device).
2. Start a debug session, open the Peripherals view, and open the PERI_MS peripheral
3. Observe the register addresses

**Expected behaviour:**
The register addresses should be:
PPU_PR[0].SL_ADDR 0x40010000
PPU_PR[0].SL_ADDR 0x40010004
PPU_PR[1].SL_ADDR 0x40010040
PPU_PR[1].SL_ADDR 0x40010044
PPU_PR[2].SL_ADDR 0x40010080
PPU_PR[2].SL_ADDR 0x40010084
...

**Actual behaviour:**
It looks like the Peripherals view did not take the cluster's dimIncrement into account:
PPU_PR[0].SL_ADDR 0x40010000
PPU_PR[0].SL_ADDR 0x40010004
PPU_PR[1].SL_ADDR 0x40010000
PPU_PR[1].SL_ADDR 0x40010004
PPU_PR[2].SL_ADDR 0x40010000
PPU_PR[2].SL_ADDR 0x40010004
...

This issue is not exactly the same as [#150](https://github.com/gnu-mcu-eclipse/eclipse-plugins/issues/150) which is already fixed--the clusters have addressOffset=0 so they are not affected by #150.

### Versions

* Plugin version: 4.1.2.201804191501
* Eclipse version: 4.7.1.v20171009-0410
* Java version: 1.8.0_151-b12
* Windows 7 SP1 x64 (6.1.7601)
* gcc-arm-embedded 7-2018-q2-update

ilg-ul, 2019-01-15:
Thank you for the report, SVDs are tricky to process. Time permitting, I'll investigate.

martinKupec, 2019-03-12:
I believe that I have come to the same problem. My problem is with ATSAMC21G18A svd provided by Atmel. The PORT groups are not working.

I have found solution for my problem and with my limited knowledge of the code it seems it should work also for other similar problems.
The fix is:
```
diff --git a/bundles/ilg.gnumcueclipse.debug.gdbjtag/src/ilg/gnumcueclipse/debug/gdbjtag/datamodel/SvdClusterDMNode.java b/bundles/ilg.gnumcueclipse.debug.gdbjtag/src/ilg/gnumcueclipse/debug/gdbjtag/datamodel/SvdClusterDMNode.java
index 686ca86b..53e40fad 100644
--- a/bundles/ilg.gnumcueclipse.debug.gdbjtag/src/ilg/gnumcueclipse/debug/gdbjtag/datamodel/SvdClusterDMNode.java
+++ b/bundles/ilg.gnumcueclipse.debug.gdbjtag/src/ilg/gnumcueclipse/debug/gdbjtag/datamodel/SvdClusterDMNode.java
@@ -104,5 +104,14 @@ public class SvdClusterDMNode extends SvdDMNode {
                }
        }
 
+       @Override
+       public BigInteger getBigRepeatIncrement() {
+               BigInteger bigRepeatIncrement = getBigArrayAddressIncrement();
+               if (bigRepeatIncrement != BigInteger.ZERO) {
+                       return bigRepeatIncrement;
+               }
+               return null;
+       }
+
        // ------------------------------------------------------------------------
 }
```

ilg-ul, 2019-03-12:
Thank you for the code. 

I need to take a closer look, hopefully before the next release (2019-03).

ilg-ul, 2019-09-15:
Fixed in 4.6.1 on 2019-09-15.

The patch provided by @martinKupec was used as is, and the `subset.svd` provided by @cypress-neil was helpful to test the patch. Thank you both!

ilg-ul, 2019-09-15:
I just pushed a prerelease on 

http://gnu-mcu-eclipse.netlify.com/v4-neon-updates-experimental

You can Install new software from there and test before the official release is out.

closed 2019-09-23 by ilg-ul