eclipse-embed-cdt/eclipse-plugins #338: Cortex-R52 and armv8-R architecture not available

ahenric, 2019-01-15:
When using the GNU Arm embedded toolchain, the Cortex-R52 option is not available under the Target Processor drop down list. Also, the armv8-r option is not available under Architecture. 

Will that be still added?

TM1234, 2019-01-15:
In the meantime @ahenric can you just set Target Processor and Architecture to Toolchain default and then manually specify the CPU and arch in Other target flags? E.g. -mcpu=cortex-r52 -march=armv8-r? Maybe -march is redundant here if the tools infer the default arch from the cpu?

ahenric, 2019-01-16:
The -mcpu=cortex-r52 switch conflicts with -march=armv8-r. So either or setting should be used.

ilg-ul, 2019-01-16:
The point was that until the plug-in will be updated, you can set the graphical selection to default, and manually add whatever options you need in the separate Other field.

TM1234, 2019-01-16:
Yes - that's what I meant alright.
Configure for toolchain defaults for Target processor and Architecture but override the defaults in the Other target flags input field. 
I guess that that should work for the moment.

closed 2019-09-23 by ilg-ul