eclipse-embed-cdt/eclipse-plugins #339: CMSIS-Pack: store compressed on disk

flit, 2019-02-03:
### Description

CMSIS-Packs should be stored on disk in compressed form in order to reduce disk usage, which can be considerable if you have many packs installed.

Ideally, files in a pack would be extracted in memory at runtime. Next best would be to extract a pack to a temporary directory while it's in use.

flit, 2019-02-07:
As of version 0.16.0, pyOCD now has support for using CMSIS Packs to add device support. I was planning on updating the GME plugin to automatically pass the pack and device selected in the project build settings to pyOCD. Right now pyOCD requires a compressed .pack file, though I'll be adding support for exploded packs specifically to support them in the GME plugin.

ilg-ul, 2019-02-07:
as you wish.

the original packs are on the disk in the cache folder, you can get them from there, but I do not guarantee that the location remains unchanged.

ilg-ul, 2019-02-07:
and what do you plan to use from CMSIS Packs?

I don't know if you noticed, but the other debug plug-ins can use a separate *.svd (or *-xsvd.json) files, without the need to have an associated CMSIS Pack.

ilg-ul, 2019-09-29:
FYI, in v4.6.1 the default location of the CMSIS Packs folder changed.

Anyway, you should not use it directly, but read the configuration variable.

ilg-ul, 2020-12-21:
A quick update, I'm migrating more and more content to the xPacks, for example the STM32F4 package now has the interrupt vectors in C, not assembly.

- https://github.com/micro-os-plus/devices-stm32f4-xpack/tree/develop

In the same packages I'll add the SVD files (most probably in compressed format, since the new plug-ins already accept .svd.zip), so, at least for my use case, there will be no need for the CMSIS Packs at all.

flit, 2020-12-29:
Fyi, I'm looking again at integrating the pyocd plugin with Embed CDT's DFP support. Currently pyocd uses the device information from the PDSC (memory regions, etc) and the .FLM flash algo files. I'm also working on supporting debug sequences in the PDSC for any required custom device connect/etc logic (more and more vendors are starting to define debug sequences for their devices when they require custom logic). It will also support the .dbgconf files at some point.

Btw, I totally agree with your concern about the size of certain packs. The Keil packs for STM32 devices are the primary offender since they include STM32Cube. On the opposite end, the NXP packs are quite small and focused on single devices, and split DFP from BSP.