eclipse-embed-cdt/eclipse-plugins #340: GDB SEGGER J-Link Debugging plug-in: Allow special characters for IP address

SEGGER-LG, 2019-02-27:
### Description

Inside the "debugger" configuration tab provided by the GDB SEGGER J-Link Debugging plug-in, the input field for the IP does not allow special characters (e.g. # § $ % &).
This is problematic, as the "IP" that is passed to the J-Link does not necessarily have to be an IP address when connecting to a J-Link via a tunneled connection using the J-Link Remote Server.
(More information about the J-Link Remote Server can be found here: https://www.segger.com/products/debug-probes/j-link/tools/j-link-remote-server/).
An example of such a connection string would be:

tunnel:$MyNickname§:#MyPassword#

Which is not possible to enter into the IP field at the moment.

### Steps to Reproduce

In Eclipse...
1. Open **Run** -> **Debug Configurations...**
2. Select a debug configuration from **GDB SEGGER J-Link Debugging**
3. Switch to the  **Debugger** tab
4. Try to enter "**tunnel:$MyNickname§:#MyPassword#**" into the "IP" input field

**Expected behaviour:**

I expect to be able to enter my tunnel-connection string into the IP-field.

**Actual behaviour:**

It is not possible for me to enter special characters like $ § # and so on...

### Versions

* Plug-in: 4.5.1
* Eclipse: Neon.3 Release (4.6.3)
* Java version: 1.8.0_201
* OS: Windows 10 Home

SEGGER-LG, 2019-02-27:
The described above applies to **IP configurations only**.
The IP connection string is limited to ASCII characters only (< 0x80).
Other than that there are no restrictions.

ilg-ul, 2019-09-10:
Fixed in 4.6.1, on 2019-09-10.

closed 2019-09-23 by ilg-ul