eclipse-embed-cdt/eclipse-plugins #341: Packages folder issue

vredzhepov, 2019-03-25:
I've changed default path for packages folder. Each time I open Eclipse IDE the old Packages folder appears again.

Plugin version: 1.1.5.201901011632
Eclipse version: 2019-03 (4.11.0)

ilg-ul, 2019-03-26:
Can you list the exact steps you did, so I can reproduce the issue?

vredzhepov, 2019-03-26:
Hi @ilg-ul 

Yes. I can reproduce the steps.
1. Install Eclipse version: 2019-03 (4.11.0)
2. Plugin version: 1.1.5.201901011632
3. Open packages manager in Eclipse and start update
4. Go to Eclipse preferences and change default location of packages folder (MCU Packages) that by default located in home directory in Windows
5. Update packages
6. Go to newly created packages directory and check that there's new data after packages update
7. Delete default packages directory
8. Start working with Eclipse
9. The default Packages directory appears again
10. Go to step 7

ilg-ul, 2019-03-26:
Did you try to exit Eclipse after making the change?

vredzhepov, 2019-03-26:
Do you mean after changing default location of Packages directory?

ilg-ul, 2019-03-26:
Yes, before updating for the second time.

vredzhepov, 2019-03-26:
I don't remember that. I've just started Eclipse IDE and checked out the settings and they show that the new location is saved but the old packages directory appeared again. How can I disable the old directory to appear again?

ilg-ul, 2019-03-26:
if you read the settings page now, what do you see, the original or the new path?

ilg-ul, 2019-03-26:
anyway, for various reasons, after changing the path you need to exit Eclipse, enter again, and run the update.

vredzhepov, 2019-03-27:
On the settings page I see the new path. It hasn't been changed since I modified it. Could you suggest me what to do in the current case. Do I need to reinstall Eclipse or something else. By the way how does plugin know about the old path if it has already changed? Where does the plugin hold its settings?

ilg-ul, 2019-03-28:
I'll investigate when I'll work on the 2019-03 release.

on my mac I move the Packages folder where I need it, and make a soft link to it. I don't know if folder links are now available for Windows.

vredzhepov, 2019-03-28:
@ilg-ul I contend that I watch the same behavior on both Windows and Linux. Where does plugin actually hold its settings?

vredzhepov, 2019-06-07:
Hi. Have you managed to reproduce this bug?

ilg-ul, 2019-06-08:
unfortunately I had some hardware problems and now I'm in the middle of upgrading my machines.

I'll take a look at this as soon as possible, but it'll still take some time.

vredzhepov, 2019-06-16:
Hi. Is there any news regarding this issue?

ilg-ul, 2019-06-16:
not yet

vredzhepov, 2019-07-02:
Author isn't capable to fix this issue during 3 months. I gave up using this plugin.

ilg-ul, 2019-07-02:
The project is under major restructuring, all tools are moved to the separate xPacks project, to facilitate their use outside Eclipse (in command line applications, in Visual Studio Code and possibly other environments).

As soon as this is completed, the plug-ins and the documentation will be updated.

We'll keep this ticket open, to be sure it'll be fixed.

ilg-ul, 2019-09-02:
I finally had some time to check this and I can confirm that a folder named Packages is created in the same location as the workspace, but in my case it is empty, the content downloaded by the Update button is in the expected location and so far I cannot see any functional anomalies.

I'll further investigate.

vredzhepov, 2019-09-03:
I can confirm too that that folder named Packages exists after installing plugin in my case its $HOME/Packages. But if you change the path for this folder let's say to /opt/packages the plugin will use /opt/packages but $HOME/Packages will be created each time you run Eclipse.

ilg-ul, 2019-09-03:
So for now the bug to fix is to no longer create the useless Packages folder? I'll fix this in the next release.

If you find other anomalies, please let me know.

vredzhepov, 2019-09-04:
Yes. If the path has changed no needed to create additional Packages folder.

ilg-ul, 2019-09-11:
Fixed in 4.6.1 on 2019-09-11.

ilg-ul, 2019-09-11:
@vredzhepov, I fixed this issue, and also changed the default folder location [see #357].

Could you check the pre-release version available from:
 
http://gnu-mcu-eclipse.netlify.com/v4-neon-updates-experimental

vredzhepov, 2019-09-11:
At first glance it seems to be working as expected. I've removed the previous version of the plugin and installed the latest plugin provided by link above. The directory named "Packages" not created and plugin uses the directory that I specified in the preferences. So I think this issue has been fixed. Thank you.

ilg-ul, 2019-09-11:
Ok, great!

Thank you for testing and for taking the time to report this bug.

closed 2019-09-23 by ilg-ul