eclipse-embed-cdt/eclipse-plugins #343: ilg.gnumcueclipse.debug.gdbjtag.datamodel.SvdUtils - typo in computeScale method (scale T does not processed)

balokir, 2019-04-08:
private static long computeScale(String str) {
	...
			if (lastChar.matches("[kmgtKMGT]")) {
				lastChar = lastChar.toLowerCase();
				if ("k".equals(lastChar)) {
					scale = 1024;
				} else if ("m".equals(lastChar)) {
					scale = 1024 * 1024;
				} else if ("g".equals(lastChar)) {
					scale = 1024 * 1024 * 1024;
				} else if ("7".equals(lastChar)) {  
// ***************************** seems to be a mistake  - should be "t"
					scale = 1024 * 1024 * 1024 * 1024;
				}
			}
		}
		return scale;
	}

ilg-ul, 2019-04-08:
you are right, it is typo.

I'll fix it in the next release, due soon.

thank you,

Liviu

ilg-ul, 2019-09-10:
fixed in 4.6.1 on 2019-09-10.

closed 2019-09-23 by ilg-ul