eclipse-embed-cdt/eclipse-plugins #348: J-Link support for RISC-V

kenrabold, 2019-06-14:
It looks like GDB SEGGER J-Link debugging only supports ARM. Will it be updated to support RISC-V as well?

When using version 4.3.4 and Eclipse 20190-03 all setup with JLink v6.46d and RISC-V Embedded GCC 8.2.0-2.2 - I can build and download a simple program for the new SiFive HiFive1B which has SEGGER JLink debug capabilities.  The GDB debug connection starts up, but seems to failout when accessing some of the registers:

SEGGER J-Link GDB Server V6.46d Command Line Version

JLinkARM.dll V6.46d (DLL compiled Jun  7 2019 17:33:40)

Command line: -if jtag -device FE310 -endian little -speed 1000 -port 2331 -swoport 2332 -telnetport 2333 -vd -ir -localhostonly 1 -log C:\Save\eclipse-workspace\jlink_gnumcu.log -singlerun -strict -timeout 0 -nogui
-----GDB Server start settings-----
GDBInit file:                  none
GDB Server Listening port:     2331
SWO raw output listening port: 2332
Terminal I/O port:             2333
Accept remote connection:      localhost only
Generate logfile:              on
Verify download:               on
Init regs on start:            on
Silent mode:                   off
Single run mode:               on
Target connection timeout:     0 ms
------J-Link related settings------
J-Link Host interface:         USB
J-Link script:                 none
J-Link settings file:          none
------Target related settings------
Target device:                 FE310
Target interface:              JTAG
Target interface speed:        1000kHz
Target endian:                 little

Connecting to J-Link...
J-Link is connected.
Firmware: J-Link OB-K22-SiFive compiled May 27 2019 15:39:22
Hardware: V1.00
S/N: 979000465
Checking target voltage...
Target voltage: 3.30 V
Listening on TCP/IP port 2331
Connecting to target...
J-Link found 1 JTAG device, Total IRLen = 5
JTAG ID: 0x20000913 (RISC-V)
Connected to target
Waiting for GDB connection...Connected to 127.0.0.1
Reading all registers
Read 1 bytes @ address 0x00000000 (Data = 0x00)
Read 1 bytes @ address 0x00000000 (Data = 0x00)
Read 1 bytes @ address 0x00000000 (Data = 0x00)
Read 1 bytes @ address 0x00000000 (Data = 0x00)
Read 1 bytes @ address 0x00000000 (Data = 0x00)
Read 1 bytes @ address 0x00000000 (Data = 0x00)
Received monitor command: speed 1000
Target interface speed set to 1000 kHz
Received monitor command: speed auto
Select auto target interface speed (4000 kHz)
Received monitor command: flash breakpoints 0
Flash breakpoints disabled
Downloading 100 bytes @ address 0x20010000 - Verified OK
Downloading 9950 bytes @ address 0x20010064 - Verified OK
Downloading 672 bytes @ address 0x20012744 - Verified OK
Downloading 4 bytes @ address 0x200129E4 - Verified OK
Downloading 112 bytes @ address 0x200129E8 - Verified OK
Comparing flash   [....................] Done.
Verifying flash   [....................] Done.
Writing register (fp = 0x20010000)
Read 1 bytes @ address 0x20010000 (Data = 0x97)
Read 1 bytes @ address 0x20010000 (Data = 0x97)
Read 1 bytes @ address 0x20010000 (Data = 0x97)
Read 1 bytes @ address 0x20010000 (Data = 0x97)
Read 1 bytes @ address 0x20010000 (Data = 0x97)
Reading 14 bytes @ address 0x20010000
Read 1 bytes @ address 0x20010000 (Data = 0x97)
Read 1 bytes @ address 0x20010000 (Data = 0x97)
Read 1 bytes @ address 0x20010000 (Data = 0x97)
Read 1 bytes @ address 0x20010000 (Data = 0x97)
Read 1 bytes @ address 0x20010000 (Data = 0x97)
Read 1 bytes @ address 0x20010000 (Data = 0x97)
Read 1 bytes @ address 0x20010000 (Data = 0x97)
Reading 14 bytes @ address 0x20010000
Read 1 bytes @ address 0x20010000 (Data = 0x97)
Read 1 bytes @ address 0x20010000 (Data = 0x97)
Read 1 bytes @ address 0x20010000 (Data = 0x97)
Received monitor command: regs
X0 = 00000000, X1 = 00000000, X2 = 00000000, X3 = 00000000
X4 = 00000000, X5 = 00000000, X6 = 00000000, X7 = 00000000
X8 = 20010000, X9 = 00000000, X10= 00000000, X11= 00000000
X12= 00000000, X13= 00000000, X14= 00000000, X15= 00000000
X16= 00000000, X17= 00000000, X18= 00000000, X19= 00000000
X20= 00000000, X21= 00000000, X22= 00000000, X23= 00000000
X24= 00000000, X25= 00000000, X26= 00000000, X27= 00000000
X28= 00000000, X29= 00000000, X30= 00000000, X31= 00000000
PC = 00000000
Reading all registers
Read 1 bytes @ address 0x00000000 (Data = 0x00)
Read 1 bytes @ address 0x00000001 (Data = 0x00)
Read 1 bytes @ address 0x00000000 (Data = 0x00)
Read 1 bytes @ address 0x00000001 (Data = 0x00)
Read 1 bytes @ address 0x00000002 (Data = 0x00)
Read 1 bytes @ address 0x00000003 (Data = 0x00)
Read 1 bytes @ address 0x00000002 (Data = 0x00)
Read 1 bytes @ address 0x00000003 (Data = 0x00)
Read 1 bytes @ address 0x00000004 (Data = 0x00)
Read 1 bytes @ address 0x00000005 (Data = 0x00)
Read 1 bytes @ address 0x00000004 (Data = 0x00)
Read 1 bytes @ address 0x00000005 (Data = 0x00)
Read 1 bytes @ address 0x00000006 (Data = 0x00)
Read 1 bytes @ address 0x00000007 (Data = 0x00)
Read 1 bytes @ address 0x00000006 (Data = 0x00)
Read 1 bytes @ address 0x00000007 (Data = 0x00)
Read 1 bytes @ address 0x00000008 (Data = 0x00)
Read 1 bytes @ address 0x00000009 (Data = 0x00)
Read 1 bytes @ address 0x00000008 (Data = 0x00)
Read 1 bytes @ address 0x00000009 (Data = 0x00)
Read 1 bytes @ address 0x0000000A (Data = 0x00)
Read 1 bytes @ address 0x0000000B (Data = 0x00)
Read 1 bytes @ address 0x0000000A (Data = 0x00)
Read 1 bytes @ address 0x0000000B (Data = 0x00)

When I use the SiFive FreedomStudio, which looks to be based off of the GNU MCU plugins, I can download and debug. I use the same J-Link V6.46d in both cases, so it looks to be the plug-in commands being sent to JLink that is the issue.

Attached are logs from both for more details.

[jlink_gnumcu.log](https://github.com/gnu-mcu-eclipse/eclipse-plugins/files/3292425/jlink_gnumcu.log)
[jlink_sifive_freedomstudio.log](https://github.com/gnu-mcu-eclipse/eclipse-plugins/files/3292426/jlink_sifive_freedomstudio.log)

I would really like to see support for JLink from the GNU MCU!

Thanks

ilg-ul, 2019-06-15:
I'll take a look, but unfortunately I do not have a new HiFive1 to test.

ErichStyger, 2019-06-15:
I do have a board available. If time permits I'll try it out.

ilg-ul, 2019-06-15:
Thank you Erich. If you could identify which command confuses the debugger, that would be very helpful.

ilg-ul, 2019-06-15:
@kenrabold, could you also compare the gdb trace logs? they are easier to read than the j-link logs. you need to enable them in the preferences pages.

kenrabold, 2019-06-17:
I collected the GDB trace logs and found my problem. The Debug Configuration was picking up the wrong GDB exe. I had to force the GDB Client Setup to the path of the risc-v gdb. The variables ${cross_prefix}gdb${cross_suffix} were not set, so the default GDB was picked up which was for mingw32.

Setting the correct RISC-V gdb fixed the problem for me.

Thanks

ilg-ul, 2019-06-17:
> The variables ${cross_prefix}gdb${cross_suffix} were not set

was this a project-less debug session?

ilg-ul, 2019-06-17:
not to mention that the documentation repeatedly reminds **not** to install toolchains in the system paths. there is a reason for this...

kenrabold, 2019-06-17:
It was a project from an existing makefile. It looks like I have some issues with the plugins install. I may try a clean install of Eclipse and the GNU MCU tools and see if I have the same problems.

ilg-ul, 2019-06-17:
The macros are substituted only for managed projects. For makefile projects, you have to explicitly define the values, the plug-ins do not have the required data to do it for you.

closed 2019-06-17 by ilg-ul