eclipse-embed-cdt/eclipse-plugins #349: GDB PyOCD Debugging debug configuration opens very slowly on Windows

AndriiLishchynskyi, 2019-06-19:
### Description

When you open PyOCD debug configuration Eclipse hangs for approximately 15 seconds, only after that time debug configuration is available. Opening PyOCD debug configurations is extremely painful when you need to make some fixes few times in a row.

Key notes:
- reproducible on Windows only
- not reproducible if PyOCD path points to wrong location
- there are a lot of TCP/IP traffic when you open PyOCD debug configuration - for some reason plugin communicates with pyocd gdb
- this is not machine-specific, I have heard from lots of people about this problem

### Steps to Reproduce
1. Create new fresh workspace
2. Create new Project
 - go to **File** -> **New** -> **C/C++ Project**
 - select **C Managed Build** and press **Next**
 - fill in **Project name**
 - select **Project type** = **Empty Project**
 - select **Toolchains** = **Cross GCC**
 - Press **Next**, **Next**, then **Finish**
3. Create new debug configuration
 - go to **Run** -> **Debug Configurations**
 - right click the **GDB PyOCD Debugging** and choose **New Configuration**. Creation takes up to 15 seconds (for OpenOCD time is < 1 second)
4. close **Debug Configurations** window
5. go to **Run** -> **Debug Configurations** and select PyOCD debug configurations. Observe it hangs for ~ 15 seconds

Screen capture: https://drive.google.com/open?id=19quDPapnm-weUgKgK6Nfpx4AqBjn_6nH

**Expected behaviour:**
PyOCD debug configurations opens faster (like OpenOCD does, it takes up to 1 second)
**Actual behaviour:**
PyOCD debug configurations opens ~ 15 seconds

### Versions

* pyocd 0.21.0
* GNU MCU Eclipse 4.5.1.201901011632
* Eclipse IDE for C/C++ Developers. Version: 2018-09 (4.9.0). Build id: 20180917-1800
* Eclipse CDT Version: 9.5.5.201811180605
* Java 8 Update 211
* Windows 10 Pro x64
* Cross GCC, but that actually does not matter

---

ilg-ul, 2019-06-19:
@flit?

ilg-ul, 2019-09-29:
I don't know how the PyOCD plug-in works, it might have a problem, or it might wait for PyOCD to return something before displaying the configuration page.

flit, 2020-09-28:
Whoops, totally missed the notification for this. Sorry about that!

I'm not sure exactly why opening the plugin would be slower on Windows. It may depend on the connected board(s). Specifically for boards with on-board STLink, it can take noticeable time under Windows to map the STLink USB device to the mounted drive letter and read the board ID (to auto-detect the target type). The delay would be longer for each additional STLink-based board that is connected.

The plugin needs to be changed to asynchronously populate the popup menu with connected boards.

Regarding the TCP accesses, I don't have any idea what that would be. The pyOCD plugin runs `pyocd-gdbserver` to generate a JSON list of connected boards (similar to `pyocd list` but in JSON), but that is returned via stdout. It doesn't use the network for anything.

I'll see if I can reproduce this in my system.

ilg-ul, 2020-09-28:
No problem, you're only one year late. :-(

ilg-ul, 2020-12-20:
@flit, v6.0.0 is out now, if you want to fix any issues, please do so, in mid January I plan to release 6.1.0 and we can include your changes there.

flit, 2020-12-20:
I'm working on it! 😄

ilg-ul, 2020-12-20:
Great! When ready, please submit a PR.

ilg-ul, 2020-12-21:
Chris @flit, all my changes for this new release were pushed, so please be sure you rebase your work on the latest commits and submit your PR against develop.

PR #466 (2020-12-29): Numerous pyocd updates
This pull request combines updates, bug fixes, and general improvements for the pyOCD launch config plugin.

Major changes:
- Support for modern `pyocd` combined tool (instead of deprecated `pyocd-gdbserver` tool).
- The majority of the changes are related to the fix for #349. All calls out to pyocd are performed asynchronously on a DSF executor thread and the results returned via a `DataRequestMonitor<?>`. While waiting for a call to pyocd to finish, a message is shown in the launch config window.
- When a launch config doesn't have a probe saved in it (e.g. a new confi [... cut]
closed 2020-12-29 by ilg-ul