eclipse-embed-cdt/eclipse-plugins #350: post-build script

esaulenka, 2019-07-31:
Please, change makefile generator.

I am using post-build script to convert output hex-file to our own format.
Now `make` working in next order:
- create elf-file
- run post-build
- convert elf -> hex

But in this way my script handles old version of hex-file, remaing from previous run.
Please, create dependency that `post-build` target requires `secondary-outputs`

ilg-ul, 2019-07-31:
Unfortunatelly the post-build step is handled by CDT itself, not the GME plug-ins.

I suggest you simply disable the current conversion step (in the Toolchain tab) and use only your custom script.

esaulenka, 2019-07-31:
Ok, thanks!
I came up the same way: first i run objcopy and then process output hex.

But this method looks ugly...
```
    import subprocess
    in_file = os.path.splitext(out_file)[0] + '.elf'
    in_file = '"' + in_file + '"'
    out_file = '"' + out_file + '"'
    cmd = 'arm-none-eabi-objcopy -O ihex ' + in_file + ' ' + out_file
    res = subprocess.call(cmd, shell=not IsRunOnWindows())
    if res: exit()
```

closed 2019-07-31 by esaulenka