eclipse-embed-cdt/eclipse-plugins #354: Debug RISC-V with QEMU failed: qemu-system-riscv32.exe: --verbose: invalid option

howardliu7874, 2019-09-01:
I tried to run QEMU (RISC-V) with the latest Eclipse CDT and the MCU plug-ins. What I did is to set up a debug configuration under GDB QEMU Debugging. But after setting, I run debug but got an error prompt: _qemu-system-riscv32.exe: --verbose: invalid option_.  I didn't find a checkbox available to disable this option in the Eclipse debug settings configure plug-in.

ilg-ul, 2019-09-01:
The QEMU debugging plug-in uses features available only in the qemu-system-gnuarmeclipse, it does not work with qemu-system-riscv32.

closed 2019-09-01 by ilg-ul

fanghuaqi, 2020-11-13:
Did latest version support riscv qemu?

ilg-ul, 2020-11-13:
Unfortunately no.

fanghuaqi, 2020-11-13:
Any plan about supporting RISC-V QEMU?

ilg-ul, 2020-11-13:
I did not check the most recent version, but the upstream QEMU, including RISC-V, does not play very well with Eclipse, it has no way to inform the debug plug-ins when the server started properly.

The solution is to start QEMU manually and then start the debug plug-in.

The xPack QEMU Arm was modified to automate starting it from Eclipse.

fanghuaqi, 2020-11-16:
Thank you @ilg-ul, I can get it work by start QEMU manually and connect with gdb remotely, if possible, I would like there is a feature to directly use the qemu debug plugin for risc-v

ilg-ul, 2020-11-16:
You're welcome!

Could you pass me a QEMU RISC-V console log from the beginning to the end, to check if there is any message to inform when the server is ready to accept connections?

fanghuaqi, 2020-11-17:
Thanks. Below is the command I used.

~~~
E:\qemu/qemu-system-riscv32.exe -machine gd32vf103_rvstar -serial stdio -nographic -kernel C:\Users\11653\NucleiStudio_workspace\1111\Debug\1111.elf -nodefaults -S -s
~~~

The qemu we built is using https://github.com/isrc-cas/plct-qemu/tree/plct-nuclei

Thanks
Huaqi

TommyMurphyTM1234, 2020-11-17:
You didn't provide what @ilg-ul asked for: the output console log/transcript for a run of qemu.

fanghuaqi, 2020-11-17:
Hi, I just opened a terminal in eclipse and type command above to launch qemu and start gdb server(-S -s).

For the qemu, there is no too much output in console, it just output the uart message from the program I run.

Here are the sceenshots:

[image]

This screenshot showed the debug configuration I used, I disabled qemu in this debug configuration, and run qemu in the eclipse terminal by itself.

[image]

This screenshot showed that I debugged the application by connecting to localhost:1234 which started by qemu debugger server.

Thanks
Huaqi

ilg-ul, 2020-11-17:
> For the qemu, there is no too much output in console

As long as there is no way to convince QEMU to output a message when the GDB server is ready to accept commands, it is not realistic to expect for an Eclipse plug-in to know when to start the GDB client; that's the reason why full integration is possible only with a customised QEMU.

TommyMurphyTM1234, 2020-11-17:
Anything useful here?

https://unix.stackexchange.com/questions/20847/can-libvirt-check-if-virtual-machine-is-fully-booted-can-another-tool

Or maybe this is beyond the scope or capabilities of the plugin?

ilg-ul, 2020-11-17:
Thank you, Tommy, but how exactly do you think we can use this? 

The problem is that if the server is not yet ready to accept TCP connections, the GDB client will enter a TCP timeout & retry cycle, which takes like forever for the waiting user. We can add a small fixed delay, but how long should it be? It depends on the virtualised device/board, thus it must be configurable, which makes things even more flimsy. 

Many years ago when I asked the upstream maintainers to add a message after the GSB server starts, they refused, thus I had to make the current fork.

fanghuaqi, 2020-11-17:
According to my simple test, it seems that the qemu started quickly, maybe a simple configurable delay is enough, how did you handle this for openocd + gdb case?

laomaolaile, 2020-11-17:
After QEMU is started, what should be returned? What kind of signal do you have to receive here to continue？

ilg-ul, 2020-11-17:
> how did you handle this for openocd + gdb case?

openocd accepts a command line option to pass commands, and I pass an `echo` command followed by a string, that I 'hunt' later in the plug-in.

> it seems that the qemu started quickly

Yes, for your simple board and your (probably) fast computer.

ilg-ul, 2020-11-17:
> After QEMU is started, what should be returned? What kind of signal do you have to receive here to continue？

Correction, the message should not be displayed after QEMU is started, but **after** the GDB server is ready to accept connections. Any message is fine.