eclipse-embed-cdt/eclipse-plugins #356: Breakpoints in C source doesn't seem to stick when debugging Windows

AHSauge, 2019-09-10:
I am in the process of upgrading the toolchain and I've hit what I believe to be a bug in the Eclipse plugin or in the communication with the associated parts around it.

The device in question is: AT91SAM9G20

Old setup:
Kepler SR2 (Build id: 20140224-0627)
GNU ARM C/C++ Cross Compiler Support 1.8.1.201402180751
GNU Tools for ARM Embedded Processors 7.4.1.20130913

New setup:
Eclipse IDE 2019‑06 (Build id: 20190614-1200)
GNU MCU C/C++ ARM Cross Compiler 2.6.4.201901011632
GNU MCU C/C++ J-Link Debugging 4.3.4.201901011632
GNU Tools for ARM Embedded Processors 7.4.1.20130913

The old setup was using J-Link v4.90. The new one has been tested with v4.90, v6.22a and v6.50.

So the old setup has of course worked for years with no issues. However, after upgrading I no longer am able to set breakpoints in the C code. It will be shown as a breakpoint, but not set on the device (no check mark and no error indicated in the icon). If I press "Show breakpoints supported by selected target" in the breakpoints view, it disappears. They reappear if I deselect it. If I manually set breakpoints in the Debugger Console, it works. If I go in the disassembler view and set breakpoints on the assembly, it works too! If I step into/over/return it all works fine.

I believe this from the Debugger Console might be on to something:
```
GNU gdb (GNU Tools for ARM Embedded Processors) 7.4.1.20130913-cvs
Copyright (C) 2012 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "--host=i586-mingw32 --target=arm-none-eabi".
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Warning: the current language does not match this frame.
No source file named C:\\temp\\eclipse-test\\FirstProject\\src\\main.c
```
The breakpoint I'm trying to set is indeed in main.c, and if one ignores the double \\ the path is correct. 
For the record, this works fine on Linux with identical versions of the items listed above.
This leads me to believe there is something Windows specific about this, possibly related to paths. 

Are there any way for me to get a more verbose output to know what GDB is actually started with and how the breakpoint is sent to GDB? It would be nice to rule out some kind of issue with the path provided.
As for the warning: I have tried "show language" in the debugger console, and that gives back
```
The current source language is "auto; currently c".
```
That is admittedly a bit contradictory to the warning it provides earlier. It also makes little sense since when I try to set additional breakpoints, it will spit out an additional
```
No source file named C:\\temp\\eclipse-test\\FirstProject\\src\\main.c.
```

So it seems to be is some issue with resolving the path when the breakpoint is set from source, but I'm not able to pin it down further than this at the moment.

AHSauge, 2019-09-10:
Just some additional information.

I tried to set some break points in the debugger console in Eclipse and that also works. However, note the warning about no source. Both breakpoint worked, but neither were inserted into Eclipse with the correct indication and both disappeared when using the "Show breakpoints supported by selected target" button
```
b ..\src\main.c:55
Breakpoint 12 at 0x20006bc0: file ../src/main.c, line 55.
No source file named C:\\temp\\eclipse-test\\FirstProject\\src\\main.c.
b ../src/main.c:57
Breakpoint 14 at 0x20006bc4: file ../src/main.c, line 57.
No source file named C:\\temp\\eclipse-test\\FirstProject\\src\\main.c.
```

ilg-ul, 2019-09-10:
managing paths on windows was always a challenge, so I'm not surprised you encountered such problems.

on the other side, the toolchain you use is very, very old. please try again using the latest version (and the latest version of the plug-ins) and let me know it it makes any difference.

TM1234, 2019-09-10:
> Are there any way for me to get a more verbose output to know what GDB is actually started with and how the breakpoint is sent to GDB? 

What about the gdb trace view which should show everything that is being sent by eclipse/plugins to gdb?

AHSauge, 2019-09-10:
> managing paths on windows has always been a challenge, so I'm not surprised you encountered such problems.
> 
Yes, as it turns out this is an underlying issue with GDB. I have essentially re-discovered this issue: https://answers.launchpad.net/gcc-arm-embedded/+question/240419
I am unsure what has changed between the versions to introduce a use of absolute paths from Eclipse, but I can only assume that's the case here. Doing an objdump as suggested in the link shows me relative paths to the source. I tried manually to set additional breakpoints and there seems to be absolutely path resolving at all in GDB. Unfortunately some of the source seems to be built with "src/./Foobar.c" as the name, and it would seem that it has to match exactly when setting the breakpoint. Thus even "src/Foobar.c" doesn't work.

> on the other side, the toolchain you use is very, very old. please try again using the latest version (and the latest version of the plug-ins) and let me know it it makes any difference.

I'm well aware of the age of the toolchain. I would very much like to upgrade it, so I will test it out. However, upgrading the toolchain is not my decision to make.

AHSauge, 2019-09-10:
> 
> 
> > Are there any way for me to get a more verbose output to know what GDB is actually started with and how the breakpoint is sent to GDB?
> 
> What about the gdb trace view which should show everything that is being sent by eclipse/plugins to gdb?

Ah, I was not aware of that. I will try that out.
Thanks!

ilg-ul, 2019-09-10:
there were several attempts to fix Windows paths in the GDB sources, sometimes they were fortunate, sometimes not.

I understand that the decision to upgrade the toolchain is not yours, but I would appreciate if you could check if this problem is still present when using the latest toolchain. if it is, I would need a way to reproduce it.

AHSauge, 2019-09-10:
> there were several attempts to fix Windows paths in the GDB sources, sometimes they were fortunate, sometimes not.
>
> I understand that the decision to upgrade the toolchain is not yours, but I would [...]

I can confirm that this works fine on GNU Arm Embedded Toolchain 7-2018-q2 
(I got "Couldn't determine a path for the index cache directory." with 8-2019-q3-update and others have identical issues with 8-2018-q4-major, hence the chosen version).

ilg-ul, 2019-09-10:
Ok, so GDB still has a problem with paths (not a surprise).

But do you think there is anything (reasonable) to be done in Eclipse to avoid these bugs? 

Did you get any more info from the trace logs, as suggested by Tommy?

ilg-ul, 2019-09-10:
BTW, the general recommendation with GNU tools and Eclipse, on Windows, is to use Posix paths, not Windows paths. I am aware that sometimes this is difficult.

AHSauge, 2019-09-10:
> 
> 
> Ok, so GDB still has a problem with Windows paths (not a surprise).
> 

Sorry, maybe I didn't make it clear. It works fine with GNU Arm Embedded Toolchain 7-2018-q2. Breakpoints and everything works there. Now I just need to convince the necessary people to upgrade the toolchain.

However, if I choose a newer version GDB prints out "Couldn't determine a path for the index cache directory." during start-up and then GDB shuts down fairly soon after that. I believe that is a separate issue that I am far from certain is caused by Eclipse or any plugins.
Also, I re-tried the latest version (8-2019-q3-update), and now it suddenly works even though the warning is there. So it is very weird and I wouldn't pay much attention to that unless it's reproduced.
My conclusion therefore is there must be something else going on that I need to track down if the issue reappears.

>
> But do you think there is anything (reasonable) to be done in Eclipse to avoid these bugs?
> 

It's a bit hard to tell. It works with a new version. I don't know how stable the code for paths in gdb is to really provide any input here. If it has been rewritten and is now in a fixed state, then there isn't much to be done. I certainly don't expect anyone to really fix bugs for a 6(?) year old version of a toolchain.

> Did you get any more info from the trace logs, as suggested by Tommy?

I tried, but it gives back "Tracing is not supported" in the "Trace control" view. I'm not sure why that occurs. I'll have to look more into that if I am to provide anything from that unfortunately.

TM1234, 2019-09-10:
> I tried, but it gives back "Tracing is not supported" in the "Trace control" view.

Not the trace control view - I meant the GDB trace view/console.
But using a recent version of the GME plugins I can't seem to see where to enable this myself...
It used to be one of the buttons where the Debugger Console etc. appear.

TM1234, 2019-09-10:
Ah - https://www.eclipse.org/lists/cdt-dev/msg32184.html

ilg-ul, 2019-09-10:
that's correct, it is not an issue with the plug-ins but with CDT defaults. you can enable the gdb trace window somewhere in the preferences > c/c++ > debug.

ilg-ul, 2019-09-10:
> I don't know how stable the code for paths in gdb is 

it is hard to tell.

the version used in 8-2018 was notoriously problematic. in 8-2019 it was somehow improved, but since you still have problems, it is probably not yet stable.

TM1234, 2019-09-10:
> it is not an issue with the plug-ins but with CDT defaults.

Of course - didn't mean to imply that it was an issue with the GME plugins.
I posted the link that explains how to enable the gdb trace view above.
You might also want to increase the buffer size as the gdb trace text can grow quite large.

ilg-ul, 2019-09-29:
Any conclusion?

sedlacekdavid, 2019-12-29:
Having the same issue :(

AHSauge, 2020-01-14:
Hi again
A colleague of mine provided some helpful insight. Turns out there is a sneaky "target remote localhost:2331" in the startup script tab. This is not needed, and is in fact causing issues. For GDB up to and including "7 2018-q2-update" it worked fine somehow, but from "8 2019-q3-update" (or the release before that) and onwards it's causing issues. Removing this line in the startup seems to fix the issues I was seeing with newer versions of GDB. I can so far reliably debug without any issues.

ilg-ul, 2020-01-14:
Great!

Good luck with your projects.

If the issue is solved, please close this ticket.

closed 2020-01-14 by AHSauge

TM1234, 2020-01-14:
Isn't port 2331 the default port for the J-Link remote serial protocol server?

AHSauge, 2020-01-14:
@TM1234 Correct. I would assume you're right about the second attempt failing gracefully in older versions.