eclipse-embed-cdt/eclipse-plugins #363: plugin was not compatible with eclipse v4.13.0

zerobikappa, 2019-09-21:
eclipse version: 2019-09 R (4.13.0)
eclipse build id: 20190917-1200
java version: openjdk-12.0.2.u10-1

operating system: 
Operating System: Arch Linux 
KDE Plasma Version: 5.16.5
KDE Frameworks Version: 5.62.0
Qt Version: 5.13.1
Kernel Version: 5.3.0-arch1-1-ARCH
OS Type: 64-bit

toolchain version: arm-none-eabi-gcc 9.2.0-1

ilg-ul, 2019-09-21:
can you be more specific on the error you get?

ilg-ul, 2019-09-21:
I could not replicate your issue.

I just installed the latest version of the plug-ins (the pre-release of v4.6.1) on the macOS Eclipse 2019-09 R and I was able to create a project using the F4 template, build and debug using qemu.

zerobikappa, 2019-09-22:
> I could not reproduce your issue.
> 
> I just installed the latest version of the plug-ins (the pre-release of v4.6.1) on the macOS Eclipse 2019-09 R and I was able to create a project using the F4 template, build and debug using qemu.

Thanks a lot for your great hint. I found the additional details of plugin in marketplace did not include the latest eclipse version number "2019-09 (4.13)". As a result, the marketplace client rejected to install plugin with an error message "the following solutions are not compatible with this version of eclipse".

As a workaround, I added the update site "http://gnu-mcu-eclipse.netlify.com/v4-neon-updates" and manually installed the plugin.

zerobikappa, 2019-09-22:
Duplicate of #228

closed 2019-09-22 by zerobikappa

ilg-ul, 2019-09-22:
I added 2019-09 to the list of supported versions on marketplace, now it should work on that route too.