eclipse-embed-cdt/eclipse-plugins #365: Plugins not available for MCUXpresso IDE 11.0.1

miguel9554, 2019-10-16:
The newest version of MCUXpresso IDE (11.0.1) is not able to install plugins from http://gnuarmeclipse.sourceforge.net/updates. The IDE gives the following error:

```
An error occurred while collecting items to be installed
session context was:(profile=DefaultProfile, phase=org.eclipse.equinox.internal.p2.engine.phases.Collect, operand=, action=).
No repository found containing: binary,MCUXpressoIDE.executable.gtk.linux.x86_64,11.0.0.201908271452
```

Previous version 11.0.0 allows it with no problem. I don't know if it's a bad backward compatibility implemented by NXP or a bad configuration on the sourceforge server

ilg-ul, 2019-10-17:
Are you sure that this is a GME plug-ins bug?

miguel9554, 2019-10-26:
I solved the issue, it was a problem with the IDE. Thanks for answering, I will change the install site for future installs.

closed 2019-10-26 by miguel9554

ilg-ul, 2019-10-26:
> it was a problem with the IDE

Could you document the issue, and the solutions, to help others encountering the same problem?