eclipse-embed-cdt/eclipse-plugins #366: [OpenOCD][GDB] Cannot debug if entry point not at start of flash

bfiset, 2019-10-31:
### Description

I have a project that make use of a custom bootloader. Since the bootloader is placed at the start of the flash to intercept reset vector, the main application is offset farther in flash.

If there is no bootloader present on flash, debugging session with only the main application starts only without Pre-run/Restart reset enabled. If enabled, execution stalls at 0x0 (due to nothing programmed at start of flash. Also, without Pre-run/Restart reset enabled, the "Restart" button in eclipse isn't working.

It looks like OpenOCD or GDB isn't taking into account the entry point of the firmware's elf.

Using STM32l115RET6 with ST-Link v2.

### Steps to Reproduce

1. Build program with entry point not at reset vector
2. Launch openocd/gdb debug session with Pre-run/Restart reset enabled
3. Pause execution

**Expected behaviour:** 

Pausing execution somewhere in the code

**Actual behaviour:**

Execution is stalled at 0x0 because elf entry point isn't at start of flash.

### Versions

* [plug-in version]
2.7.1.201909231407
* [Eclipse version]
2019-03 (4.11.0)
* [Java version]
Java(TM) SE Runtime Environment (build 1.8.0_201-b09)
* [operating system]
Xubuntu 18.04 amd64, kernel 5.0.0-32-generic
* [toolchain version]
binutils 2.30-21ubuntu1~18.04.2
GNU Make 4.1

### OpenOCD logs

```
GNU MCU Eclipse 64-bit Open On-Chip Debugger 0.10.0+dev-00462-gdd1d90111 (2019-01-18-11:37)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
adapter speed: 300 kHz
adapter_nsrst_delay: 100
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
none separate
srst_only separate srst_nogate srst_open_drain connect_assert_srst
debug_level: 2
Started by GNU MCU Eclipse
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : clock speed 300 kHz
Info : STLINK V2J29S7 (API v2) VID:PID 0483:3748
Info : Target voltage: 2.927559
Info : stm32l1.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : Listening on port 3333 for gdb connections
Info : accepting 'gdb' connection on tcp/3333
Info : Halt timed out, wake up GDB.
Error: timed out while waiting for target halted

Info : Device: STM32L1xx (Cat.5/Cat.6)
Warn : STM32L flash size failed, probe inaccurate - assuming 512k flash
Info : STM32L flash has dual banks. Bank (0) size is 256kb, base address is 0x8000000
Warn : negative reply, retrying
Warn : negative reply, retrying
Info : Unable to match requested speed 300 kHz, using 240 kHz
Info : Unable to match requested speed 300 kHz, using 240 kHz
adapter speed: 240 kHz
target halted due to debug-request, current mode: Thread 
xPSR: 00000000 pc: 00000000 msp: 00000000
STM32L: Enabling HSI
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
adapter speed: 1800 kHz
semihosting is enabled
Info : Unable to match requested speed 300 kHz, using 240 kHz
Info : Unable to match requested speed 300 kHz, using 240 kHz
adapter speed: 240 kHz
target halted due to debug-request, current mode: Thread 
xPSR: 00000000 pc: 00000000 msp: 00000000, semihosting
STM32L: Enabling HSI
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
adapter speed: 1800 kHz
target halted due to breakpoint, current mode: Thread 
xPSR: 0x61000000 pc: 0x2000000e msp: 00000000, semihosting
target halted due to breakpoint, current mode: Thread 
xPSR: 0x61000000 pc: 0x2000000e msp: 00000000, semihosting
target halted due to breakpoint, current mode: Thread 
xPSR: 0x61000000 pc: 0x2000000e msp: 00000000, semihosting
target halted due to breakpoint, current mode: Thread 
xPSR: 0x61000000 pc: 0x2000000e msp: 00000000, semihosting
target halted due to breakpoint, current mode: Thread 
xPSR: 0x61000000 pc: 0x2000000e msp: 00000000, semihosting
target halted due to breakpoint, current mode: Thread 
xPSR: 0x61000000 pc: 0x2000000e msp: 00000000, semihosting
target halted due to breakpoint, current mode: Thread 
xPSR: 0x61000000 pc: 0x2000000e msp: 00000000, semihosting
target halted due to breakpoint, current mode: Thread 
xPSR: 0x61000000 pc: 0x2000000e msp: 00000000, semihosting
target halted due to breakpoint, current mode: Thread 
xPSR: 0x61000000 pc: 0x2000000e msp: 00000000, semihosting
target halted due to breakpoint, current mode: Thread 
xPSR: 0x61000000 pc: 0x2000000e msp: 00000000, semihosting
Warn : keep_alive() was not invoked in the 1000ms timelimit. GDB alive packet not sent! (5795). Workaround: increase "set remotetimeout" in GDB
Info : Unable to match requested speed 300 kHz, using 240 kHz
Info : Unable to match requested speed 300 kHz, using 240 kHz
adapter speed: 240 kHz
target halted due to debug-request, current mode: Thread 
xPSR: 00000000 pc: 00000000 msp: 00000000, semihosting
Info : Unable to match requested speed 300 kHz, using 240 kHz
Info : Unable to match requested speed 300 kHz, using 240 kHz
adapter speed: 240 kHz
target halted due to debug-request, current mode: Thread 
xPSR: 00000000 pc: 00000000 msp: 00000000, semihosting
===== arm v7m registers
(0) r0 (/32): 0x00000000
(1) r1 (/32): 0x00000000
(2) r2 (/32): 0x00000000
(3) r3 (/32): 0x00000000
(4) r4 (/32): 0x00000000
(5) r5 (/32): 0x00000000
(6) r6 (/32): 0x00000000
(7) r7 (/32): 0x00000000
(8) r8 (/32): 0x00000000
(9) r9 (/32): 0x00000000
(10) r10 (/32): 0x00000000
(11) r11 (/32): 0x00000000
(12) r12 (/32): 0x00000000
(13) sp (/32): 0x00000000
(14) lr (/32): 0xFFFFFFFF
(15) pc (/32): 0x00000000
(16) xPSR (/32): 0x00000000
(17) msp (/32): 0x00000000
(18) psp (/32): 0x00000000
(19) primask (/1): 0x00
(20) basepri (/8): 0x00
(21) faultmask (/1): 0x00
(22) control (/2): 0x00
===== Cortex-M DWT registers
```

TM1234, 2019-10-31:
Try this without using the GME plugins at all - i.e. use openocd and gdb from the command line.
If the problem persists there then it's definitely not a problem with the GME plugins.
And it may simply be an issue with how your program/debug session is set up rather than with the tools at all.

TM1234, 2019-10-31:
BTW - if you want to debug a program with no bootloader at 0x00000000 and you are loading and running your program from a higher address then you should be programming the Vector Table Offset Register (VTOR) to point at the actual program's vector table which is at 0x08008000 in the absence of a bootloader. And something will need to load the initial SP and PC from 0x08008000 and 0x08008004 respectively. Progam load should load the PC but I can't recall if you need to manually init the SP.

TM1234, 2019-10-31:
If in the absence of a bootloader at 0x00000000 (i.e. that memory is erased to 1's or 0's or contains garbage) then if a debugger driven reset of the target is causing execution to start there then all bets are off since it will be executing an invalid program and will end up in a fault, double fault and then lockup.

ilg-ul, 2019-10-31:
I concur with Tommy's suggestion, try this without using the GME plugins at all - i.e. use openocd and gdb from the command line.

However, in your use case, I'm not sure what you expect. Do you expect the debug session to take into account the ELF entry point and start the debug session from there? I would say it does, but by enabling the **Pre-run reset** you ask the debugger to issue a hardware reset and this simply overrides the entry point.

Anyway, the idea of debugging devices with code starting at a higher address and leaving the bootloader space empty is very risky.

In case you do not know exactly how things work, here are some details.

To start a debug session, first the plug-ins ask gdb which asks openocd to issue a reset. This is equivalent with pressing the reset button. Immediately after this, the plug-in sends to gdb which sends to openocd a command to halt; openocd starts to talk via the JTAG protocol with the device and asks the JTAG logic to halt execution. Ideally halt should occur before executing the first instruction, which is a hard wired jump to the Reset vector, but due to the limited speed of the logic inside the plug-ins, inside gdb, inside openocd and especially the speed of the communication via JTAG, which is initially set to a low speed, it is possible that the core will start to execute something, and if you leave the flash empty, you may get unexpected behaviour.

For most cases this is not relevant, the core may be in any state, but the JTAG interface should always be functional and it will be able to write the flash, and possibly to set any breakpoints in the debug circuitry, which is separated from the core.

Then comes the second reset (the Pre-run reset), which resets again the core, but this time with the flash expectedly valid, and the breakpoints set, so if you have a breakpoint at the Reset_Handler, it will trigger.

As far as I remember (sorry if I miss something) the ELF entry point is used by the debugger to set the PC before continuing after the initial reset. 

If you do not enable the Pre-run reset, and the debugger was fast enough to halt the core before triggering an exception, this entry point will behave as a jump that will be executed, regardless of the address of the Reset_Handler.

If you enable the Pre-run reset, this will be executed, and the address of the Reset_Handler will be used, regardless of the ELF entry point.

Given the above, I suggest you reconsider the use case, and, if you agree that this is not a bug in the plug-ins, close this ticket.

For any other questions and support, please use the project forum.

bfiset, 2019-11-01:
Thanks for the lengthy explanation. I must admit inner workings of JTAG debugging are still a mystery to me.

Anyway, what I can grasp of all this is it's not possible to reliably debug a program on MCU if start of flash isn't correctly populated; using OpenOCD/GDB combo at least.

I was under the impression it was possible. I recently migrated a project from CrossWorks and their debugging suite will make it work. Flash between 0x08000000 and 0x08007FFF is empty and I can succesfully start a debug session reliably and reset execution to start of program. I thought the same could be possible with OpenOCD/GDB.

closed 2019-11-01 by bfiset

ilg-ul, 2019-11-01:
> CrossWorks

I don't know how debugging in CrossWorks work, it is probably fast enough to halt the core immediately after reset.

But from your message I understood that with the secondary reset disabled it also works with OpenOCD. And, in case the debugger does not use the ELF entry point, you can always set it by hand.