eclipse-embed-cdt/eclipse-plugins #367: Path display on Windows is not consistent

ilg-ul, 2019-11-06:
While updating the CMSIS Packs, the console displays some paths with POSIX syntax, and some with Windows syntax.

An initial check showed that the code correctly uses `getCanonicalPath()` which should generate the POSIX path, but, for unknown reasons, sometimes this does not work.

TM1234, 2019-11-06:
Apologies if this might be slightly off topic but isn't there a strong argument for making "/" the default path separator regardless of OS since it's compatible/portable across Windows, linuxes, Mac?

ilg-ul, 2019-11-06:
Yes, during the latest changes to improve the CMSIS Pack updates, I also updated all code displaying lines on the console to using `getCanonicalPath()`, expecting to see only nice POSIX paths in the output.

For reasons beyond my understanding, some lines still show back-slashes, and I'm a bit confused by this, I don't know what I did wrong.

Can you confirm that this happens on your Eclipses too?