eclipse-embed-cdt/eclipse-plugins #368: Documentation: Adding new template

ajaybhargav, 2020-01-04:
I want to create a new template for my custom hardware based on ARM processor. Is there any documentation available on how to get started to use the existing plugin and add a new template to it as an add-on. I could not find much about build environment setup etc. I did install maven (m2e), JDK but I don't know what to do next.

I may sound silly as I am not a java/eclipse developer but I hope given the right steps I can create template for my board. Thanks in advance.

ilg-ul, 2020-01-04:
You can take a look at the STM32 template project, for example the F4:

https://github.com/gnu-mcu-eclipse/eclipse-plugins/blob/develop/bundles/ilg.gnumcueclipse.templates.stm/templates/stm32f4xx_exe_c_project/template.xml

ajaybhargav, 2020-01-04:
Thanks for response @ilg-ul I have already seen the sources. I understand what to do but I just dont know how to build or package the template **only**. So all I want is If I have gnu-mcu-eclipse plugin installed, I can just create template without need to download and building the whole plugin again. Am I getting right?

ilg-ul, 2020-01-04:
>  I can just create template without need to download and building the whole plugin again. 

Nope. For the template to work, it needs to be fully described to Eclipse, which means a proper Eclipse project must exist. And if you want to be able to install the template from an Eclipse update site, you must also define a feature project, and a repository project.

So you need at least three Eclipse projects. There are maven tasks to build them, but it is still quite a lot of work to put things together.

I do not know of any documentation apart from the CDT source code, which I also used to write my plug-ins.

ajaybhargav, 2020-01-04:
I do not want to create my own plugin at all. I do not believe in reinventing the wheel 😄 so maybe an easy way could be to just build your plugin with added project template for my board. Can you please guide to setup build environment? Currently I have installed following:
1. Eclipse PDE (marketplace)
2. M2E (http://download.eclipse.org/technology/m2e/releases/)
3. JDK
4. Project imported in eclipse
[image]

There was an error for Base API setup which I fixed (I believe 😄) Now lot of warnings
1. Description	Resource	Path	Location	Type
'Automatic-Module-Name' header is required to be discoverable as a Java module in a Java modular setup (Java 9 and later)	MANIFEST.MF	/ilg.gnumcueclipse.codered/META-INF	line 1	Plug-in Problem
2. Description	Resource	Path	Location	Type
A plug-in manifest must contain at least one extension or extension point	plugin.xml	/ilg.gnumcueclipse.debug.packs	line 1	Plug-in Problem
3. Description	Resource	Path	Location	Type
CompilerCommandLineGenerator extends non-API type ManagedCommandLineGenerator	CompilerCommandLineGenerator.java	/ilg.gnumcueclipse.managedbuild.cross/src/ilg/gnumcueclipse/managedbuild/cross	line 9	API Usage Problem

and so on...  I am really sorry that you are dealing with a complete noob here. Maybe this will help others too like me to get started with this.

ilg-ul, 2020-01-04:
> I do not want to create my own plugin at all.

Then I miss the point.

> Can you please guide to setup build environment? 

https://gnu-mcu-eclipse.github.io/developer/#how-to-build

ajaybhargav, 2020-01-04:
Wow how did i miss that. Thanks again i will go through it and maybe bug you again if I stumble.

ilg-ul, 2020-01-04:
Please use the project forum for further support.

https://www.tapatalk.com/groups/xpack/

closed 2020-01-04 by ilg-ul