eclipse-embed-cdt/eclipse-plugins #370: Prepare for Eclipse legal review checklist

jonahgraham, 2020-01-09:
There are a number of items that need to be done to the code base to prepare it for the legal review. Some of these items may be deferred until after legal review

- [ ] Update to EPLv2
- [ ] Naming Conventions followed
- [ ] Provider information is set to the project’s formal name:
- [ ] Create CQs for dependencies aka third-party content
- [ ] All source files include a copyright and license header;
- [ ] CONTRIBUTING file (or equivalent) in the root;
- [ ] License file and Notice file in root of repo
- [ ] Apply https://www.eclipse.org/projects/handbook/#legaldoc-plugins

Items that can probably be delayed until after review & checkin, but must be done before first release

- [ ] [Incubation branding](https://www.eclipse.org/projects/handbook/#starting-project-phases) is included on distributed artifacts
- [ ] All download artifacts are (if technical feasible) [signed](https://wiki.eclipse.org/IT_Infrastructure_Doc#Sign_my_plugins.2FZIP_files.3F)

Sources for the above, there are other items in the handbook that need the be followed, but not included in the above because they are action items (e.g. "Project content (code and documentation) does not violate trademarks owned by other organizations")
- https://www.eclipse.org/projects/handbook/#checklist
- https://www.eclipse.org/projects/handbook/#legaldoc
- https://www.eclipse.org/projects/handbook/#legaldoc-plugins

ilg-ul, 2020-01-10:
I checked all files for copyright and license, and I discovered that the `bundles/ilg.gnumcueclipse.core/src/com/github/zafarkhaja/semver` files are third party files, licensed under MIT.

Probably these files should not be included in the initial review, but as separate CQs.

There are also several .jar file in `ilg.gnumcueclipse.core`, that I have no idea what licenses use.

ilg-ul, 2020-01-10:
I updated the feature provider name to Eclipse Embedded CDT

ilg-ul, 2020-01-10:
I updated the feature names to start with Embedded C/C++ instead of GNU MCU C/C++

ilg-ul, 2020-01-10:
License file and Notice file in root of repo checked.

ilg-ul, 2020-01-10:
All kind of details addressed; features and plug-ins should have the required license content.

One question: should I rename the `bundles` folder to `plugins`?

ilg-ul, 2020-01-10:
A preview release can be installed from:

- http://gnu-mcu-eclipse.netlify.com/v4-neon-updates-experimental

The major number was bumped from 4 to 5, to mark the migration.

@jonahgraham, from my point of view we are ready for the review. Apart from the package names and ids that cannot be changed without breaking compatibility, if you identify other details that need to be fixed, please let me know.

jonahgraham, 2020-01-11:
> There are also several .jar file in `ilg.gnumcueclipse.core`, that I have no idea what licenses use.

The ones that are already part of [Orbit](https://download.eclipse.org/tools/orbit/downloads/drops/R20191126223242/) will be easier/faster to use as you only need a piggy-back CQ. A quick looks sees that most of the jars in there are already in Orbit, but not the same versions.

I would be happy to help submit the CQs for this and to adapt the build code to use Orbit

> I discovered that the `bundles/ilg.gnumcueclipse.core/src/com/github/zafarkhaja/semver` files are third party files, licensed under MIT.
> 
> Probably these files should not be included in the initial review, but as separate CQs.

Yup - we can do a CQ for this code too and add it to orbit. Current (last updated 5 years ago) URL is https://github.com/zafarkhaja/jsemver

jonahgraham, 2020-01-11:
> License file and Notice file in root of repo checked.

Once all the CQs are submitted the Notice file can be auto generated from https://www.eclipse.org/projects/tools/documentation.php?id=iot.embed-cdt

jonahgraham, 2020-01-11:
It looks good to me. You can submit the CQ when you are ready - it can be updated if the name change becomes required (https://projects.eclipse.org/projects/iot.ecdt/cq/create) -- Note that some of the PMI (Project Management Interface) still has project name as iot.ecdt. Hopefully that will be sorted soon. Let me know if I can assist with the CQ.

ilg-ul, 2020-01-11:
> The ones that are already part of Orbit will be easier/faster to use as you only need a piggy-back CQ. A quick looks sees that most of the jars in there are already in Orbit, but not the same versions.

Should we adjust the code to use the Orbit libs first?

I'm not familiar with Orbit at all, so any help will be appreciated.

There is also a library that I forked and patched: 

- https://github.com/gnu-mcu-eclipse/Liqp

jonahgraham, 2020-01-11:
> Should we adjust the code to use the Orbit libs first?

No need to do it first. The ip team aren't really checking your code compiles, just licenses and that you haven't copied code from anywhere (i.e. That licenses and copyright actually match code provenance)

I believe you will get parallel ip (see handbook for details) which means you will be on hold for a short time only.

ilg-ul, 2020-01-11:
> that you haven't copied code from anywhere

Well, I copied heavily from CDT ;-) ... I hope I did not mess the copyright headers... 

If I got it right, I have to make a snapshot of the current repo without

- the jars
- the zafarkhaja/semver classes
- the submodules

pack it as an archive and submit it via the url you provided.

Planned for Monday.

jonahgraham, 2020-01-11:
> > that you haven't copied code from anywhere
> 
> Well, I copied heavily from CDT ;-) ... I hope I did not mess the copyright headers...

Yup. Premature end of sentence on my part. 
> 
> If I got it right, I have to make a snapshot of the current repo without
> 
> * the jars
> * the zafarkhaja/semver classes
> * the submodules
> 
> pack it as an archive and submit it via the url you provided.
> 
> Planned for Monday.

Wonderful!

ilg-ul, 2020-01-13:
I added '(incubation)' to feature/plugins/p2 names, since the documentation mentioned that this allows to get parallel ip.

Then done this:

```bash
cd $HOME/tmp
mkdir embed-cdt
cd embed-cdt
git clone -b embed-cdt https://github.com/gnu-mcu-eclipse/eclipse-plugins.git eclipse-plugins.git

cd eclipse-plugins.git

rm -rf .git .gitattributes .github .gitignore .gitmodules .travis.yml 

rm plugins/ilg.gnumcueclipse.core/lib/liqp-0.6.8.jar
rm plugins/ilg.gnumcueclipse.core/lib/antlr-runtime-3.5.1.jar
rm plugins/ilg.gnumcueclipse.core/lib/jackson-databind-2.2.3.jar
rm plugins/ilg.gnumcueclipse.core/lib/jackson-core-2.2.3.jar
rm plugins/ilg.gnumcueclipse.core/lib/json-simple-1.1.1.jar
rm plugins/ilg.gnumcueclipse.core/lib/ST4-4.0.7.jar
rm plugins/ilg.gnumcueclipse.core/lib/jackson-annotations-2.2.3.jar

rm -rf plugins/ilg.gnumcueclipse.core/src/com/github/zafarkhaja/semver

cd $HOME/tmp

tar czf embed-cdt.tgz embed-cdt 
```

Submitted the request and attached the archive:

https://dev.eclipse.org/ipzilla/show_bug.cgi?id=21415

ilg-ul, 2020-12-07:
All issues addressed, CQs approved, ready for v6.0.0.

closed 2020-12-07 by ilg-ul