eclipse-embed-cdt/eclipse-plugins #372: What’s hard, really hard and really frustrating

duaneellissd, 2020-01-13:
Who exactly is managing the “build system” used in GNU-MCU eclipse stuff? Is that this project, because I find some things utterly horrifyingly hard to get done and make this work.

In my situation I have about 6 applications, that share about 8 common libraries.  I continuously find it hard if not impossible to setup projects

For me, I have a single flat top level directory with 8 “libWHATEVER” directories, and 6 “appWHATEVER” sub directories, this is sort of my ‘workspace’.  Note that I am very open to another configuration or structure but don’t have a good solution that work to build what I need, put another way:

The directory looks like this (this is a simple config)

```
<root> — 
   + .git
   + ProjectDocs
   + app_main
        +include
        +src
   + app_bootloader
        +include
        +src
   + app_factorytest
        +include
        +src
   + libfoo
        +include
        +src
   + libfreertos
        +include
        +src
   + liblwip
        +include
        +src
   + libCOMPANY_A
        +include
        +src
   + libTECHNOLOGY_B
        +include
        +src
   + libIMAGEPROCESSING
        +include
        +src
   + libbsp_modelA
        +include
        +src
   +libbsp_modelB
        +include
        +src
```

When I try to create some means to build & compile static LIBs + APPS, it’s painful really painful  - I must be doing something wrong

**I must be doing something wrong.**

Managed build - works - if your project is 100% self contained - copy all your source code into a few directories and you are done - but that means I have 6 copies of the same library buried in the 6 projects - aka: Clone And Own...  This does not scale very well.

But for me, I have many: -I paths, that are relative to the current project and point “up an over” to the neighbor projects, ie: `-I${ProjPathDir}/../libFOO” 

Using CMake to create GNU-MCU eclipse project files is non-trivial and not very clear. If somebody can point me to this, that would be great ... but I don’t see a solution here.

Using Makefile Projects would be nice, but I need to run on Windows and - well yes MAKE is present - but NONE of the helpful Unix tools are present and/or usable ie: simple things like using awk, sed, grep, perl, or python - is painful on windows because there are so many providers/flavors of these tools (msys, cygwin, git-bash, etc, powershell) - I’d really like to be cross platform (win/mac/linux) this helps with other automation I’d like to have in place.

FYI - The “Freedom Studio” - from SiFIVE is nice - the last time I used it (6 months ago) part (1) it actually builds using MSSYS on windows and not simple MSDOS CMD.EXE and Part (2) they require this so they can run ./configure scripts, and (3) In their last release. I used they added the ability to install more tools into their private MSYS instance, so for example if you need a unix tool on windows you can install it the extra tools :-) it’s not a closed solution

GNU-Eclipse does not have that option - the Makefile solution is purely windows centric and that is all you get. Sigh....

yes, there are other build systems that work in Eclipse, ie: SCons... and Ant and ... you name it.  The further away from Eclipse core things the harder it is for team members to understand how to be productive.

**To illustrate the point**

One of the super hard things is just getting the include directories (ie: compiller “-I/path/to/some/dir” type options ) consistent across (X) library projects with (Y) applications and (Z) configurations ... it is horrifying how hard this is

Take for instance FreeRTOS, it has freertosconfig.h -(LWiP has the same thing) as does MBED-TLS  - each of these are libraries - with configuration files that taylor the library at compile time.

How can I setup the 5 library projects to switch configurations? Damn that’s hard, when I switch from APP(A) to (B) - I really don’t want to edit all the project files that is horrible.

For instance: must modify 6 different “-I” settings for each project (The C/C++/ASM paths TIMES Debug/Release = 3 * 2 = 6),  times 4 libraries + 1 app = means 30 dialog screens, and no means to cut and paste between projects dialog boxes.

There’s another 30 in the -D command line defines,  add another app or two - I’m at over 100 dialog screens to hand edit. This does not scale.

There are another 30 screens to get Warnings and Options consistant.

Oh of course, you can also edit XMLfiles and do this the way Real Men do their job

**What are my other options**

Internal to various companies (ie: Texas Instruments) they have a “eclipse project creator tool” that is custom to their environment

Any suggestions? This is beyond frustrating.

ilg-ul, 2020-01-13:
Are you reporting a bug? 

If so, please provide the required details.
If not, for discussions please use the project forum, as clearly stated in the intro message.

TM1234, 2020-01-13:
With regard to having one or more app projects that depend on/link in multiple libs that are contained in their own projects I don't see what the issue is. You can have all of these as separate projects and then configure the lib projects that each app depends on in each project. Yes each app project will have its own copy of the include directives, link library settings etc but I don't see that as a problem or unexpected? Definitely you don't have to copy/duplicate the library sources into each app project.

ilg-ul, 2020-01-13:
> Who exactly is managing the “build system” used in GNU-MCU eclipse stuff? 

GME uses the managed build system provided by Eclipse CDT, with extensions on how paths to multiple toolchains are handled, and an extensive set of configuration settings.

For the rest of the very pertinent questions, please try to split your current message into separate issues, and let's discuss them in the forum.

TM1234, 2020-01-13:
FWIW I agree that the original query should be split into separate topics (on the forum not here) clearly stating what the individual perceived issues are with clear examples of how things work (or not) versus how one might expect them to work to aid discussion/analysis of the issues. And a more meangful and less emotional title would help too! :-)

TM1234, 2020-01-20:
BTW - even when it's necessary to configure the same preprocessor include dirs and defines in multiple projects the workload can be ameliorated by using the export settings option. E.g. 

1. - right click on your project 
2. - select Export... > C/C++ > C/C++ Project Settings
3. - click Next > 
4. - Select Project and Select Configuration
5. - Select settings to export
6. - Browse and select an output folder/file 
7. - Click Finish
8. - (There seems to be a bug whereby if you just enter the filename in the Export to file field clicking Finish does nothing)
9. - 

Then you can import those settings into other projects.
Unfortunately it looks like you need to export Debug and Release (and any other) build target settings individually.

But this feature mightt be useful in this context.

closed 2020-03-15 by ilg-ul