eclipse-embed-cdt/eclipse-plugins #375: Repositories is initially hidden on Repositories preference page

cdwilson, 2020-01-23:
For some reason, Repositories is always blank when I first launch MCU packages preferences, and then I have to click "Edit" button for it to display the repository line.

[image]

```
Eclipse IDE for GNU ARM & RISC-V C/C++ Developers

Version: 2019-12 (4.14.0)
Build id: 20200115-1949
```

ilg-ul, 2020-01-23:
Could you try with a fresh workspace too?

That version is from Eclipse, the plug-ing have their own versions.

ilg-ul, 2020-01-27:
I cannot reproduce the issue.

I installed a fresh Eclipse from 20200115-1949-gnumcueclipse-4.7.1-2019-12-R-macosx.cocoa.x86_64.tar.gz, with a fresh workspace, went straight to the Preferences page and URL is there.

I suggest you retry in the same conditions. On macOS, be sure you remove the `com.apple.quarantine` attribute, otherwise the behaviour may be unexpected.

cdwilson, 2020-03-15:
I retried with a fresh install and a fresh workspace and everything works as expected (I can't reproduce the issue seen previously).

closed 2020-03-15 by cdwilson

ilg-ul, 2020-03-15:
Ok, glad it works for you now.