eclipse-embed-cdt/eclipse-plugins #379: proposal: SDCC support

talsegev, 2020-05-05:
Hi,

I recently found out that eclipse is missing support for [SDCC](http://sdcc.sourceforge.net). 

In short SDCC is "Small Device C Compiler", AKA C compiler for most 8-bit MCU's.

My proposal is to add support for SDCC. I think It will be a nice addition for this plugin.

Regards,
Tal

ilg-ul, 2020-05-05:
Thank you for taking the time to make suggestions to this project.

As far as I can see in the SDCC home page, it supports mostly 8-bit MCUs, which, in my opinion, are not exactly the future.

talsegev, 2020-05-05:
Hi, 

I'm amazed to find how different we see the market.

I'll start with what I think we do agree on; 
A. _eclipse-embed-cdt_ is for embedded MCUs
B. 8-bit MCUs are still MCUs
C. 8-bits are less than 16-bits which are less than 32-bits etc...

I invite you to google up  "_8 bit microprocessor market share_" and see for your self why I'm so amazed to read your comment "_not exactly the future_". My thoughts are exactly the opposite from yours, I think the 8-bits MCUs are the future, especially in the IoT era where devices are disposable and not fixable.

Regards,
Tal

ilg-ul, 2020-05-05:
If 8 bit is the future, then it won't take long for someone with a better vision than mine will step in and sponsor de development of such a new feature.

flit, 2020-05-05:
From my experience working in the silicon industry for about 17 years with a focus on MCUs, 8-bit devices are most definitely _not_ the future. The vast majority of 8-bit devices sold are old devices being sold into existing products. There is comparatively very little new R&D for 8- or 16-bit devices. A major reason is that the power/performance numbers makes 8-bit devices very unattractive for IoT and a lot of other markets, especially in advanced process nodes ("advanced" for MCUs being 40nm with flash, or 28nm without flash).

TM1234, 2020-05-05:
> A. eclipse-embed-cdt is for embedded MCUs

Pedantic point - we (Microsemi/Microchip), and I suspect others, successfully use the plugins to support our FPGA based soft IP CPU cores which are technically not MCUs. So don't necessarily presume that the target is a "fixed" (e.g. ASIC) MCU. :-)