eclipse-embed-cdt/eclipse-plugins #384: Switch to the "pomless" layout

ilg-ul, 2020-07-18:
From: Alexander Fedorov <alexander.fedorov@arsysop.ru>
Subject: Re: [embed-cdt-dev] GNU MCU Eclipse IDE for C/C++ Developers 2019-12 20200115
Date: 16 January 2020 at 06:27:58 EET
To: Liviu Ionescu <ilg@livius.net>

Hi Liviu,

Please create a ticket(s) with maven issues you have and I'll take a look.
I would recommend to switch to the "[pomless](https://www.vogella.com/tutorials/EclipseTycho/article.html)" layout like [here](https://www.vogella.com/tutorials/EclipseTycho/article.html).

ilg-ul, 2020-07-25:
@ruspl-afed, Alexander, you suggested to switch to pomless layout, and provided a link, but I could not figure out exactly what the required changes are.

Could you give me some clues? If the changes are not very complicated, perhaps we can include them in the initial 5.1.1 release.

ruspl-afed, 2020-07-25:
If you are mostly ready to declare 5.1.1 I would suggest to postpone this one - as this is not a cosmetic change and may require some time to be introduced.
If you have a number of things to do before the release anyway - I can try to help with this one.

ilg-ul, 2020-07-25:
I'm fine with postponing this for the next release.

ilg-ul, 2020-07-25:
In this case I'll freeze 5.1.1 and work on the EPP config to create the new packages.

ilg-ul, 2020-07-31:
Fixed in 2020-07-30 by Alexander.

closed 2020-07-31 by ilg-ul