eclipse-embed-cdt/eclipse-plugins #390: The project icon is represented twice in the About window

ilg-ul, 2020-07-25:
The problem occurred after performing the legal changes required by the migration.

The second icon is associated with the J-Link debug.

File comparisons between the J-Link and OpenOCD features did not reveal any differences. Same for the plug-ins.

There must be something subtle that differentiate the J-Link from OpenOCD (and others).

jld01, 2020-07-25:
Could trailing space chars in *.properties be responsible?

ilg-ul, 2020-07-25:
The definitions in the .properties file were copied from OpenOCD.

ilg-ul, 2020-07-25:
Does anyone know which Eclipse project draws the About window, to check the source code and try to identify what is different in the J-Link feature/plug-in definitions that puts it in a separate category?

ruspl-afed, 2020-07-25:
Does it appear after installing of https://download.eclipse.org/embed-cdt/releases/5.1.1/ upon the "Eclipse for C/C++ Developers" 2020-06 package or what are the steps to reproduce?

ilg-ul, 2020-07-25:
Good question. In my case it showed after installing 5.1.1 over the previous version of the plug-in, with both 2020-03 and 2020-06. I did not try a fresh install, could you confirm?

ruspl-afed, 2020-07-25:
> Does anyone know which Eclipse project draws the About window

https://git.eclipse.org/c/platform/eclipse.platform.ui.git/tree/bundles/org.eclipse.ui.workbench/Eclipse%20UI/org/eclipse/ui/internal/dialogs/AboutDialog.java?id=c1aa656c39e929445847dbe6d180c8b653b8142f#n112

ilg-ul, 2020-07-25:
To be noted that the issue was introduced by the recent changes when we updated the licenses to EPL2 and related metadata, to satisfy the legal requirements for the migration to the Foundation. There were no functional changes to the debug plug-ins.

I tried to compare the repo with the previous version, but could not spot anything unusual.

ilg-ul, 2020-07-25:
Thank you, Alexander, I'll take a look, and, if necessary, install the sources, to run a debug session; but it'll take some time, this is new territory for me; if you spot something faster, please let me know.

ruspl-afed, 2020-07-25:
1. I cannot install from your site running with Java 8
```
Cannot complete the install because some dependencies are not satisfiable
  Software being installed: a.jre.javase 9.0.0
  Software currently installed: C/C++ Meson Build Support - Preview 9.11.1.202004012023 (org.eclipse.cdt.meson.feature.group 9.11.1.202004012023)
  Cannot satisfy dependency:
    From: C/C++ Meson Build Support - Preview 9.11.1.202004012023 (org.eclipse.cdt.meson.feature.group 9.11.1.202004012023)
    To: org.eclipse.equinox.p2.iu; org.eclipse.tools.templates.freemarker.java11 [1.1.100.202001140305,1.1.100.202001140305]
  Cannot satisfy dependency:
    From: Template Engine Freemarker Support (Java 11) 1.1.100.202001140305 (org.eclipse.tools.templates.freemarker.java11 1.1.100.202001140305)
    To: osgi.ee; (&(osgi.ee=JavaSE)(version=11))
```
It may be CDT 2020-06 problem, that should not appear in 2020-09 as it will be Java 11 based
So I switched launcher to Java 11 to install the `ilg.gnumcueclipse.repository-5.1.1-202007251009.zip`

2. I can repeat your problem

3. The only difference that I can see without deep dive is that your branding bundle `ilg.gnumcueclipse.debug.gdbjtag.jlink` does not have the trailing space in provider name while others have it. I mean 
"bundle.vendor=Eclipse Embedded CDT"
vs
"bundle.vendor=Eclipse Embedded CDT "

ilg-ul, 2020-07-25:
> Java 11

Yes, that's a 2020-06 issue, it requires Java 11.

> trailing space

Oops! That's exactly the opposite of what I expected, J-Link is ok, and all other are wrong. :-(

Also to be noted is that the field used for grouping is `bundle.vendor` from the branding plug-in, not the `providerName` from the feature, as I expected.

ruspl-afed, 2020-07-25:
This is what @jld01 suggested to check first, hope it will help to fix the problem.

ilg-ul, 2020-07-25:
> This is what @jld01 suggested to check first

Right, just that I checked the `providerName` in the feature, actually I copied most of the properties from OpenOCD, not the `bundle.vendor` from the branding plug-in. :-(

> hope it will help to fix the problem

It did, I recreated the binaries and now it is ok.

Thank you very much!

And, if you allow me a small suggestion, perhaps the About code could trim the string, to become insensitive to such small inconsistencies.

ilg-ul, 2020-07-25:
Fixed in 2020-07-25.

ruspl-afed, 2020-07-26:
Should it be closed?

ilg-ul, 2020-07-26:
It will be closed when the release is out.

Thank you very much!

closed 2020-07-28 by ilg-ul