eclipse-embed-cdt/eclipse-plugins #397: Generate "source" features that constains "source" bundles

ruspl-afed, 2020-08-01:
Our p2 repository should contain sources for all the bundles. I think this is one of requirements to join SimRel but I cannot find it to reference here.
At least that's what all good Eclipse citizens do.

ilg-ul, 2020-08-01:
Agreed.

ilg-ul, 2020-08-01:
Fixed in 2020-08-01 by Alexander.

closed 2020-08-01 by ilg-ul

ilg-ul, 2020-08-02:
Alexander, what is the purpose of the source packages? Only to provide the source files for the Java classes when debugging?

Because I checked some source .jar files, and they only include the ilg.gnumcueclipse.* folders, and some minimal metadata.

All other files were skipped. When I check in Eclipse the `build.properties` file, the Binary Build lists has most files selected, but the Source Build section has no files selected at all. Is this ok?

ruspl-afed, 2020-08-02:
> Alexander, what is the purpose of the source packages

I think the preimary reason is compliance with EPL-2.0, but also other benefits like source navigation that you already mentioned.

> 3. REQUIREMENTS
> 
> 3.1 If a Contributor Distributes the Program in any form, then:
> 
>     a) the Program must also be made available as Source Code

As for the content of the source build I plan to revisit all the bundles as a part of #402

ilg-ul, 2020-08-02:
Ok, great.

Please note that the SiFive template project has some git submodules, with the code used to create RISC-V projects.