eclipse-embed-cdt/eclipse-plugins #402: Normalize content of the sources bundles

ruspl-afed, 2020-08-02:
From #397 by @ilg-ul 

> ...  I checked some source .jar files, and they only include the ilg.gnumcueclipse.* folders, and some minimal metadata.
> 
> All other files were skipped. When I check in Eclipse the build.properties file, the Binary Build lists has most files selected, but the Source Build section has no files selected at all. Is this ok?

Let's use Eclipse Platform bundles as a sample to normalize the content

ilg-ul, 2020-08-02:
> Let's use Eclipse Platform bundles as a sample to normalize the content

Good plan.

I would also take a look at the CDT projects.

ilg-ul, 2020-08-02:
Please note that the SiFive template project has some git submodules, with the code used to create RISC-V projects.

(Comment moved here from #397).

ruspl-afed, 2020-08-02:
> I would also take a look at the CDT projects.
Yes, to fix it too :)

ilg-ul, 2020-08-03:
#404 is the only change required?

ruspl-afed, 2020-08-03:
It looks so, I walked through all the bundles that I can see.

ilg-ul, 2020-08-03:
Ok, when done, you can close the issue.

I changed the strategy, I no longer keep them open until the actual release, I tag them with a milestone and it is enough to keep track of the changes related to a release.

closed 2020-08-03 by ruspl-afed