eclipse-embed-cdt/eclipse-plugins #405: Proposal: Eclispe Embedded CDT core feature

kolipakakondal, 2020-08-04:
Can we have the core plugin features with the following plugins?

```
ilg.gnumcueclipse.core
ilg.gnumcueclipse.debug.core
lg.gnumcueclipse.debug.gdbjtag
ilg.gnumcueclipse.debug.packs
ilg.gnumcueclipse.managedbuild.cross
ilg.gnumcueclipse.managedbuild.packs
```

So that I should be able to build customized OpenOCD plugin(for example, esp32 OpenOCD) by just installing "eclipse embedded CDT core feature".

ilg-ul, 2020-08-04:
I'm not sure I understand the rationale of your request.

Without this change how would you build your customised plug-in?

ilg-ul, 2020-08-04:
@ruspl-afed, what do you think about having a core feature including the core plug-in?

The only advantage that I can see is that, by having the same version as the core plug-in, it makes the version more visible, otherwise by inspecting an installed configuration, there is no way to tell the version (5.1.1 in our case).

ruspl-afed, 2020-08-04:
Perhaps we use the term "core" in different meanings. 
Generally I do support the separation of headless part. But, currently `ilg.gnumcueclipse.core` contains a lot of UI dependencies.

As feature is used mostly to isolate dependencies - what dependencies do we want to isolate here?

kolipakakondal, 2020-08-04:
I'm sorry if I didn't make it clear enough. Here is what I'm trying to build.

`com.espressif.idf.debug.openocd` is our plugin which is derived from `ilg.gnumcueclipse.debug.gdbjtag.openocd` with customized configuration for OpenOCD debugging.

Here are my plugin dependencies which it has.
 -  ilg.gnumcueclipse.core,
 -   ilg.gnumcueclipse.debug.core,
 -   ilg.gnumcueclipse.debug.gdbjtag,
 -   ilg.gnumcueclipse.managedbuild.cross

Now when I install **Eclipse Embedded CDT OpenOCD** feature and **our customized OpenOCD plugin** - they will see two debug configuration options "GDB OpenOCD Debugging" and "ESP GDB OpenOCD Debugging". This might be confusing to the end-users. To avoid this confusion to the user I want to show only "ESP GDB OpenOCD Debugging" but not the other one. 

I was thinking I could achieve this if there is a way to install only core gnumcueclipse plugins

Thanks

ilg-ul, 2020-08-04:
> Perhaps we use the term "core" in different meanings. ... currently ilg.gnumcueclipse.core contains a lot of UI dependencies.

When I wrote the plug-ins I used the term core to mean common, and I was not very concerned on separating the UI functionality, so I'm not sure what you are trying to achieve is that easy to implement. :-(

ruspl-afed, 2020-08-05:
> I was thinking I could achieve this if there is a way to install only core gnumcueclipse plugins

Why not to declare your own feature that has only the bundles that you need?

For me the feature content that you requested looks specific a bit, so I would ever declare more granular features that can make sense for more use cases or leave it all "as is".

ruspl-afed, 2020-08-05:
> and I was not very concerned on separating the UI functionality, so I'm not sure what you are trying to achieve is that easy to implement.

The main reason to extract headless part is to cover it with the headless unit tests in the pipeline. As a bonus such a separation can enable command line usage of the functionality.

ilg-ul, 2020-08-05:
> The main reason to extract headless part is to cover it with the headless unit tests in the pipeline. As a bonus such a separation can enable command line usage of the functionality.

I see... 

However in this case, the OpenOCD debug plug-in does not offer CLI functionality (those who need CLI will go directly to `gdb`), and there are no unit tests for it, so the separation does not bring much.

We'll probably leave it as it is for now.

kolipakakondal, 2020-08-05:
> Why not to declare your own feature that has only the bundles that you need?
>
> For me the feature content that you requested looks specific a bit, so I would ever declare more granular features that [...]

Thanks for your reply @ilg-ul 

For that, I need to replicate gnumcueclipse plugins in my repo and build a feature. I think from the code maintenance perspective, that's going to be tedious.

Could you please suggest any other way to do this?

Thanks.

ilg-ul, 2020-08-05:
> I need to replicate gnumcueclipse plugins in my repo and build a feature

I thought that you need to replicate only the openocd plug-in, not the whole repository.

jonahgraham, 2020-08-05:
> For that, I need to replicate gnumcueclipse plugins in my repo and build a feature. I think from the code maintenance perspective, that's going to be tedious.

@kolipakakondal  I assume you mean git repo - you can have a feature in your git repo that includes plug-ins not in your git repo. The feature in this case would be called something like "com.espressif.idf.debug.openocd.feature"

If you did mean p2 repo, p2 can point to additional repos for content.

(HTH - this bug report has gone in multiple different directions, so I hope I have answered the question you were originally asking.)

kolipakakondal, 2020-08-20:
I'm sorry for replying late on this. I'm talking about git repo @jonahgraham. Maybe I'm complicating the issue, I'm sorry!

com.espressif.idf.debug.gdbjtag.openocd <<< Here is my plugin and it's dependecies
   -  ilg.gnumcueclipse.debug.core
   -  ilg.gnumcueclipse.debug.gdbjtag
   -  ilg.gnumcueclipse.managedbuild.cross

How do I get above dependencies to my target Eclipse without installing the "Embedded CDT OpenOCD Debugging" feature? Is it possible? Could you please point to some resource to achieve this? That would be a great help.

 Please find the below attachment. As you can see I've two OpenOCD Debugging configurations.  All I want is to get rid of "GDB OpenOCD Debugging" option

[image]

 Thanks for your reply @jonahgraham @ilg-ul

ilg-ul, 2020-08-20:
The plug-in that provides the OpenOCD functionality also contributes the 'GDB OpenOCD Debugging' entry to the GUI. With the current structure of the code, I don't think you can get one without the other.

The same happened with the 'GDB Hardware Debugging', my plug-ins inherit functionality from them.

ruspl-afed, 2020-08-20:
@kolipakakondal In that case you can try to suppress the unwanted contribution with "activities", please have a look at this article http://blog.vogella.com/2009/07/13/eclipse-activities/

kolipakakondal, 2020-08-20:
Thanks @ruspl-afed @ilg-ul,  I'll check the link.

kolipakakondal, 2020-08-20:
I'm closing this issue as we can't separate the plugins. Thank you all.

closed 2020-08-20 by kolipakakondal

jonahgraham, 2020-08-20:
Hi @kolipakakondal - it looks like a resolution has been achieved here. 

The GDB Hardware Debugging UI and Core mix can be split into separate bundles and CDT would accept that patch. I assume the same is true for separating OpenOCD, if anyone has the time and desire to do it.

kolipakakondal, 2020-08-24:
Sure @jonahgraham, I can take a look at this. It might take some time for me considering my work schedule.

ilg-ul, 2020-11-09:
Given that the code is under major restructuring, we can revisit this issue.

I already added a Core feature, but for now it includes only the top core plug-in.

The issue with OpenOCD is also open, I tried to spit it into `.core` and `.ui`, but failed, since I could not identify the non-UI classes that should go into `.core`.