eclipse-embed-cdt/eclipse-plugins #407: Add Aarch64 support

ilg-ul, 2020-08-12:
After Eclipse Platform and Eclipse CDT were updated to support Aarch64, I tried to integrate everything and run my EPP build script, but I encountered something possibly related:

```
[INFO] Resolving dependencies of MavenProject: org.eclipse.epp:epp.package.embedcdt:4.17.0-SNAPSHOT @ /Users/ilg/My Files/WKS Projects/gnu-mcu-eclipse.github/org.eclipse.epp.packages.git/packages/org.eclipse.epp.package.embedcdt.product/pom.xml
[INFO] {osgi.os=linux, osgi.ws=gtk, org.eclipse.update.install.features=true, osgi.arch=aarch64}
[ERROR] Cannot resolve project dependencies:
[ERROR]   Software being installed: epp.package.embedcdt 4.17.0.qualifier
[ERROR]   Missing requirement: org.eclipse.cdt.autotools.core 2.1.0.202007040820 requires 'osgi.ee; (&(osgi.ee=JavaSE)(version=1.8))' but it could not be found
[ERROR]   Cannot satisfy dependency: epp.package.embedcdt 4.17.0.qualifier depends on: org.eclipse.equinox.p2.iu; org.eclipse.cdt.autotools.feature.group 0.0.0
[ERROR]   Cannot satisfy dependency: org.eclipse.cdt.autotools.feature.group 10.0.0.202007040820 depends on: org.eclipse.equinox.p2.iu; org.eclipse.cdt.autotools.core [2.1.0.202007040820,2.1.0.202007040820]
[ERROR] 
```

Also, while working on adding Aarch64 support to CDT we encountered an error in

```
Stand-Alone C/C++ Debugger Product 10.0.0-SNAPSHOT . FAILURE [  0.065 s]

22:52:16 
[ERROR] Failed to execute goal org.eclipse.tycho:tycho-packaging-plugin:1.7.0:build-qualifier-aggregator (default-build-qualifier-aggregator) on project org.eclipse.cdt.debug.application.product: Execution default-build-qualifier-aggregator of goal org.eclipse.tycho:tycho-packaging-plugin:1.7.0:build-qualifier-aggregator failed: Could not resolve plugin org.eclipse.equinox.launcher.gtk.linux.aarch64_null; Path to dependency: -> [Help 1]
```
The full log is in:

https://ci.eclipse.org/cdt/job/cdt-verify-test-cdt-ui-only-pipeline/2099/console

For the moment I removed the references to aarch64, but this probably still needs some attention.

---

To me this seems like the Equinox package is not happy with aarch64.

ruspl-afed, 2020-08-12:
@ilg-ul please post a link to a branch or Gerrit review here to have a look at changes you did.

ilg-ul, 2020-08-12:
I first created the `embed-cdt-develop` branch from `embed-cdt`, which is my main branch, merged the last `master` and updated  all `embed-cdt` related folders.

The build passed, and I got the 3 archives with the 2020-09-M2 packages.

Then I added `aarch64` to the array of environments:

https://github.com/gnu-mcu-eclipse/org.eclipse.epp.packages/commit/d51ae3273566727668cf4a9a9992e5ebc47e8b05

And the build failed with the unresolved dependencies.

ilg-ul, 2020-08-12:
I checked the rt.equinix.* repos. 

If we take `ppc64le` as reference, most references to `ppc64le` are followed by references to `aarch64`, so apparently `aarch64` should have the same functionality as `ppc64le`.

However, since there are no packages built for `ppc64le`, this might not be enough.

Checking for `x86_64` returns way more references, which probably explains why `aarch64` is not yet functional.

For example there is no `org.eclipse.equinox.security.linux.aarch64` bundle, to name only an easily visible difference.

It looks like the Equinox repos need some more `aarch64` definitions.

jonahgraham, 2020-08-12:
> Also, while working on adding Aarch64 support to CDT we encountered an error in

This is fully expected as CDT does not build against a target platform that includes Aarch64 - See [bug 565836 c30](https://bugs.eclipse.org/bugs/show_bug.cgi?id=565836#c30)

jonahgraham, 2020-08-12:
The key line in your error is:

> [ERROR]   Missing requirement: org.eclipse.cdt.autotools.core 2.1.0.202007040820 requires 'osgi.ee; (&(osgi.ee=JavaSE)(version=1.8))' but it could not be found

but  that looks unrealted to aarch64 - presumably your build works fine without https://github.com/gnu-mcu-eclipse/org.eclipse.epp.packages/commit/d51ae3273566727668cf4a9a9992e5ebc47e8b05 ?

ilg-ul, 2020-08-12:
>  presumably your build works fine without gnu-mcu-eclipse/org.eclipse.epp.packages@d51ae32 ?

that's correct. without that aarch64 added, the build is fine with Java 11.

ilg-ul, 2020-08-12:
> CDT does not build against a target platform that includes Aarch64

CDT? you mean EPP?

how old is the platform used to build 2020-09-M2? 

Alexander mentioned that the aarch64 changes were merged into master about a week ago.

jonahgraham, 2020-08-12:
I don't see any sign of you adding p2 repos that have aarch64 yet? The simrel (https://download.eclipse.org/releases/2020-09/ and https://download.eclipse.org/staging/2020-09/) does not yet have aarch64 in it yet. It should have it when the 2020-09 M3 is ready on 27 Aug. In the meantime you need to point at pre-published repos (e.g. https://ci.eclipse.org/cdt/job/cdt-master/lastSuccessfulBuild/artifact/releng/org.eclipse.cdt.repo/target/repository/ for CDT and suitable URL chosen from https://download.eclipse.org/eclipse/downloads/ - choose a recent Integration build aka "I-build")

jonahgraham, 2020-08-12:
> how old is the platform used to build 2020-09-M2?

It will be the Platform I-build from around that date (29 July's build IIRC).

If you want to know how the simrel is populated and what is in the current staging p2 repo see https://git.eclipse.org/c/simrel/org.eclipse.simrel.build.git/tree/ (when embed-cdt is ready, a new file will be added there with the pointer to the embed-cdt p2 repo to contribute to simrel)

jonahgraham, 2020-08-12:
> CDT? you mean EPP?

Note that it is bad practice to build your project that you are contributing to simrel against simrel. However lots of projects do it. CDT does not, the CDT target platform points at the individual dependencies we need so that CDT can control what it gets. Generally we update to new Eclipse Platform milestones only (e.g. every 3 weeks or so). So CDT currently uses the 20200729 build of Platform and will change again next Friday.

The reason we update on milestones only is that milestones are more rigorously tested than nightly (aka Integration) builds.

ilg-ul, 2020-08-12:
> The simrel (https://download.eclipse.org/releases/2020-09/ and https://download.eclipse.org/staging/2020-09/) does not yet have aarch64 in it yet.

Ah, that explains it.

> you need to point at pre-published repos (e.g. https://ci.eclipse.org/cdt/job/cdt-master/lastSuccessfulBuild/artifact/releng/org.eclipse.cdt.repo/target/repository/ for CDT and suitable URL chosen [...]

The only definition that I see in my EPP fork is:

```
    <eclipse.simultaneous.release.repository>http://download.eclipse.org/staging/2020-09/</eclipse.simultaneous.release.repository>
```

If there is any nightly or integration build that includes the the platform and CDT with Aarch64 support, I can test it, otherwise I'll wait for M3, the EPP code is too complicated for me to make it point to different locations.

jonahgraham, 2020-08-12:
> If there is any nightly or integration build

No, at the moment the simrel (staging repo)  the updates to the aggr files I [linked to ](https://git.eclipse.org/c/simrel/org.eclipse.simrel.build.git/tree/) are normally only done with each M and RC build.

ilg-ul, 2020-08-12:
> No ...  M and RC build.

Then we'll resume testing next Friday using M3.

jonahgraham, 2020-08-12:
> Then we'll resume testing next Friday using M3.

On Friday 21st, Platform contribute their M3 build and then CDT starts using it. Then on Mon 24th, CDT contributes their M3 build. In the end all the projects update and the new complete staging repo is ready on 27 Aug. See https://wiki.eclipse.org/Category:SimRel-2020-09 for more info if desired.

ilg-ul, 2020-08-28:
With 2020-09-M3, the build passed.

closed 2020-08-28 by ilg-ul