eclipse-embed-cdt/eclipse-plugins #409: GDB path not set correct

mgiaco, 2020-08-24:
Hi,
I use the latest tools (xpacks gcc, xpack openocd and build tools, Eclipse 2020-06 with latest updates and openjdk14). I think there is something wrong in the gdb path setting. 
When I start a GDB session eclipse do not use the xpack path. 
When I set the Executable name in the GDB Client Setup in the debug launch setting everything works fine.

ilg-ul, 2020-08-24:
> When I start a GDB session eclipse do not use the xpack path.

Then you have a path problem. 

Download a new eclipse package, install it in a separate location, create a new workspace and run the blinky arm tutorial.

ilg-ul, 2020-08-28:
Any progress?

mgiaco, 2020-08-31:
Hi, I found out that the java version is the problem i switched to openjdk 14. Made a new worksspace and everything works fine.

closed 2020-08-31 by mgiaco

ilg-ul, 2020-08-31:
Your initial message also mentioned openjdk14. What platform are you using?

mgiaco, 2020-08-31:
The problem was eclipse did not start with openjdk 14.

ilg-ul, 2020-08-31:
And now what java are you using?

mgiaco, 2020-08-31:
openjdk 14.

ilg-ul, 2020-08-31:
Aha, so you claim that 'eclipse did not start with openjdk 14' and that now you are using openjdk 14. Does it make any sense?

mgiaco, 2020-08-31:
On my machine there was the latest open version from Oracle installed (java version "1.8.0_261") and openjdk 14. But the path was pointed to the Oracel java and i did not check that. So for the first time i taught I am using openjdk 14.... 
No I added the following line in eclipse.ini 
-vm
c:\openjdk-14.0.2_windows-x64_bin\jdk-14.0.2\bin\

and now everything works fine.

ilg-ul, 2020-08-31:
Right, all recent Eclipse require Java 11, this is already mentioned in the old web site.

I added a Prerequisites section in the download page:

https://projects.eclipse.org/projects/iot.embed-cdt/downloads