eclipse-embed-cdt/eclipse-plugins #426: Add an org.eclipse.embedcdt feature

ilg-ul, 2020-11-04:
Add a top feature to install most other features, similar to the Marketplace configuration.

ilg-ul, 2020-11-04:
Feature added, but for now only the `org.eclipse.embedcdt.core` plug-in is there.

@jonahgraham, @ruspl-afed, any suggestion what plug-ins should be listed by the main feature?

If I got it right, CDT pulls several `.core` plug-ins.

My main goal of adding this feature is somehow hilarious, but it was impossible to identify the top version only by looking at the list of features, you had to go to the list of plug-ins and identify `ilg.gnumcueclipse.core`, which was not exactly user friendly.

jonahgraham, 2020-11-04:
CDT pulls in a set of plug-ins for basic functionality. I am not sure how to replicate that for embed-cdt because you probably want a managedbuilder and debugger plug-in for that - but how do you choose what belongs in that set. Your original comment said "top feature to install most other features" - I would agree with that, put it (almost) all in.

ilg-ul, 2020-11-04:
Ok, in this case I'll refer to the same features as the Marketplace.

ilg-ul, 2020-11-04:
Fixed on 2020-11-05.

closed 2020-11-04 by ilg-ul

ilg-ul, 2020-11-05:
Unfortunately including this feature alone somehow prevents the p2 trick to trigger and I ended with duplicated plug-ins.

I had to revert it to the initial almost empty content, serving only to show the core version.