eclipse-embed-cdt/eclipse-plugins #431: Releases should have the top version, not a plug-in version

ilg-ul, 2020-11-07:
Currently the features inherit somehow the version of the main plug-in.

Change it to the top version.

@ruspl-afed, any suggestion how to fix this?

ruspl-afed, 2020-11-07:
@ilg-ul I'm not sure that I understood what is needed to do. Please provide a bit more details: what is now and what is desired. Technically the version of feature and its main (branding?) plug-in can be different.

ilg-ul, 2020-11-07:
In CDT all features use the top/branding version, like 10.0.0.

In Embed CDT, currently features display the version of one of the plug-ins, which is fine if the feature includes only one plug-in, but not so fine if the feature includes multiple plug-ins.

The main problem I have with the current versioning scheme is that the release/branding version is not visible when seeing the list of features during install. I tried to improve this by adding a 'Core' feature, which includes the `.core` plug-in, which has the top version, but it is still confusing.

My proposal is to use the same versioning scheme as CDT, i.e. all features to have the top/branding version.

ilg-ul, 2020-11-07:
BTW, if you take a look at the code, please use the `develop-v6` branch.

ruspl-afed, 2020-11-07:
OK, now it is more clear for me. Personally I prefer to have the "top/branding" version for all the top-level features. By "top-level feature" I mean the feature that can be installed/uninstalled by the user. All the "internal" features are versioned according to SemVer rules.

ilg-ul, 2020-11-07:
CDT uses explicit `version="10.0.0.qualifier"` for all features.

Unless you know a trick with a variable that expands to the top version, we can do the same.

ilg-ul, 2020-11-07:
> top-level/internal features

I think we don't have internal features in Embed CDT, all are installable by the user.

ruspl-afed, 2020-11-07:
>  a trick with a variable that expands to the top version

I would avoid implementing such a trick, it will add a lot of headache without notable savings.

ilg-ul, 2020-11-07:
Ok, I changed all explicit versions to `6.0.0`; they'll be updated by a bulk search/replace, it shouldn't be a problem.

ilg-ul, 2020-11-07:
Fixed on 2020-11-07.

closed 2020-11-07 by ilg-ul