eclipse-embed-cdt/eclipse-plugins #432: Check if feature 'license-feature' is needed

ilg-ul, 2020-11-07:
The CDT features refer to `org.eclipse.license`, something like this:

```
<feature
      id="org.eclipse.cdt"
      label="%featureName"
      version="10.0.0.qualifier"
      provider-name="%providerName"
      license-feature="org.eclipse.license"
      license-feature-version="0.0.0">
...
```

@jonahgraham, @ruspl-afed, should Embed CDT do the same?

ruspl-afed, 2020-11-07:
Yes, to reuse a lot of EPL2 related content from its official source

ilg-ul, 2020-11-07:
If I add this reference, should I remove some local content?

ruspl-afed, 2020-11-07:
Yes, 
1) license files and 

2) license info from `feature.properties`
`licenseURL`
`license`

ilg-ul, 2020-11-07:
Great, I'll do it.

Thank you!

ilg-ul, 2020-11-08:
I identified the project:

https://git.eclipse.org/c/cbi/org.eclipse.license.git/

However it has no README, nor any other documentation page, so using it is totally obscure.

I added the two attributes to all `<feature>` definitions:

```xml
<feature
      id="org.eclipse.embedcdt"
      label="%featureName"
      version="6.0.0.qualifier"
      provider-name="%providerName"
      plugin="org.eclipse.embedcdt.core"
      license-feature="org.eclipse.license"
      license-feature-version="0.0.0">

</feature>
```

I removed the `licenseURL` and `license` definitions from `feature.properties`

```
licenseURL=license.html
license=\
Eclipse Foundation Software User Agreement\n\
...
```

And I also removed the `license.html` file.

The result is that the build fails:

```console
ilg@wks eclipse-plugins.git % mvn clear verify
[INFO] Scanning for projects...
[INFO] Computing target platform for MavenProject: org.eclipse.embedcdt.plugins:org.eclipse.embedcdt.core:6.0.0-SNAPSHOT @ /Users/ilg/My Files/WKS Projects/eclipse-embed-cdt.github/eclipse-plugins.git/plugins/org.eclipse.embedcdt.core/.polyglot..META-INF_MANIFEST.MF
[INFO] Fetching p2.index from https://download.eclipse.org/releases/2020-09/202009161000/ (142B)
[INFO] Adding repository https://download.eclipse.org/releases/2020-09/202009161000
[INFO] Resolving dependencies of MavenProject: org.eclipse.embedcdt.plugins:org.eclipse.embedcdt.core:6.0.0-SNAPSHOT @ /Users/ilg/My Files/WKS Projects/eclipse-embed-cdt.github/eclipse-plugins.git/plugins/org.eclipse.embedcdt.core/.polyglot..META-INF_MANIFEST.MF
[INFO] Resolving class path of MavenProject: org.eclipse.embedcdt.plugins:org.eclipse.embedcdt.core:6.0.0-SNAPSHOT @ /Users/ilg/My Files/WKS Projects/eclipse-embed-cdt.github/eclipse-plugins.git/plugins/org.eclipse.embedcdt.core/.polyglot..META-INF_MANIFEST.MF
[INFO] Computing target platform for MavenProject: org.eclipse.embedcdt.features:org.eclipse.embedcdt:6.0.0-SNAPSHOT @ /Users/ilg/My Files/WKS Projects/eclipse-embed-cdt.github/eclipse-plugins.git/features/org.eclipse.embedcdt-feature/.polyglot.feature.xml
[INFO] Resolving dependencies of MavenProject: org.eclipse.embedcdt.features:org.eclipse.embedcdt:6.0.0-SNAPSHOT @ /Users/ilg/My Files/WKS Projects/eclipse-embed-cdt.github/eclipse-plugins.git/features/org.eclipse.embedcdt-feature/.polyglot.feature.xml
[INFO] {osgi.os=linux, osgi.ws=gtk, org.eclipse.update.install.features=true, osgi.arch=x86_64}
[ERROR] Cannot resolve project dependencies:
[ERROR]   Software being installed: org.eclipse.embedcdt.feature.group 6.0.0.qualifier
[ERROR]   Missing requirement: org.eclipse.embedcdt.feature.group 6.0.0.qualifier requires 'org.eclipse.equinox.p2.iu; org.eclipse.license.feature.group 0.0.0' but it could not be found
[ERROR] 
[ERROR] See http://wiki.eclipse.org/Tycho/Dependency_Resolution_Troubleshooting for help.
[ERROR] Cannot resolve dependencies of MavenProject: org.eclipse.embedcdt.features:org.eclipse.embedcdt:6.0.0-SNAPSHOT @ /Users/ilg/My Files/WKS Projects/eclipse-embed-cdt.github/eclipse-plugins.git/features/org.eclipse.embedcdt-feature/.polyglot.feature.xml: See log for details -> [Help 1]
[ERROR] 
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR] 
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/MavenExecutionException
ilg@wks eclipse-plugins.git % 
```

I saw CDT uses a `cdt.target` file with some definitions that seem related to this issue, but I don't use such a file, and I don't know how to fix it.

@ruspl-afed, any suggestions?

jonahgraham, 2020-11-08:
Have a look at https://wiki.eclipse.org/CBI#CBI_license_bundle it shows you the repo you need to add.

jonahgraham, 2020-11-08:
https://help.eclipse.org/2020-09/index.jsp?topic=%2Forg.eclipse.pde.doc.user%2Ftasks%2Fpde_shared_license.htm also has docs on this.

ilg-ul, 2020-11-08:
Thank you, Jonah, adding the repo fixed the problem:

```xml
		<repository>
			<id>license-feature</id>
			<url>http://download.eclipse.org/cbi/updates/license/</url>
			<layout>p2</layout>
		</repository>
```

I also removed the license.html from `build.properties`.

I also removed the `<license>` element from `feature.xml`:

```xml
   <license url="%licenseURL">
      %license
   </license>
```

The documentation does not mention if it is required or not, but since I removed the actual definitions from `feature.properties`, I though they are no longer needed.

ilg-ul, 2020-11-08:
https://ci.eclipse.org/embed-cdt/job/build-plug-ins/job/develop-v6/11/

Fixed on 2020-11-08.

closed 2020-11-08 by ilg-ul